MarlinFirmware/Marlin #20628: BLTOUCH V3.1 MARLIN 2.0 does not deploy or self test tho a cmd is sent [BUG] (short description)

cobra777, 2021-01-02:
### Bug Description

MARLIN 2.0.07 bug fix branch -- CREALITY ENDER 3 4.2.2 Motherboard STM32F103RET6 at its core, BLTOUCH V3.1

Other than turning on power initially the BLtouch does not physically deploy, stow, or self-test.  Obviously I can not do a G28 Homing as it will crash into the bed everytime due this issue.

Of course after successful compile (visual studio + platform.io) with pronterface I have tried issuing the following cmds with the results below:

M280 P0
Which reads out 90
M280 P0 S10
Which reads out 10

However, there is no physical change on the bltouch.

### Configuration Files

attached is my config and config_adv

### Steps to Reproduce

Checkout marlin 2.0.x bugfix
Checkout marlin config

Setup for STM32F103RET6_CREALITY platform

using visual studio with platform.io compile.

**Expected behavior:**

From either pronterface or lcd menu, it should deploy when you hit the deploy button
it should stow when u hit the stow button, etc.

**Actual behavior:**

BLtouch remains up with the orange light on at all times.  If a G28 command is issued it will crash into bed everytime.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-01-02:
This is an issue with pwm plug of the BLtouch,

there are two ways to plug it in.

1) the default, is to plug it in to the 5 pin bltouch socket. It must be wired correctly. 

[image] 

2) or your using a "pin 27 board" that plugs the servo plug into the LCD connector, in this case you must HAS_PIN_27_BOARD
add #define HAS_PIN_27_BOARD to your Configuration.h

Either way, these are support questions. 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-01-02 by ellensp

cobra777, 2021-01-02:
Thank u this solved my issue!!!!

ellensp, 2021-01-02:
and for other who see this the issue was?

cobra777, 2021-01-02:
Issue was i had mistakenly cloned wrong branch.

Upon cloning the correct branch bugfix-2.0

I was able to issue the #define has pin 27

As i was using the lcd breakout board with the bltouch plugged into that....

Worked like a charm!!! Thank you again for swift response!!!!