MarlinFirmware/Marlin #20638: [BUG] (HW PWM not working for stm32f1 PA0)

zbyrek, 2021-01-02:
### Bug Description

In boards which use PA0 of STM32F1 (like creality V4) hardware PWM is not working due to conflict with timer2 which is used for temp interrupt, changing temp timer to timer4 didn't resolve issue. 

### Configuration Files

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5760873/config.zip)

Uncommented:
`#define FAST_PWM_FAN`

Commented out:
`#define FAN_SOFT_PWM`

*also one in: pins_CREALITY_V4.h*

### Steps to Reproduce

1. Compile code for creality board (or other STM32F1 which uses PA0 for fan)
2. Upload to printer
3. Try to change cooling fan percantage

**Expected behavior:**

There should be PWM signal at gnd fan connection (positive one is always +24V it is controled by N-channel mosfet) with duty depending on set percentage.

**Actual behavior:**

Solid GND signal, even for 0%, fan always run at 100% nevertheless of setting

.

#### Additional Information
Sorry for crude photo, but i wanted to show both: printer lcd and scope:
[image]

Software pwm is working alright however i need hardware pwm, because sw is only 7,8Hz which is too low for laser.

I think about two solutions possible:
1. Change pwm timer of port PA0 to timer5 or timer8:
[image]

2. Change temp timer to other than 2, however i think (correct me if i'm wrong) that interrupt from that timer is also used in other parts of fw.

rhapsodyv, 2021-01-02:
It only happen on PA0? do you need to use specifically this pin?

I'm asking it, so we can have a better clue if it's a peripheral conflict on PA0, or if there's a general bug on fast pwm...

The fastest way to check of a peripheral conflict, is to use STM Cube MX, and configure all peripheral currently used. It will tell conflicts and invalid combinations.

rhapsodyv, 2021-01-02:
Two random examples of peripheral conflict: On F4 mcu, if we are using OTG, we can't use TIM8; If we are using SPI1 on F4, we can't use PA6 for PWM.

zbyrek, 2021-01-02:
To be honest i didn't try with other pin, because that pin is dedicated to fan0 and have mosfet, two other pins that are avalaible on that board are: PB1 and PB0, however they are dedicated to bltouch, so they are directly connected to mcu which means if i want use them for fan/laser i need to add additional hw (which i could do but prefer to found sw solution if exist)

sjasonsmith, 2021-01-02:
I am testing some changes right now.

sjasonsmith, 2021-01-02:
Well, I thought it would be a simple change but I don't have my scope in the room with this printer and something is still wrong.

PWM output should be possible with PA0 on TIM2 and TIM5 (not TIM8), but something is still not quite working right with any of the configurations I've tried.

zbyrek, 2021-01-02:
I also tried with tim5, my solution was:
1. in CHITU_F103\board.cpp:
change line 82:
` {&gpioa, &timer2, &adc1,  0, 1,    0}, /* PA0 */`
to:
` {&gpioa, &timer5, &adc1,  0, 1,    0}, /* PA0 */`

2. in HAL\STM32F1\timers.h:
change line 72:
` #define STEP_TIMER_NUM      5  // for other boards, five is fine.`
to:
`#define STEP_TIMER_NUM      4  // for other boards, five is fine.`

effect was different: at 0% it turns off fan, but at any % > 0 (even 1%) it goes at 100% fan speed.

sjasonsmith, 2021-01-02:
Why are you changing a CHITU variant for a Creality board?

zbyrek, 2021-01-02:
As far as i see there is no dedicated creality, it uses same file (or VSC is lying to me)

zbyrek, 2021-01-02:
Ok, i found where is hidden board.cpp for creality, it seems that it is provided by platformio itself, it sits in: `%USERPROFILE%\.platformio\packages\framework-arduinoststm32-maple\STM32F1\variants\generic_stm32f103r`

rhapsodyv, 2021-01-04:
I don't think worth fixing it on current STM32F1 hal (maple), because probably we don't have this issue on the new STM32 hal.
But as we didn't migrate this board to STM32 yet, so I think we should keep this open, until we migrate it, so we can test this behaviour.

GermanJack, 2021-02-04:
I think everybody who tries to connect a laser to the fan connector on the 4.2.7 board will have this problem.
So, I would be happy when this issue will be solved.

GabrieleMezzera, 2021-02-09:
Hi, is there any update on this problem? I'm using a Creality Ender 3 V2 which mounts a Creality 32-Bit Motherboard V4.2.2 and I'm trying to use a laser for some "8-bit greyscale Engraving" but at the moment if i set PWM < 255 the laser (and the same does the part cooling fan) it just blinks because of a really slow PWM Frequency. Reading @zbyrek comments i tried to "force" the HW PWM to work hoping in an higher frequency but i haven't got any luck, the frequency didn't change at all. 

Anyway... after editing "Configuration.h" commenting "#define FAN_SOFT_PWM" out and uncommenting "#define FAST_PWM_FAN" i've tried to:

Try 1:

- Setting "#define STEP_TIMER_NUM 4 // for other boards, five is fine." in "Marlin/src/HAL/STM32F1/timers.h"
- Setting "{&gpioa, &timer5, &adc1, 0, 1, 0}, /* PA0 */" in "/Users/%USERPROFILE%/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/boards.cpp"

Try 2:

- Leaving "#define STEP_TIMER_NUM 5 // for other boards, five is fine." in "Marlin/src/HAL/STM32F1/timers.h"
- Setting "{&gpioa, &timer5, &adc1, 0, 1, 0}, /* PA0 */" in "/Users/%USERPROFILE%/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/boards.cpp"

NB. Every time i recompiled the firmware i started from an untouched version of the "Marlin Bugfix-2.0.x" Frimware

Am i doing something wrong? Does anybody have any suggestion?

zbyrek, 2021-02-09:
> Reading @zbyrek comments i tried to "force" the HW PWM to work hoping in an higher frequency but i haven't got any luck, the frequency didn't change at all.

Did you removed `#define FAN_SOFT_PWM` which is hardcoded in `Marlin\src\pins\stm32f1pins_CREALITY_V4.h`?

GabrieleMezzera, 2021-02-09:
> > Reading @zbyrek comments i tried to "force" the HW PWM to work hoping in an higher frequency but i haven't got any luck, the frequency didn't change at all.
> 
> Did you removed `#define FAN_SOFT_PWM` which is hardcoded in `Marlin\src\pins\stm32f1pins_CREALITY_V4.h`?

Nope 😅

Anyway now i've tried removing that too... Being honest i'm feeling kinda dumb. Everything is working like a charm! Thanks a lot @zbyrek !

Just another quick query... Did you noticed any problem occurring when 3D Printing using this firmware? Or do you simply swap the firmwares when needed?

P.S. For anyone who's interested here's the full list of changes to the firmware:

- Set "{&gpioa, &timer5, &adc1, 0, 1, 0}, /* PA0 */" in "/Users/%USERPROFILE%/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/boards.cpp"
- Commented "#define FAN_SOFT_PWM" in "Marlin/src/pins/stm32f1/pins_CREALITY_V4.h" 
- Commented "#define FAN_SOFT_PWM" in "Marlin/Configuration.h" (DO NOT comment out "#define SOFT_PWM_SCALE 0")
- Set "#define STEP_TIMER_NUM 4 // for other boards, five is fine." in "Marlin/src/HAL/STM32F1/timers.h"

zbyrek, 2021-02-09:
> Just another quick query... Did you noticed any problem occurring when 3D Printing using this firmware? Or do you simply swap the firmwares when needed?

Well i don't swap firmwares however I added custom gcode to change pwm frequency with it, and i keep low frequency for fan and change to high when need laser, because high frequency for fan have two sides effects:
1. noise - it is soudn frequency, so motor in fan generates that sound
2. problems with low pwm values, DC motor in fan works like low pass filter, and short, quick pulses don't work very well, so it would starts spinning at higher pwm rates, also it could slip.

GabrieleMezzera, 2021-02-09:
Makes perfectly sense... In the end Noise is somehow acceptable, the most worrying problem is that LPF that can effectively lead to some hardware damage.
Using G-Code to control the PWM Frequency sounds awesome. Anyway, while hoping that in a not so far future your Feature Request #20642 will get implemented, i'll go with Custom G-Code. Thanks a lot man!

Mnnovak95, 2021-02-20:
Hello, i have the same problem with Ender 3V2 (mainboard 4.2.2 32BIT) with LASER 500mw. Im spending alot of hours trying solve this, so i hope someone can help me to solve it. Im absolutely begginer and this was first time, i was working in visual studio. Problem started, when i connect laser to hotend fan port, outline engraving worked perfect, but full engraving/rastering looks so ugly, it looks like there is slow PWM to ON/OFF laser, thats why it looks so bad. Before i found this issue here, i tried a lot of configurations/extensions in inkscape, but it must work only with CREALITYWORKSHOP - blackwhite edit, because same gcode and 500mw laser works fine on ENDER 5 printer... So here its problem somewhere in SW/HW on ender 3V2.. 

So what do i need to do to solve this? New mainboard? OR some setup? I tried  use //#define FAN_SOFT_PWM" in "Marlin/src/pins/stm32f1/pins_CREALITY_V4.h"  for example, but after that, laser is still ON. The same happen after i delete this comment.

Cant find boards.cpp for creality, only in CHITU_F103, but i dont know how to acess boards for creality via visual studio core

How im doing it?

Downloaded Marlin 2.0 zip - unzip
Downloaded configurations zip - unzip
Copy configurations Ender 3v2 to Marlin folder as show in manual
Open Visual studio core - platformio
Open folder marlin 2.0
Set STM32F103RET6_creality in platformio.ini

Than tried //#define FAN_SOFT_PWM in creality_V4.H
Tried use //#define FAN_SOFT_PWM" in "Marlin/Configuration.h"
Tried add #define STEP_TIMER_NUM 4 // for other boards, five is fine." in "Marlin/src/HAL/STM32F1/timers.h" - Maybe here needs to be add // before #define ??
Tried change timer to 5 in boards.cpp but i found boards.cpp only CHITU_F103

Hopefully someone can help to try solve this problem, or does it work for example with SKR E3 mainboard ?? 
THANK YOU SO MUCH
GREETINGS FROM CZECH

Mnnovak95, 2021-02-20:
[image]

zbyrek, 2021-02-21:
> Cant find boards.cpp for creality, only in CHITU_F103, but i dont know how to acess boards for creality via visual studio core

You can't find that board, because it's not a part of marlin but it uses definitions from paltformio.

* Open windows explorer
* Type in path bar: `%USERPROFILE%\.platformio\packages\framework-arduinoststm32-maple\STM32F1\variants\generic_stm32f103r`
* Here is your `board.cpp` where you need make changes:
```diff
-        {&gpioa, &timer2, &adc1,  0, 1,    0}, /* PA0 */
+        {&gpioa, &timer5, &adc1,  0, 1,    0}, /* PA0 */
```

GabrieleMezzera, 2021-02-21:
Hi, @Mnnovak95, i believe you missed some steps and don't worry, so did i the first time i tried ;)

So, i think you missed to:

- Uncomment `#define FAST_PWM_FAN` in `Marlin/Configuration.h`

- Set `{&gpioa, &timer5, &adc1, 0, 1, 0}, /* PA0 */`in `/Users/%USERPROFILE%/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/boards.cpp` (By default it should be `&timer2` so you only need to change the number at the end from 2 to 5.) 

Please note that, for the 2nd step, as it can be seen in the path, this `boards.cpp` file isn't located into the marlin folder since it's already provided by PlatformIO itself. So you should look for it into the `.platformio` folder which location changes on the OS you're running. The official docs say:

> The default value is the user’s home directory:
> 
> Unix (Linux or MacOS): `~/.platformio`
> Windows: `%HOMEPATH%\.platformio`

Hope this helps

GabrieleMezzera, 2021-02-21:
Ok, @zbyrek is definitely faster than me

Mnnovak95, 2021-02-21:
thank you so much for fast reply @zbyrek and @GabrieleMezzera. I changed boards.cpp via text editor now and uncomment #define fast pwm, but now i have this error. Maybe i need to use marlin 2.0.x BUGFIX? Im trying it with Marlin 2.0.X.. I will try this process again and i will let you know. 

Thank you !!
[image]

GabrieleMezzera, 2021-02-21:
@Mnnovak95 being honest i never tried using the stable version. I've used the bugfix branch for all the tests i've done. so my suggestion is try with the latest `bugfix-2.0.x` (remember to download the right configurations file since it differs between the stable and the bugfix verison).

Anyway @zbyrek is more experienced than me so maybe he can see a problem i don't see.

Mnnovak95, 2021-02-21:
@GabrieleMezzera I think thats my problem ! I tried MARLIN 2.0.7.2 and the same error. I think its because im using configurations for 2.0.x.. I didnt know, that i need to download specific configurations.  Thank you!

zbyrek, 2021-02-21:
There is sanity check in stable version which prevents you from enabling it, you must use `bugfix-2.0.x` for this to work.

Here is [issue #19952](https://github.com/MarlinFirmware/Marlin/pull/19952)

Mnnovak95, 2021-02-21:
@zbyrek I cant figure out whats wrong now, it looks fine with BUGFIX-2.0.x, laser was ON/OFF after upgrade so fast (it makes some new specific noise) but i was not abble to move Z up manualy from settings.. so i think that was because i forget update screen too.. So i updated screen, make the same process via visual studio for new BIN file, and after that, laser is still ON. Tried it for example 5times with the same result. It happen after commented //#define FAN_SOFT_PWM
[image]

Please do you still have working BIN file for creality ender 3v2? I will keep trying, but for me its hard to figure out whats wrong.

Thank you !

zbyrek, 2021-02-21:
> Please do you still have working BIN file for creality ender 3v2?

I have ender 5, also getting some random bin files from strangers is bad idea, they could even break your printer.

Mnnovak95, 2021-02-21:
Well, after xx upgrades its still doing the same. After this setup shown in picture, FAN/LASER is still ON after power ON 3d printer.. I also changed {&gpioa, &timer2, &adc1,  0, 1,    0}, /* PA0 */ to {&gpioa, &timer5, &adc1,  0, 1,    0}, /* PA0 */

Downloaded BUGFIX 2.0 which was last Updated 6 hours ago by marciot - i dont know, if there are any changes from bugfix 2.0 version you used..

Maybe another BUG, thanx for your time @zbyrek  @GabrieleMezzera 

[image]

zbyrek, 2021-02-21:
[image]

Here's your problem, it should look like this:

[image]

trengtor, 2021-02-23:
> Я не думаю, что стоит исправлять это на текущем STM32F1 hal (клен), потому что, вероятно, у нас нет этой проблемы на новом STM32 hal.
> Но поскольку мы еще не перенесли эту плату на STM32, я думаю, что мы должны оставить ее открытой до тех пор, пока мы ее не перенесем, чтобы мы могли проверить это поведение.

https://github.com/MarlinFirmware/Marlin/issues/21129

RA7AH, 2021-04-20:
Hello, I have mounted a Neje Laser on my Ender 3 with a 4.2.2 mainboard and I have the same problem: between 1% and 99% the laser is pulsing. There is a firmware to solve this issue? Thank you.

TheYarkin, 2021-05-13:
Hi, I'm trying to achieve a similar thing to @zbyrek and @GabrieleMezzera but with an SKR mini E3 V2 board, and a laser that has a TTL pin. So I need a 3.3-12V PWM signal and can't use the normal fan pins because they have the MOSFET on the negative terminal (and are 24V anyway).

I have been trying to reprogram the Neopixel pin (PA8) to use it for PWM. Any suggestions on how to do this?

I have enabled FAST_PWM_FAN tried assigning PA8 to FAN2_PIN as well as modifying the board.cpp file similar to above to put that pin on timer5 or timer8. Nothing is working so far. Can I achieve this without making it a fan pin? Thanks!

RA7AH, 2021-05-13:
@TheYarkin Hello, Ender 3 has also a 24V Fan, I have used the schematics below found on internet that uses an 4N25 octocoupler, 2  resistors (470 Ohm and 1k5 Ohm) and a DC DC step down converter. Really easy to made, and it costs less than 10€. For me the only problem is the PWM frequency of the Creality 4.2.2 board mounted on my Ender 3. 
I have tested the laser on my Anycubic Predator (also 24 Volt) and is working fine. 
Hope it helps. Please  see pictures below:

[image]
[image]
[image]
[image]
[image]
[image]

TheYarkin, 2021-05-14:
@RA7AH Thanks for this! I actually found the same post as you and implemented it when I first started on this problem a month ago. 
It works well for simple on/off or low frequency switching, but I really want to do "grayscale" engraving and need the higher frequency. Perhaps the optocoupler I had lying around wasn't fast enough - it didn't work well. Have you tried PWM frequencies ~1kHz?

I originally had the Creality 4.2.2 board and bought the SKR mini because it has a couple of extra pins I could use (plus linear advance), so I'd really love to get this to work through software - I just don't quite understand the interaction between the timers and pin definitions etc., so need help setting them up. The default 50 Hz PWM on that PA8 pin just isn't fast enough - just need to increase that.

RA7AH, 2021-05-14:
@TheYarkin Hello, unfortunately I have not tried with high PWM frequencies. 
Neither I was able to implement the changes on Marlin for the Creality v4.2.2 described by @zbyrek and @GabrieleMezzera ...

HofaTheRipper, 2021-05-25:
> gcode to change pwm frequency

> Well i don't swap firmwares however I added custom gcode to change pwm frequency with it, and i keep low frequency for fan and change to high when need laser, because high frequency for fan have two [...]

How do you set the PWM frequency via gcode? could you please explain this to me.

Thanks!

zbyrek, 2021-05-25:
> How do you set the PWM frequency via gcode?

I wrote piece of my own code to change variable with frequency.

zbyrek, 2021-05-25:
It is fast and dirty piece of code without any checks, so i give no warrnaty that it won't crash anything (especially with wrong input).

Modifications in `gcode.h`:
```diff
...
+   static void M9891();
...
```
Add file ` gcode\M9891.cpp`:
```cpp
#include "../../inc/MarlinConfig.h"

#include "../gcode.h"

void GcodeSuite::M9891() {
  const uint8_t pport = parser.byteval('P');

    const uint16_t dfreq = parser.ushortval('F');

    // Set frequency for port
    
    set_pwm_frequency((pin_t)pport, (int)dfreq);
}
```

I don't remember if there were more modifications (that was quite time ago).

Send to printer:
`M9891 P<portname> F<frequency in Hz>`

HofaTheRipper, 2021-05-25:
Am i right to send to printer: "M9891 PPA0 F10" to set 10Hz? or should it be: M9891 PA0 F10?

HofaTheRipper, 2021-05-25:
Ok i'll try... no permanent change is ok - i will add it to the start gcode for printing and for engraving.

Thank you very much!

vojto1980, 2021-05-26:
> > > Reading @zbyrek comments i tried to "force" the HW PWM to work hoping in an higher frequency but i haven't got any luck, the frequency didn't change at all.
> > 
> > 
> > Did you removed `#define FAN_SOFT_PWM` which is hardcoded in `Marlin\src\pins\stm32f1pins_CREALITY_V4.h`?
> 
> Nope 😅
> 
> Anyway now i've tried removing that too... Being honest i'm feeling kinda dumb. Everything is working like a charm! Thanks a lot @zbyrek !
> 
> Just another quick query... Did you noticed any problem occurring when 3D Printing using this firmware? Or do you simply swap the firmwares when needed?
> 
> P.S. For anyone who's interested here's the full list of changes to the firmware:
> 
> * Set `{&gpioa, &timer5, &adc1, 0, 1, 0}, /* PA0 */` in `/Users/%USERPROFILE%/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/boards.cpp`
> * Commented `#define FAN_SOFT_PWM` in `Marlin/src/pins/stm32f1/pins_CREALITY_V4.h`
> * Uncommented `#define FAST_PWM_FAN` in `Marlin/Configuration.h`
> * Commented `#define FAN_SOFT_PWM` in `Marlin/Configuration.h` (DO NOT comment out `#define SOFT_PWM_SCALE 0`)
> * Set `#define STEP_TIMER_NUM 4 // for other boards, five is fine.` in `Marlin/src/HAL/STM32F1/timers.h`

> > > Reading @zbyrek comments i tried to "force" the HW PWM to work hoping in an higher frequency but i haven't got any luck, the frequency didn't change at all.
> > 
> > 
> > Did you removed `#define FAN_SOFT_PWM` which is hardcoded in `Marlin\src\pins\stm32f1pins_CREALITY_V4.h`?
> 
> Nope 😅
> 
> Anyway now i've tried removing that too... Being honest i'm feeling kinda dumb. Everything is working like a charm! Thanks a lot @zbyrek !
> 
> Just another quick query... Did you noticed any problem occurring when 3D Printing using this firmware? Or do you simply swap the firmwares when needed?
> 
> P.S. For anyone who's interested here's the full list of changes to the firmware:
> 
> * Set `{&gpioa, &timer5, &adc1, 0, 1, 0}, /* PA0 */` in `/Users/%USERPROFILE%/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/boards.cpp`
> * Commented `#define FAN_SOFT_PWM` in `Marlin/src/pins/stm32f1/pins_CREALITY_V4.h`
> * Uncommented `#define FAST_PWM_FAN` in `Marlin/Configuration.h`
> * Commented `#define FAN_SOFT_PWM` in `Marlin/Configuration.h` (DO NOT comment out `#define SOFT_PWM_SCALE 0`)
> * Set `#define STEP_TIMER_NUM 4 // for other boards, five is fine.` in `Marlin/src/HAL/STM32F1/timers.h`

Thanks a lot, this solved the problem with changing the PWM frequency. The turning laser on and off was still a bit slow causing some extra/missing lines during engraving. So I found this commented line 
#define LASER_SYNCHRONOUS_M106_M107 
uncommenting this line the laser became faster response
I'm showing pictures of worst case 7,8 Hz PWM, changed 32 kHz PWM and uncommenting LASER_SYNCHRONOUS_M106_M107
The only disadvantage is loosing of manual control of the fan (the fan/laser is then controlled only by g-code ignoring the manual fan speed settings... :(
Last picture shows connection for the dimming transistor (just cut off the isolation of the laser control PCB inside the laser body) and you can then connect laser fan to +24 and GND and dimm the laser with transistor or optocoupler.

[image]
[image]
[image]
[image]

dcwalmsley, 2021-06-12:
@zbyrek

Need some assistance in compiling the M9891 GCode you posted earlier.

1.  I created the file "M9891.cpp" and placed in the Marlin/src/gcode folder after adding the following.

      #include "../../inc/MarlinConfig.h"
      #include "../gcode.h"
      void GcodeSuite::M9891() {
      const uint8_t pport = parser.byteval('P');
      const uint16_t dfreq = parser.ushortval('F');
      // Set frequency for port
      set_pwm_frequency((pin_t)pport, (int)dfreq);
      }

2.  In "gcode.h" I added "static void M9891();" to the bottom of the file just above, 
     line: 1101    static void M9891();
     line: 1102  
     line: 1103    static void T(const int8_t tool_index);
     line: 1104
     line: 1105    };
     line: 1106
     line: 1107    extern GcodeSuite gcode;

So when I compiled this is the message I received....

Compiling .pio\build\STM32F103RET6_creality\src\src\gcode\M9891.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\gcode\bedlevel\G42.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\gcode\bedlevel\M420.cpp.o
Marlin\src\gcode\M9891.cpp:4:10: fatal error: ../gcode.h: No such file or directory
 #include "../gcode.h"
          ^~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RET6_creality\src\src\gcode\M9891.cpp.o] Error 1

Where is my mistake?

Best Regards,
Doug W

zbyrek, 2021-06-12:
> 1. placed in the Marlin/src/gcode folder

you should place it one level deeper, for example in config or control

dcwalmsley, 2021-06-12:
@zbyrek 

I placed tht cpp file inside "control" folder and compiled with no issues.  Flashed the bin file to my Ender-3 V2 and recycled power with no issues.

However, when I attempt to run the GCode from Ponterface it comes back with "echo:Unknown command: "M9891 PA0 F10"".

Where did I go wrong with you instructions?

HofaTheRipper, 2021-06-12:
> @zbyrek
>
> I placed tht cpp file inside "control" folder and compiled with no issues. Flashed the bin file to my Ender-3 V2 and recycled power with no issues.
>
> However, when I attempt to run the GCode [...]

You have to add:

case 9891: M9891(); break;

to gcode.cpp to get this working.

dcwalmsley, 2021-06-12:
Ok, I did everything I was certain that would work.  Compile completed with no errors but when I attempted to execute M9891 I still receive "echo:Unknown command: "M9891 PA0 F10"".

Attached is a zipped folder of only those files I added or modified including their paths.

Please review and see where I missed something.  I'm not a programmer but I'm learning how to modify files and compile them.  Still not clear how to read or develop code.  

Thanks again.
[Marlin-For Ender-3 Using Laser.zip](https://github.com/MarlinFirmware/Marlin/files/6642533/Marlin-For.Ender-3.Using.Laser.zip)

HofaTheRipper, 2021-06-12:
> Ok, I did everything I was certain that would work. Compile completed with no errors but when I attempted to execute M9891 I still receive "echo:Unknown command: "M9891 PA0 F10"".
>
> Attached is a [...]

You didnt change the platformIO files?

Look at my repo. There is a FastPWM.txt with every change you need to do.

zbyrek, 2021-06-12:
> Ok, I did everything I was certain that would work. Compile completed with no errors but when I attempted to execute M9891 I still receive "echo:Unknown command: "M9891 PA0 F10"".
>
> Attached is a [...]

You shouldn't surround that new command block with if statement since you don't define it anywhere else, remove these preprocessor directives.

```diff
- #if ENABLED(9891)
  { case 9891:M9891();
    break;
  }
- #endif
```

luchomasco, 2021-06-13:
Hi @vojto1980 , Where do you set the frequency value?
Sorry for my noob question. I'm now with this stuff.
Greetings, Luciano.

dcwalmsley, 2021-06-14:
Ok, this is a bit frustrating.  LOL

So I reviewed @HofaTheRipper "FastPWM.txt" file from is repository and noted that all steps he has listed have been performed. I even went so far as to copy and paste each requirement into appropriate file and their locations.

But I still receive errors when compiling.  See next line.
--------------------------------------------------------------
Marlin\src\gcode\gcode.cpp:1131:5: error: expected unqualified-id before 'case'
     case 9891: M9891(); break;
     ^~~~
Marlin\src\gcode\gcode.cpp:1131:25: error: expected unqualified-id before 'break'
     case 9891: M9891(); break;
                         ^~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\gcode\gcode.cpp.o] Error 1
------------------------------------------------------------------

And to be clear, here is the verbatim list of tasks I followed from Hofa's list.

-----------------------------------------------------------------------------------
Set {&gpioa, &timer5, &adc1, 0, 1, 0}, /* PA0 */ in /Users/%USERPROFILE%/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/boards.cpp

Commented #define FAN_SOFT_PWM in Marlin/src/pins/stm32f1/pins_CREALITY_V4.h

Uncommented #define FAST_PWM_FAN in Marlin/Configuration.h

Commented #define FAN_SOFT_PWM in Marlin/Configuration.h (DO NOT comment out #define SOFT_PWM_SCALE 0)

Set #define STEP_TIMER_NUM 4 // for other boards, five is fine. in Marlin/src/HAL/STM32F1/timers.h

********************************************************************************************************
gcode.h:
********************************************************************************************************

...
+   static void M9891();
...

********************************************************************************************************
gcode.cpp:
********************************************************************************************************
...
+   case 9891: M9891(); break;
...

********************************************************************************************************
Add file gcode\M9891.cpp:
********************************************************************************************************

#include "../../inc/MarlinConfig.h"

#include "../gcode.h"

void GcodeSuite::M9891() {
    const uint8_t pport = parser.byteval('P');

    const uint16_t dfreq = parser.ushortval('F');

    // Set frequency for port
    
    set_pwm_frequency((pin_t)pport, (int)dfreq);
}
-----------------------------------------------------------------------------------

zbyrek, 2021-06-14:
I guess (because i didn't see whole code) you put case outside switch (after switch closing `}` )

dcwalmsley, 2021-06-15:
After doing some reviewing, IO now understand what you were referring to.  I moved the line "case 9891: M9891(); break;" up to and between 

     #if ENABLED(MAX7219_GCODE)
        case 7219: M7219(); break;                                // M7219: Set LEDs, columns, and rows
      #endif
          
      case 9891: M9891(); break;                                 //Send GCode command to printer to set port and frequency desired (P<portname> F<frequency in Hz>)

        default:
          parser.unknown_command_warning();
          break;

This allowed the complying to complete with no error message.

I also performed via serial connection and received and "ok" message one I hit return on M9891.  Is there a process to see the entered command was successful other than an "ok" response?

abhinavjha86, 2021-11-12:
want to thank @zbyrek for creating this bug and explaining the issue with a provided fix(work-around), what really helped me getting grayscale laser engraving to work on my Ender 3 v2 in addition to @zybrek changer was:

uncommenting '#define LASER_SYNCHRONOUS_M106_M107' suggested by @vojto1980.

cheers guys👍😊

VukIgor, 2021-11-23:
@descipher are those fixes included in bugfix or Marlin firmware ? Have Creality 4.2.2 board and after compiling bugfix and also Marlin HW pin is still dead ,board is new ,everything works fine even that k1 fan but when i connect laser to it just blinks laser and nothing else

descipher, 2021-11-23:
@VukIgor #23125 was merged after your compile, please use bugfix-2.0.x which now contains the fix.

VukIgor, 2021-11-23:
@descipher thank you very much :) Hope it won't need any aditional coding becouse i'm noob at this ,have few laser engravers running on marlin and ramps 1.4 for more than year without issue

VukIgor, 2021-11-24:
Bugfix works for my creality 4.2.2 boards ,thank you

thisiskeithb, 2021-12-12:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/23125.

closed 2021-12-12 by thisiskeithb

MichalsonCZ, 2021-12-20:
Hi there,
I want to share my observations with the Ender 3 v2 and 500mW laser.
I also had a problem with slow PWM. So I did all the tweaks suggested by @zybrek and @vojto1980.
Awesome, it worked! Nice and precise engraving with the possibility to control the laser power.

BUT, then I started to burn 15 min program. After ca. 10 mins, something happened and the output from the connector now gives 100% power (24V) all the time. It is no more possible to control this output. I also tried stock firmware, but it still gives 100%.

For me, it looks, something may be burned in the mainboard. For that "fatal" engraving I had used about 5% of laser power and the mainboard fan (which was connected to the same connector) was not running at all, so maybe it got overheated, I do not know. Fortunately, the other features are OK, I can 3D print as before, only the fan rotates at full speed now.

Take care,
Michal

zbyrek, 2021-12-20:
> Hi there, I want to share my observations with the Ender 3 v2 and 500mW laser. I also had a problem with slow PWM. So I did all the tweaks suggested by @zybrek and @vojto1980. Awesome, it worked! Thank you. Nice and precise engraving with the possibility to control the laser power.
> 
> BUT, then I started to burn 15 min program. After ca. 10 mins, something happened and the output from the connector now gives 100% power (24V) all the time. It is no more possible to control this output. I also tried stock firmware, but it still gives 100%.
> 
> For me, it looks, something may be burned in the mainboard. For that "fatal" engraving I had used about 5% of laser power and the mainboard fan (which was connected to the same connector) was not running at all, so maybe it got overheated, I do not know. Fortunately, the other features are OK, I can 3D print as before, only the fan still rotates at full speed now.
> 
> Take care, Michal

You probably burned transistor, higher frequency switching create bigger losses due to capacitance of transistor which leads to more power dissipated, so it could overheat, only way to check this is to desolder transistor and measure it outside circuit.

[17 more replies]