MarlinFirmware/Marlin #20641: Speed is multiplied by 2 whenever top/bottom layers are printed

Stormwing01, 2021-01-03:
### Bug Description

My printer multiplies it's print speed whenever top/bottom layers are printed. if I set 1 top/bottom layer it only speeds up 1 layer. if I set it to 4, it does it for 4.

thing is, I have set my top/bottom layer speeds to be exactly the same as my wall speeds, but you can clearly see it speeding up and then slowing down once the layers are finished.

the printer itself shows speed as 100 percent the whole time.

i have tested with multiple versions of cura and the problem persists.

flow values also seems to be affected a bit.

### Configuration Files

[firmware.zip](https://github.com/MarlinFirmware/Marlin/files/5761040/firmware.zip)

### Steps to Reproduce

Sadly, I have no ideas what the steps are, because it just suddenly happened when I flashed marlin on my skr mini v2 motherboard. my old creality 1.1.5 motherboard does not suffer the issue.

**Expected behavior:**

Speeds should match what the slicer gcode dictates.

**Actual behavior:**

top/bottom layers are printed at twice the set speed and with more flow rate than is set in my slicer.

#### Additional Information

Please let me know what information else is needed, I will try to provide the information!

I will try to upload a video in an hour.

I use a 1.2mm nozzle.

ellensp, 2021-01-03:
A firmware.bin file is not your Configuration files.

Stormwing01, 2021-01-03:
[image]

https://youtu.be/uT_saYAjqWE

I apologize for my crusty english.

As you can see in the video and screenshot, my top and bottom layer speeds are 25mm/s, the same as all my other speeds. Marlin obviously ignores those speeds and the printer prints the top and bottom layers both at something like 50 or 60 mm/s.

it can be counteracted by decreasing my flow rate and speed by 50 percent via post processing in cura, but I would have to edit post processing for each different nozzle and profile I use, which is unpractical, but it does prove the point that speed and flow is 2 times faster.

Stormwing01, 2021-01-03:
> 
> 
> A firmware.bin file is not your Configuration files.

Yes there is. the file is inside the zip file. Not sure how you can miss it.

[image]

[image]

ellensp, 2021-01-03:
You need to attach Configuration.h and Configuration_adv. and any other files you modified. Not the firmware.bin file

Stormwing01, 2021-01-03:
Edit: when I make the wall could in cura more than 1 wall, the speed increase goes to the entire print, not just the top and bottom layers.

I still don't think it is the slicer, because I have tried multiple versions of the slicer and it also worked perfectly fine with my old motherboard.

> 
> 
> You need to attach Configuration.h and Configuration_adv.h and any other files you modified. Not the firmware.bin file

ooohhhh... I'll provide those files now, sorry

Stormwing01, 2021-01-03:
[firmware.zip](https://github.com/MarlinFirmware/Marlin/files/5761097/firmware.zip)

this one has the configuration and configuration_adv files

Stormwing01, 2021-01-03:
EDIT!!!!

I apologize, I was a bloody idiot!  This "problem" was just a setting (minimum layer time) doing it's job in cura!

I really should not try to figure these things out at 3 am at night without a bunch of coffee.

again sorry! my fault.

closed 2021-01-03 by Stormwing01