MarlinFirmware/Marlin #20642: [FR] Change HW PWM frequency via gcode

zbyrek, 2021-01-03:
### Description

It should be nice to be able to change frequency of HW PWM "on the fly" for example via gcode or menu item and not hardcode it in fw.

### Feature Workflow

1. Send gcode with pwm frequency
2. MCU changes PWM prescaler

#### Additional Information

Reasoning: if you use laser attachment for ender series, which is connected to fan1 output you need to have high pwm frequency for greyscale engraving but when you connect fan for 3d printing PWM frequency should be low to provide high torque, so if you hardcode it in fw you have to make some compromise (or have 2 fw versions and flash it when you want to print or engrave which is not great option).

Laser/spindle feature cannot be used because it won't allow use same pin for fan and laser.

luchomasco, 2021-06-13:
Hi @zbyrek , is there any update on this feature? I'm trying to use a Laser on my Ender3v2 with v4.2.2 motherboard, but I need at least 1kHz on FAN1 PWM.
Thanks for any all the help you can give me.
Greatings, Luciano.

cbagwell, 2022-06-30:
@luchomasco This FR would only help for case HW PWM is enabled and V4.2.2 does not have enabled by default.  Once enabled, it will default to 1kHz and so you still might not need this FR.  If you recompile, I also suggest thinking about rewiring your controller fan to be permanently on like the hotend fan is.  On V4.2.2 boards, the PWM controls both the laser *and* the controller FAN.  If you do lots of low power burns then the mosfet and stepper drivers could overheat.

The instructions to enable HW PWM are a bit out dated in #20638 so here are updates for Creality V4.2.2 and similar boards:

- Comment out FAN_SOFT_PWM in Configuration.h
- Change FAN_MIN_PWM to 0 in Configuration_adv.h
- Uncomment FAST_PWM_FAN in Configuration_adv.h
- Change MF_TIMER_STEP from 5 to 4 in Marlin/src/HAL/STM32F1/timers.h
- Change PA_0 from TIM2 to TIM5 in buildroot/share/PlatformIO/variants/MARLIN_F103Rx/PeripheralPins.c

You can optionally uncomment and modify FAST_PWM_FAN_FREQUENCY in Configuration_adv.h if you need value different than 1kHz.

This FR example of sending custom frequency in Gcode is one way to deal with issue (and a good option to have for testing) but I'm thinking about hacking up a simpler Laser Mode gcode and UI toggle command that switchings between two fixed frequencies.  I'm already comfortable with that approach on my GRBL machine when switching between spindle and laser.

cbagwell, 2022-07-21:
Having played with changing HW PWM a bit now, I think this is probably 2 feature requests.

One is for people hooking their lasers to parts cooler fan and wanting to control the PWM for this fan interface.  For this case, your already using M106 to control laser and so naturally think of it in terms of parts cooler fan settings.

RepRap M106 has already defined an Fxxx option on M106 for setting the FAN PWM so this would be a good option if this feature is accepted.

Having a menu option under temperature menu to set this frequency value in addition to fan speed would be a nice to have option but that suffers from usability in next use case.

2nd use case is for lucky people that can enable LASER_FEATURE.  Among other things, this enables a nice Laser menu option with a pre-existing option to set the PWM frequency with a range between 2000-40000.  Works great but its a bit of time commitment to scroll from 2000 up to 10000.

Having a gcode to set for this LASER_FEATURE case would also be useful but M106 doesn't make sense in that case.  I looked around for pre-existing gcode  examples that might apply here but I couldn't find any.  I could only find some examples in a subset of GRBL controllers where you can set $33 to laser PWM frequency you want to use or $28 in a 0-15 range with fixed frequency mappings.  So we'd need to invent/agree upon a new gcode to implement this part of feature request.

I-wish-2, 2023-02-01:
Greeting,
In an effort to increase my PWM frequency on my Ender 4.2.7 board I followed your instructions for editing the various configuration and other files from above.  The initial re-compile resulted in the PWM increasing from 7.87 Hz to 50Hz.
I would like to increase it again to about 1kHz per LightBurn's suggestion for a finer resolution for the Sovol 5W laser.

I did uncomment FAST_PWM_FAN_FREQUENCY in the Configuration_adv.h file but any values that I entered, from the notes above that line in the file, had no effect in changing the frequency.
I was hoping you might have some additional editing thoughts.
Regards
Doc'

cbagwell, 2023-02-02:
It looks like I left out an important change in my description.  Your report of 50Hz made it easy to find though since it hints at a servo conflict.   The change of PA0 from TIM2 to TIM5 puts it in conflict with servo also using TIM5 so we need to move that one to another free timer as well.

This is done in init/stm32f1.ini for v422 and v427 boards:

```
[STM32F103Rx_creality]
extends                     = stm32_variant
board_build.variant         = MARLIN_F103Rx
board_build.offset          = 0x7000
board_upload.offset_address = 0x08007000
build_flags                 = ${stm32_variant.build_flags}
                               -DMCU_STM32F103RE -DHAL_SD_MODULE_ENABLED
-                              -DSS_TIMER=4 -DTIMER_SERVO=TIM5
+                              -DSS_TIMER=4 -DTIMER_SERVO=TIM8
                               -DENABLE_HWSERIAL3 -DTRANSFER_CLOCK_DIV=8
```

I-wish-2, 2023-02-02:
Greetings CB,
Thank you for the information.  Being new to firmware editing process is struggle a little with out some instructions as you did for us by updating the 20638 thread. 
I'm going to see if I can advance my editing skills by looking at the information that you sent and try to figure out how to make the changes.  If you have an additional moment a step by step set of instructions word work better for me.  I will reply back with the results of my efforts.
I have included a picture of where I'm picking off the PWM signal from the gate of the Q1 fan mosfet.  Thank you again.
Doc'
[image]

I-wish-2, 2023-02-02:
I have found the init/stm32f1.ini in the extra_configs listing of the platformio file in the main source code.  But I'm at a bit of a loss as to how to edit it as I'm not finding the " init/stm32f1.ini " file to edit.  Still looking
Thanks
Doc'

cbagwell, 2023-02-02:
I am sorry to have made it harder than it should be since I gave wrong filename.  I meant the file stm32f1.ini in the "ini" directory at top level of Marlin git repo.

You can see all the changes I made related to fast pwm as well as my config file for my V4.2.2 board in the top commit of this branch:  

https://github.com/cbagwell/Marlin/commits/bugfix-2.1.x-CrealityV422

That branch also adds a new gcode so you can test changing PWM rate at run time.  I never got around to moving the logic to M106 to match RepRap as I described earlier in this FR.

I-wish-2, 2023-02-02:
Its all good.  Please look over what I just edited and see if it's close.  One note in line 131 I have used the STMF103RC as my processor is the  STMF103RCT6 version.  Is that the correct edit or should it be the STMF103RE ?
Here is my current edit of the { stm32f1.ini

Origional stm32f1.ini

Line 124 	[STM32F103Rx_creality]
Line 125	extends                     = stm32_variant
Line 126	board_build.variant         = MARLIN_F103Rx
Line 127	board_build.offset          = 0x7000
Line 128	board_upload.offset_address = 0x08007000
Line 129	build_flags                 = ${stm32_variant.build_flags}
Line 130			-DHAL_SD_MODULE_ENABLED
Line 131			-DSS_TIMER=4 -DTIMER_SERVO=TIM5
Line 132			-DENABLE_HWSERIAL3 -DTRANSFER_CLOCK_DIV=8
Line 133			-lproui -LMarlin/lib/proui
Line 134 build_unflags               = ${stm32_variant.build_unflags}
Line 135			-DUSBCON -DUSBD_USE_CDC
Line 136  extra_scripts               = ${stm32_variant.extra_scripts}
Line 137			pre:buildroot/share/PlatformIO/scripts/random-bin.py
Line 138			pre:buildroot/share/scripts/proui.py
Line 139  monitor_speed         = 115200
Line 140  debug_tool                = jlink
Line 141  upload_protocol        = jlink

Edited stm32f1.ini

Line 124 	[STM32F103Rx_creality]
Line 125	extends                     = stm32_variant
Line 126	board_build.variant         = MARLIN_F103Rx
Line 127	board_build.offset          = 0x7000
Line 128	board_upload.offset_address = 0x08007000
Line 129	build_flags                 = ${stm32_variant.build_flags}
Line 130  #  Start Edits Doc'
Line 131			DCMU_STMF103RC -DHAL_SD_MODULE_ENABLED  # My processor is a RC6T version Doc'
Line 132			-DSS_TIMER=4 -DTIMER_SERVO=TIM5
Line 133			-DSS_TIMER=4 -DTIMER_SERVO=TIM8
Line 134			-DENABLE_HWSERIAL3 -DTRANSFER_CLOCK_DIV=8
Line 135  #  End Edits Doc'
Line 136			-lproui -LMarlin/lib/proui
Line 137 build_unflags               = ${stm32_variant.build_unflags}
Line 138			-DUSBCON -DUSBD_USE_CDC
Line 139  extra_scripts               = ${stm32_variant.extra_scripts}
Line 140			pre:buildroot/share/PlatformIO/scripts/random-bin.py
Line 141			pre:buildroot/share/scripts/proui.py
Line 142  monitor_speed         = 115200
Line 143  debug_tool                = jlink
Line 144  upload_protocol        = jlink

Did I get it close
Doc'

I-wish-2, 2023-02-02:
I ran a build with both STMF103RC  &  STMF103RC in this line.  Line 131 DCMU_STMF103RC -DHAL_SD_MODULE_ENABLED # My processor is a RC6T version Doc'  It produced this result for both.

Build failed 

tforms\ststm32@12.1.1\builder\DMCU_STM32F103RE' not found, needed by target `.pio\build\STM32F103RE_creality\firmware-20230202-120430.elf'.

Kind of stuck now.  Ah, but getting closer

I-wish-2, 2023-02-02:
I changed the edit for the Original Line 130 and the Edited Line 131. To be -DHAL_SD_MODULE_ENABLED only.  I did not include either the STMF103RC or STMF103RE entries.
The compile completed and the resulting xxxxx.bin file flashed with the PWM frequency now at 1KHz.
Per your comment I did remove the Mother Board cooling fan from the PWM connection and wired it straight to the 24 vdc buss as it had a wicked 1KHz ring.  With all that completed I have started running the now Ender 3 Pro Laser with LightBurn for further testing.  It is all looking good although I'm going to take the Frequency up more to see the results in the raster burns.

Move info to come.  Thanks again for all your help.
Best Regards
Doug

I-wish-2, 2023-02-03:
You can optionally uncomment and modify FAST_PWM_FAN_FREQUENCY in Configuration_adv.h if you need value different than 1kHz.

In an effort to edit the above line to allow the frequency to be increased, what it the correct way to edit these lines?

Configuration_adv.h

Origional Lines 613 to 624  FAST_PWM_FAN_FREQUENCY

Lines
613	#define FAST_PWM_FAN    // Increase the fan PWM frequency. Removes the PWM noise but increases heating in the FET/Arduino
614	#if ENABLED(FAST_PWM_FAN)
615	//#define FAST_PWM_FAN_FREQUENCY 31400  // Define here to override the defaults below
616	//#define USE_OCR2A_AS_TOP
617	#ifndef FAST_PWM_FAN_FREQUENCY
618	#ifdef __AVR__
619	#define FAST_PWM_FAN_FREQUENCY ((F_CPU) / (2 * 255 * 1))
620	#else
621	#define FAST_PWM_FAN_FREQUENCY 1000U
622	#endif
623	#endif
624	#endif

I have not been able to successfully edit them and the compile.  Help if you have a moment.  Thank You.
Regards
Doug

rajhlinux, 2023-08-13:
> Greeting, In an effort to increase my PWM frequency on my Ender 4.2.7 board I followed your instructions for editing the various configuration and other files from above. The initial re-compile [...]

How are you able to determine that the laser frequency was changed?

rajhlinux, 2023-08-13:
> -DSS_TIMER=4 -DTIMER_SERVO=TIM5
> +                              -DSS_TIMER=4 -DTIMER_SERVO=TIM8

What exactly needs to be done to use a generic laser module and able to modulate with any frequency?

rajhlinux, 2023-08-13:
> . If you do lots of low power burns then the mosfet and stepper drivers could overheat.

Is it possible to simply use the PWM solely as a data signal and not draw power out from the 4.2.7 board and have the laser module to have it's own dedicated power source? This way 20+ Watt laser modules could be used without drawing power directly off of the 4.2.7 board. 

The board would only send voltage data signal to the laser module.

rajhlinux, 2023-08-14:
> ### Description
> It should be nice to be able to change frequency of HW PWM "on the fly" for example via gcode or menu item and not hardcode it in fw.
>
> ### Feature Workflow
> 1. Send gcode with pwm [...]

How are you able to do this? Can you make a tutorial? Thanks.