MarlinFirmware/Marlin #20643: [BUG] GTR V1 unable to build - "Unknown Development Platform... ststm32"

Protean-Man, 2021-01-03:
Just started a fresh BugFix build with the GTR V1 board.
Unable to build with the very base setup. 

In VS Code I am getting this error:
```
Processing BIGTREE_GTR_V1_0 (platform: ststm32@~10.0; board: BigTree_GTR_v1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------Platform Manager: ststm32 @ 10.0.1 is already installed
Error: Unknown development platform 'PackageItem <path=C:\Users\adwar\.platformio\platforms\ststm32 metadata=PackageMetaData <type=platform name=ststm32 version=10.0.1 spec={'owner': 'platformio', 'id': 8020, 'name': 'ststm32', 'requirements': None, 'url': None}'
The terminal process "C:\Users\adwar\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.
```

Configs:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5761101/Configuration_adv.zip)

Thanks!

ellensp, 2021-01-03:
most likely your platformio is out of date.
Click on the alien/bug head 
Look in QUICK ACCESS list
Select update all.

Protean-Man, 2021-01-03:
Hi @ellensp,

I tried that and the same issue arrises. This is the result of platformIO update:

```
PS C:\Users\adwar\OneDrive\Documents\GitHub\Marlin> pio update
Updating platformio/contrib-piohome           3.3.1 @ ~3.3.1                     [Up-to-date]
Updating platformio/contrib-pysite            2.37.191020 @ ~2.37.0              [Incompatible 2.39.201019]

Platform Manager
================
Platform atmelavr
--------
Updating platformio/atmelavr                  3.1.0                              [Up-to-date]
Updating platformio/toolchain-atmelavr        1.50400.190710 @ ~1.50400.0        [Up-to-date]
Error: Unknown development platform 'PackageItem <path=C:\Users\adwar\.platformio\platforms\ststm32 metadata=PackageMetaData <type=platform name=ststm32 version=10.0.1 spec={'owner': 'platformio', 'id': 8020, 'name': 'ststm32', 'requirements': None, 'url': None}'
```

Protean-Man, 2021-01-03:
Ok I seemed to have fixed it by:

1 - Deleting C:\Users\USER\.platformio\platforms\ststm32
2 - In VS Code, PlatformIO, CLICK PlatformIO Core CLI from the quick access list
3 - type "pio upgrade --dev"
4 - restart VS Code
5 - CLICK Upgrade All in PlatformIO - which still says: `Updating platformio/contrib-pysite 2.37.191020 @ ~2.37.0 [Incompatible 2.39.201019]` and I can't figure out why.
6 - Restart VS Code again. 

Even though the `Incompatible `error is thrown when updating it allowed me to build again.

Thanks

ellensp, 2021-01-03:
Since its not a Marlin issue, closing.

closed 2021-01-03 by ellensp

GadgetAngel, 2021-01-03:
I been having the same problems with VScode, I did what you did: erased my .Platformio sub directory and then restarted VScode to let it download python. 

I then tried to recompile and got missing path message.   So I just hit clear project. Then recompile again and it then worked.  

Try recompile twice even tho you get the error the first time. 

I also see this problem when switching environments in Platformio.ini file (switching between different platforms type boards) like going from GTR V1.0 to SKR E3 Turbo board. 

Just placing this here for others who are searching for fixes to this issue.