MarlinFirmware/Marlin #20645: [BUG] Knocking noise when moving TMC2209 (reopen #14183)

AndKe, 2021-01-03:
### Before Reporting a Bug

2.0.7.2  still has this bug: https://github.com/MarlinFirmware/Marlin/issues/14183
The fix does not seem to apply to SKR mainboards.

### Instructions

Config attached, while TMC2209 is used on SKR(1.3) - (32bit board)   if MONITOR_DRIVER_STATUS is defined, knocking is heard whenever driver is enabled, louder when moving.

### Bug Description

Same as linked

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5761356/Configuration.zip)
All changes marked with //#####

### Steps to Reproduce
see above.

ellensp, 2021-01-03:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

sjasonsmith, 2021-01-03:
> The fix does not seem to apply to SKR mainboards

The "fix" I made for AVR boards was to add a sanity check to prevent enabling `MONITOR_DRIVER_STATUS` if using SoftwareSerial to talk to TMC UARTs.

Are you suggesting that we should extend that sanity check to apply to all boards?

sjasonsmith, 2021-01-03:
Although it doesn't make sense to me that it would cause knocking when you aren't even moving...

AndKe, 2021-01-03:
Just to clearify;
The "knocking" interval is a bit over 1Hz (I did not measure) - the noise amplitude is proportional to speed or maybe torque, while the interval between "knocks" is always the same, not affected by speed or which axis is moved.

If I move steppers a bit, then they are idle, (holding current) there there is still very low knocking, and it can be felt on the main axle of the steppers.  Once I disable steppers, it gets silent.
Commenting out MONITOR_DRIVER_STATUS solved it instantly (supersmooth sound on all axis.)

I do not know whatever the SKR1.3 uses software serial at all - so I have no idea whatever the same fix is "correct" for this board.

ellensp, 2021-01-03:
Yes the SKR 1.3 uses software serial, but the skr is a 100Mhz 32 bit lpc1768 processor vs a 16 Mhz 8 bit AVR processor. It shouldn't have any issues with the work load of MONITOR_DRIVER_STATUS

grauerfuchs, 2021-01-03:
Can we check to see what the temperature is on the TMC chips and on the motors? I had this happen at first when I was driving the chips so hard they were going into thermal shutdown. Even with heatsinks in place, the chips would shut down regularly until the temperature dropped. I should note that this was happening with the **default vref settings** on some of the drivers. This also happened when current was being driven improperly via digipot on initial testing of an FFCP with TMC2208 upgrades in place. I suspect that if the current is managed in software and is set too high, it could result in the very same.

AndKe, 2021-01-03:
FWIW: I checked my new setup with thermal camera (to detect bad connection of heater-cables) and could not see anything obvious, I have three high cooling fins on the TMC2209's PCB, and a fan blowing on them.  All of my tests were related to a new setup (test runs then did not exceed 30cm of travel per test per axis) , and now that the MONITOR_DRIVER_STATUS is commented out, I have printed a 3-hr test print of ABS, without many hiccups.
So in my case - this is purely related to a setting - not heat.

grauerfuchs, 2021-01-03:
Ok, that definitely answers that then. Thank you. It does sound like the MONITOR_DRIVER_STATUS may be an issue regardless. Can someone with detailed knowledge of the Marlin libs for software serial confirm whether or not a transmission or receive operation is a blocking operation? I know many AVR-based implementations are. If this is the case, then it won't matter which chip is used for a CPU; processing power would be entirely wasted in wait states until completion of the read/write operations.

AndKe, 2021-01-03:
Despite the plentiful processing power, if something is not properly multithreaded or the software serial is a blocking function, it may very well stop anything else for the time of communication (which would be the same for the same data at a given baudrate)
AFAIK Marlin does not use ChibiOS or NuttX or other RTOS

sjasonsmith, 2021-01-03:
Since the knocking occurs when the motors **are not moving** it wouldn't matter whether they are blocking or not. Marlin is not changing the control lines to the TMC driver when it isn't trying to move motors.

ellensp, 2021-01-03:
One question I have is what type of TMC2209 stepper modules, who makes them? Perhaps they are some sort of broken clone?

teemuatlut, 2021-01-04:
With AVR the issue was that the SoftwareSerial library disables interrupts for a brief moment and this also prevents timer interrupts (like Step pulse) from firing. The result was inconsistent step rates and audible knocking noise.
As far as I know the LPC implementation of the SoftwareSerial does not disable interrupts and should not exhibit the same issue.

The driver won't try to move the motor without step pulses. We'll see if someone can confirm the issue and then look into the messages sent to the driver or if perhaps the enable line is toggled for some reason. One indication would also be a changing mscnt value on each knock.

AndKe, 2021-01-04:
The next step would be to eliminate the chance for something odd in my setup:
My FYSETC 3.0 modules are jumped on RX and TX pins, (please note, that I am very surprised to see RX/TX be on different position in version 3.0 and 3.1)

I am using a diagonal jumper like in this article, https://www.jayuk.org/bigtreetech-skr-v1-4-normal-and-turbo-and-fysetc-tmc2209-v3/

Maybe, just maybe, the fact that RX/TX is in different places, make the jumper RX/TX conduct some other signal into incorrect place?

AndKe, 2021-01-04:
I removed teh "RX/TX" humper on top of the TMC2209 modules, that caused no difference 

I ran M122 with MONITOR_DRIVER_STATUS uncommented.  then again after a while.  I see no ghostly steps
https://gist.github.com/AndKe/256330f53c42175a204311bc031f1f8e   
(This firmware, with M122, makes the noise again.) - recompiled after commenting it out, and it's good again. 
A simple homing is a good test, as knocking will be heard along all three axes, instead of a perfect smooth hum.

The Stepper documentation is here: https://wiki.fysetc.com/Silent2209/
And I am using this version 3.0 with the silk-screen error described here: 
[image]

Maybe the diagonal jumper is the problem?  (TMC connection error happens without the diagonal jumpers,  so I doubt it)

ConstantijnCrijnen, 2021-01-13:
I think I have a related problem which I've been trying to hunt down for over a year. I never mentioned it here because I always thought that it was some sort of mechanical issue, but I am now almost certain that it's not. 
Configuration files: [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5806505/Configuration.zip)
The version I'm using is the bugfix from around a week ago. Unfortunately I can't just upload and experiment with new bugfixes because the printers are running at my university and it's nearing the end of the semester so the printers have to printing non-stop.

The drivers I'm using are BTT2209 with a BTT SKR1.3

The issue is that the drivers will randomly give a loud knock and and skip a couple of steps (around 0.5 to 2mm with GT2 20T pulleys and 1.8deg motors). It doesn't happen often and its occurrence seems to correlate with the feedrate.

At 30mm/s max. speed the issue will occur around once per week with 24/7 printing. At 60mm/s about every two days, at 100mm/s 1-2 times per day and at 150mm/s every 1-2 hours. 

Since it happens so randomly, it is really hard for me to catch it but I just managed to catch it happen twice in a row (marked with orange arrow)
[image]
[image]

One thing I changed in trinamic.cpp was under
`
#if HAS_DRIVER(TMC2209)
`
the following blank time from 1 to 3 since I'm running quite high-inductance motors (6mH per phase)
`
chopconf.tbl = 0b11; // blank_time = 40
`

That seemed to have helped dramatically, since the issue was much worse where the printer would lose more steps and more often. However, I think that that issue and my fix are not related to the issue I'm having now, as it would have been easy to mistake one sort of step loss with the other. I am however quite sure, that it's not due to torque loss at higher speeds because only with a stupid amount of brute force am I able to block the axis' movement, even at 400mm/s.

I wish I could try messing with MONITOR_DRIVER_STATUS but, as I said, I can't mess with the printers for another month, which makes my observations quite useless, but maybe someone will be able to use my observations as a part of the puzzle?

ConstantijnCrijnen, 2021-01-16:
I spoke too soon! I just opened [issue 20785](https://github.com/MarlinFirmware/Marlin/issues/20785), as my testing concluded that the problems I'm having are not coming from `MONITOR_DRIVER_STATUS`.

closed 2021-02-23 by github-actions[bot]