MarlinFirmware/Marlin #20646: [BUG] CR10_STOCKDISPLAY with SKR E3 TURBO board gets a continuous SPEAKER output between POWER ON of printer and Marlin Boot Screen appearing on LCD

GadgetAngel, 2021-01-03:
### Bug Description

Hardware:
CR10_STOCKDISPLAY off of an Ender3 PRO
SKR E3 TURBO MCU board
Connect EXP3 on back of CR10_STOCKDISPLAY connected to EXP1 on the SKR E3 TURBO board
Turn the power on the SKR E3 TURBO and speaker produces a high pitch sound until the MARLIN boot screen appears. Once the Marlin boot screen appears the high pitch sound goes away.

The sound starts immediately after the firmware upload occurs.

I tried turning the cables around but then I lost the screen. 

I tried disabling the #define SPEAKER in configuration.h but the sound still was produced.  

I tried going into the pins_BTT_E3_TURBO.h file and disabling the BEEPER_PIN but it made it worse and it continuously produced a sound until the power was shut off on the printer. 

 I used this LCD adapter board from TH3D, https://www.th3dstudio.com/product/ezboard-creality-dual-lcd-adapter-board/  
[image]
 with the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER but only enabled CR10_STOCKDISPLAY in configuration.h.  The adapter board just combines the EXP1 and EXP2 into EXP3 port (I made no changes to the firmware) and the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER came up and showed the Marlin display WITHOUT the noise!  

Something is causing the speaker to start producing sound on the CR10_STOCKDISPLAY on the SKR E3 TURBO.  

Since the LCD adapter board cover up other connectors I need on the SKR E3 TURBO, I was only left with one option.  I cut the BEEPER line on the flat ribbon cable that goes to my CR10_STOCKDISPLAY.  Once the BEEPER line was cut and the cable put back on the screen and on the SKR E3 TURBO board, I rebooted and the speaker no longer produced the sound.  

### Configuration Files

[SKR E3 TUBO and CR10_STOCKDISPLAY.zip](https://github.com/MarlinFirmware/Marlin/files/5761471/SKR.E3.TUBO.and.CR10_STOCKDISPLAY.zip)

### Steps to Reproduce

Hardware:
CR10_STOCKDISPLAY off of an Ender3 PRO
SKR E3 TURBO MCU board
Connect EXP3 on back of CR10_STOCKDISPLAY to EXP1 on the SKR E3 TURBO board
Turn the power on the SKR E3 TURBO and speaker produces a high pitch sound until the MARLIN boot screen appears. Once the Marlin boot screen appears the high pitch sound goes away.

**Expected behavior:**

No high pitch sound to occur unless there is a issue with Marlin

**Actual behavior:**

High pitch sound occurs and their is not a problem with Marlin

ellensp, 2021-01-03:
This is not something that marlin can fix.
The issue is the boot loader needs to set the buzzer pin off.
You need a custom bootloader... 

If you use a Pin 27 board, this will disable the buzzer.
[image]

closed 2021-01-03 by ellensp

GadgetAngel, 2021-01-03:
Thank you, I will get a pin27 board

GadgetAngel, 2021-01-08:
I solved my issue by creating a custom bootloader for the board by modified the smoothieware bootloader. If anyone else want to solve this problem you can be the modified bootload.hex file and instructions on how to upload the modified bootloader from here: https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader