MarlinFirmware/Marlin #20649: [BUG] LIN_ADVANCE produces horrible E jerking on first layer (depending on object)

AndKe, 2021-01-03:
### Bug Description
Some files, produce violent E travel, causing it to skip steps, making awful sound, and not print correctly. (underextrude)
in this case: 0.5mm thin wall test file is affected badly.

### Configuration Files
gcode: 
[Uploading 0.5mm-thin-wall.zip…]()
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5761637/Configuration.zip)

### Steps to Reproduce
Print Gcode as-is , 
M900  K0.6 is set, so LIN_ADVANCE will be active, and produce horrific result on the first layer.
then change gcode to
M900  K0
observe normal print of all layers.

#### Additional Information
-

swilkens, 2021-01-03:
There are some known issues with LA specifically in combination with some SKR boards.

Is the sound only occurring on the first layer, or on all layers?

In your `Configuration_adv.h` file, please look for `//#define LA_DEBUG` and enable it, report back here if you see any significant debug output over the serial interface.

AndKe, 2021-01-03:
I have only really observed it on the first layer.   (I have no explanation for that)
Also: I am fairly sure I've heard this before on the Atmega2560 board I upgraded from. Last time I thought it was the hobbed gear that skipped on the filament.  (I almost never see a print job start over than via octopi , this is detected because of the fact that I replaced the 2560 board with SKR1.3 and were nearby for all basic tests.)

XDA-Bam, 2021-01-07:
@AndKe I see your `DEFAULT_EJERK` is set to `5.0`. In my experience, this is way too low for LA to work correctly. Maybe try it with `15.0`. You could also try using `CLASSIC_JERK` in stead of JD. The combination of LA and JD can cause some odd issues.

It is strange that it only happens on the first layer. But I would test these two changes first, before investigating any further.

AndKe, 2021-01-07:
@XDA-Bam  Thanks - I did not tweak DEFAULT_EJERK - maybe the default should be very different then?

XDA-Bam, 2021-01-30:
As a follow-up: The `DEFAULT_EJERK` is fine, because `LIN_ADVANCE` is not on by default.

Did you try any of the suggested settings? Did they improve anything?

AndKe, 2021-01-31:
No,  have not tried a different DEFAULT_EJERK as it's unclear what would be a decent setting.
I did not try CLASSIC_JERK eighter.

From what I see in this thread, the default config files should have a clear warning against LIN_ADVANCE & DEFAULT_EJERK (if it is known to cause trouble) and DEFAULT_EJERK should have a good default vealue.

XDA-Bam, 2021-02-02:
> 
> No, have not tried a different DEFAULT_EJERK as it's unclear what would be a decent setting.

See above: "Maybe try it with 15.0". I ran up to 20.0 on my bowden setup (standard Creality extruder), but started getting some skipped steps there, so I turned it down again.

>
> I did not try CLASSIC_JERK eighter.

Well...

> 
> From what I see in this thread, the default config files should have a clear warning against LIN_ADVANCE & DEFAULT_EJERK (if it is known to cause trouble) and DEFAULT_EJERK should have a good default vealue.

I agree that some comment/warning in the `LIN_ADVANCE` section of the config would be appropriate. But as stated above, the defaults for `DEFAULT_EJERK` are fine, because the default for `LIN_ADVANCE` is, that it's deactivated.

AndKe, 2021-03-05:
Please implement XDA-Bam 's conclusion.

thisiskeithb, 2021-03-05:
We're accepting [Pull Requests](https://marlinfw.org/docs/development/getting_started_pull_requests.html)!

Vertabreak, 2021-03-05:
would be happy to submit a comment change what exactly should it say.
something to the effect of linear advance may require adjustment of default_ejerk? 
perhaps a if condition that switch's the value when linear advance is enabled?
come up with the most ideal way to handle it and ill put in the request should others be unable to them self.

XDA-Bam, 2021-03-05:
I'd suggest a warning in the style of
```c++
#if ENABLED(LIN_ADVANCE) && DEFAULT_EJERK<10
    // Warn user about possible problems
#endif
```

Vertabreak, 2021-03-05:
sounds like more of sanity check something like the following.

`#if ENABLED(LIN_ADVANCE) && DEFAULT_EJERK<10
    
#error DEFAULT_EJERK less then 10 results in issues when LIN_ADVANCE is enabled.

#endif`

is this the consensus and is is universally true,

thisiskeithb, 2021-03-05:
> is this the consensus and is it universally true,

It's not universally true, so I'd change it to a #WARNING instead of an #ERROR and word it suggesting that users *may* experience issues. There will probably be configs from the Configurations repo with settings that will cause builds to fail with that #ERROR as well.

Vertabreak, 2021-03-05:
so we are leaning more to something like.

```
#if ENABLED(LIN_ADVANCE) && DEFAULT_EJERK<10
#WARNING DEFAULT_EJERK less then 10 with LIN_ADVANCE enabled may cause issues.
#endif
```
are we thinking sanity check or should this be added to the lin_advance config section.
when we have a consensus on how it should be handled i can then submit it see what thinkyhead has to say about it.

ManuelMcLure, 2021-03-05:
From my reading of the code, when using Linear Advance with Junction Deviation `DEFAULT_EJERK` is completely ignored and the E jerk is calculated using:
```
 891     #if HAS_LINEAR_E_JERK
 892       FORCE_INLINE static void recalculate_max_e_jerk() {
 893         const float prop = junction_deviation_mm * SQRT(0.5) / (1.0f - SQRT(0.5));
 894         LOOP_L_N(i, EXTRUDERS)
 895           max_e_jerk[E_INDEX_N(i)] = SQRT(prop * settings.max_acceleration_mm_per_s2[E_INDEX_N(i)]);
 896       }
 897     #endif
 ```
So I can't see how changing `DEFAULT_EJERK` would have any effect on this case.

Vertabreak, 2021-03-06:
it seems to me further investigation and a better a understanding is required before a patch can be made give the info presented.

XDA-Bam, 2021-03-06:
> From my reading of the code, when using Linear Advance with Junction Deviation `DEFAULT_EJERK` is completely ignored and the E jerk is calculated using:
>
> ```
>  891     #if HAS_LINEAR_E_JERK [...]

Good catch. From what I can see after a somewhat extensive search for all terms related to E-jerk, that seems to be the only exception, though. It's triggered by
https://github.com/MarlinFirmware/Marlin/blob/2f97505b93f3550707cbaa05f6118de86670f57e/Marlin/src/inc/Conditionals_post.h#L40-L43

Therefore, something like 
```c++
#if ENABLED(LIN_ADVANCE) && DISABLED(HAS_JUNCTION_DEVIATION) && DEFAULT_EJERK<10
  #WARNING DEFAULT_EJERK less then 10 with LIN_ADVANCE enabled may cause issues.
#endif
```

should be a helpful warning. It just doesn't cover the case, where JD is enabled, LA is enabled and `settings.max_acceleration_mm_per_s2[E_INDEX_N(i)] * junction_deviation_mm` is too low. We could handle that, too. But in case 
https://github.com/MarlinFirmware/Marlin/blob/b26516b3ac7cd7d15a751af4b299a38ac8ac7a57/Marlin/src/module/planner.h#L887-L893
is rewritten in the future, the mirrored calculation for the warning would have to be adapted, too.

AndKe, 2021-06-06:
This should remain open until actually solved.

AndKe, 2021-08-07:
keep active.

thisiskeithb, 2021-08-07:
A sanity check has been added: https://github.com/MarlinFirmware/Marlin/commit/5dc5d42e25d6afe5a4ed616c768c91cc4db7db66.

closed 2021-08-07 by thisiskeithb