MarlinFirmware/Marlin #20651: [BUG] TEMP_CHAMBER_HYSTERESIS

Lanwolf81, 2021-01-03:
### Bug Description
reopen [https://github.com/MarlinFirmware/Marlin/issues/20432](url)

SKR 1.4 Turbo
Configuration_adv.h

#if TEMP_SENSOR_CHAMBER
#define CHAMBER_MINTEMP 5
#define CHAMBER_MAXTEMP 60
#define TEMP_CHAMBER_HYSTERESIS 5 // (°C) Temperature proximity considered "close enough" to the target
//#define CHAMBER_LIMIT_SWITCHING
#define HEATER_CHAMBER_PIN P1_00 // Chamber heater on/off pin
//#define HEATER_CHAMBER_INVERTING false

#define TEMP_CHAMBER_HYSTERESIS did not work.
Starting heating at 1°C under the set temp.
I use M502 to reset the eeprom after flashing, but nothing change

testet with 2.0.7.2 and bugfix-2.0.x from 02.01.2021

### Configuration Files

bugfix-2.0.x 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5761748/Marlin.zip)

### Steps to Reproduce

Change the #define TEMP_CHAMBER_HYSTERESIS over 1
Flash the new firmware
Test

**Expected behavior:**

Set the chamber temrature to 40°C. Marlin heating the chamber up to it. After the tempreatur is falling under 35°C start heating again.

**Actual behavior:**

chamber heater lamps flick on and off at 40 °C (39.9 - on, 40.1 - off).

#### Additional Information

reopen  @[https://github.com/MarlinFirmware/Marlin/issues/20432](url)

ellensp, 2021-01-03:
code needs #define CHAMBER_LIMIT_SWITCHING
or it doesn't use TEMP_CHAMBER_HYSTERESIS
Ie add #define CHAMBER_LIMIT_SWITCHING to your Configuration_adv.h

sjasonsmith, 2021-01-03:
I'm going to re-open this, because it seems a bit more discussion is warranted.

The configuration file in no way implies that `CHAMBER_LIMIT_SWITCHING` should be required.

In fact, it doesn't describe what `CHAMBER_LIMIT_SWITCHING` even is!

sjasonsmith, 2021-01-03:
Not to mention the config defaults to `TEMP_CHAMBER_HYSTERESIS` without `CHAMBER_LIMIT_SWITCHING`

sjasonsmith, 2021-01-03:
It looks like this is leveraging options with the same names as are used for the bed, but you have to look at the comments around `PIDTEMPBED` in Configuration.h to make sense of them.

I definitely think these deserve some rework to make correct usage obvious.

Lanwolf81, 2021-01-04:
ok i aktivate the`#define CHAMBER_LIMIT_SWITCHING` but nothing change.
#if TEMP_SENSOR_CHAMBER
  #define CHAMBER_MINTEMP             5
  #define CHAMBER_MAXTEMP            50
  #define TEMP_CHAMBER_HYSTERESIS     5   // (°C) Temperature proximity considered "close enough" to the target
  #define CHAMBER_LIMIT_SWITCHING
  #define HEATER_CHAMBER_PIN       P1_00  // Chamber heater on/off pin

thinkyhead, 2021-04-15:
Has anyone followed up on this or are we still waiting on the Good Lord?

thinkyhead, 2021-04-15:
Apparently `PIDTEMPCHAMBER` must be disabled and `CHAMBER_LIMIT_SWITCHING` must be enabled before you have `CHAMBER_HYSTERESIS`.

```cpp
#if DISABLED(PIDTEMPCHAMBER)
  #define CHAMBER_CHECK_INTERVAL 5000   // (ms) Interval between checks in bang-bang control
  #if ENABLED(CHAMBER_LIMIT_SWITCHING)
    #define CHAMBER_HYSTERESIS 2        // (°C) Only set the relevant heater state when ABS(T-target) > CHAMBER_HYSTERESIS
  #endif
#endif
```

closed 2021-06-29 by github-actions[bot]