MarlinFirmware/Marlin #20652: TOOLCHANGE_FS_LENGTH - Not retracting active tool before toolchange

ColorGS, 2021-01-03:
I'm attempting to bring my last two printers running 2.0.5.3 up to 2.0.7.2.  For this submission, I'll use the creality ender 5 plus.

Creality Ender 5 Plus
BTT SKR pro v1.1 with BTT TFT70
1 BTT THC-01 3 in 1 out switching hotend, 3 extruders
Cura Slicer for gcode creation
Printing by SD on TFT as well as ONBOARD for testing this problem with same results.

Expected behavior:
When using 2.0.5.3, this printer performs toolchange filament swap as expected.  I use a swap length of 50 set in firmware.  Using cura sliced gcode, the active tool in the print will retract the defined 50 from active tool before loading the next active tool by a length of 50.

Observed behavior:
Using the same gcode file from a successful print from Cura, 2.0.7.2 firmware does not retract the defined 50 from the active tool before loading the next active tool by a length of 50,  thereby jamming the filament inside the hotend chamber.

Steps to reproduce:
Easiest method is to set active tool using the GUI terminal on BTT TFT 70.  Enter T0, Tool 0 will become active.  Enter T1, Extruder for T0 will perform NO retraction, then extruder for T1 loads 50.  Enter T2.  T1 extruder will NOT retract any.  T2 extruder will load 50.  Enter T0.  Extruder for T2 will not retract any.  Extruder for T0 will load 50.  Firmware 2.0.5.3 will perform as expected however.

Attempted resolution:
I've tried enabling FW Retraction, which I didn't use in 2.0.5.3, which did not change the behavior.  It has a define swap length item which I defined as 50 to attempt to solve this issue.  It did not.

Work Around:
I have setup in Cura a start and end gcode for each extruder in machine settings to work around this issue.  This doesn't help when doing troubleshooting or maintenence on printer itself.  Example.  Loading filaments.  When switching active tool to load an additional filament, I must remember to manually retract the first loaded filament by 50 before switching extruders to load the next filament into the machine.

[E5+Marlin 2.0.7.2.zip](https://github.com/MarlinFirmware/Marlin/files/5761820/E5%2BMarlin.2.0.7.2.zip)
[E5+Marlin 2.0.5.3.zip](https://github.com/MarlinFirmware/Marlin/files/5761821/E5%2BMarlin.2.0.5.3.zip)

sjasonsmith, 2021-01-17:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thinkyhead, 2021-02-17:
After a quick review of the code, it appears that `TOOLCHANGE_FS_PRIME_FIRST_USED` is required or else `tool_change_prime` is never called. I'm not sure if that is the behavior that everyone wants, but give it a try.

soundstorm, 2021-04-15:
The mentioned Param `V` in `Configuration_adv.h` for `M217` is not documented as of today https://marlinfw.org/docs/gcode/M217.html

Using `TOOLCHANGE_FS_PRIME_FIRST_USED` without `TOOLCHANGE_FS_INIT_BEFORE_SWAP` ends up in an error that `toolchange_extruder_ready` is not defined as https://github.com/MarlinFirmware/Marlin/blob/e3a12c3c28c2b6c4882eaa6549c6053b4d38953e/Marlin/src/module/tool_change.cpp#L48-L50 only lists `TOOLCHANGE_FS_INIT_BEFORE_SWAP` as dependency. Should there be a sanity check or something else?

closed 2021-06-26 by github-actions[bot]