MarlinFirmware/Marlin #20653: [BUG] #define PID_OPENLOOP causes hotend to overheat

willdmo, 2021-01-03:
### Bug Description

if I activate in the Configuration.h on an BTT skr 1.3 or 1.4 the point #define PID_OPENLOOP without any number after the command as soon as the printer receives any command to heat up the hotend to a certain temperature, the hotend will heat until the thermal runaway protection switches off the printer. It wont stop on the set temperature.
### Configuration Files

[link to my config and config adv](https://www.sharepa.com/s/DbSQaIciM7)

### Steps to Reproduce

1. activate define PID_OPENLOOP in configuration.h
2. compile and flash firmware
3. tell printer to heat up hotend to a certain temperature

**Expected behavior:**

Printer heats up hotend to set temperature
**Actual behavior:**
Printer heats up hotend until thermal runaway stops it

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-01-03:
No, your miss understand
PID_OPENLOOP means its just heats! no control.

closed 2021-01-03 by ellensp

willdmo, 2021-01-03:
Rtfm... I see. But even on an open loop response a pid tuning according
Kirchner should be possible. I think a documenting comment would be handy

Mfg Dominik Samol

Von einem Handheld gesendet. Ich bitte Schreibfehler zu verzeihen.