MarlinFirmware/Marlin #20661: PID Autotune does not work with SKR 1.4 Turbo

BenKitesurfs, 2021-01-03:
### Bug Description

I could not get my hotend to hold temp since installing an SKR 1.4 Turbo board a few weeks ago.  I have a genuine E3D V6 hotend with a silicone sock.  I downloaded, configured and flashed the latest Marlin 2.xx bugfix
I replaced the thermistor and heater cartridge as part of my trouble shooting.  I did over a dozen PID autotunes, at various temperatures and cycle counts, but I could never hold temp better than +- 5˚

Aside from replacing hardware, I also tweaked the PID_K1 value, PID_FUNCTIONAL_RANGE value, and BANG_MAX value.

I twisted my thermistor wires in an attempt to reduce EM interference from the heater leads.

I was previously running a Ramps 1.4 board for over a year with Marlin 1.xx and a few weeks before purchasing the SKR, I upgraded to Marlin 2.xx and the printer was performing fine with the Ramps 1.4

Today I flashed RepRap firmware on my board, did one Autotune, and the hotend is holding temp within 0.5˚

I don't understand the nuts and bolts of PID tuning theory, but whatever they are doing in the RepRap firmware, it works, and Marlin does not.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

https://github.com/BenKitesurfs/BensTarantula_MarlinConfigs/blob/main/Configuration%20Files.zip

### Steps to Reproduce

1. Install SKR 1.4 Turbo Board
2. Install e3d genuine hotend
3. Flash the latest Marlin 2.xx bug fix

**Expected behavior:**

Run PID Autotune, save PID parameters generated by the Autotune, set a temp, and have the Hotend hold the temp within 1˚

**Actual behavior:**

The temperature fluctuates +- 5˚ or more

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

BenKitesurfs, 2021-01-03:
I uploaded some screenshots showing the instability of the temperature running Marlin vs stable temps with RRF

https://github.com/BenKitesurfs/BensTarantula_MarlinConfigs

swilkens, 2021-01-04:
RRF uses Cohen-coon I believe Marlin uses Åström-Hägglund. A different tuning algorithm. 

From RRF github:
https://github.com/Duet3D/RepRapFirmware/blob/dev/src/Heating/Pid.cpp#L599

Maybe it can be interesting to see the performance of this other tuning method.

@BenKitesurfs have you tried going back a few marlin versions to see if the problem is a recent development?

BenKitesurfs, 2021-01-06:
@swilkens No I haven't.  I have been so frustrated with this issue for 2 weeks, tuning, replacing hardware, and scouring the internet for information.  I'm relieved to finally have working firmware, even though there's been a lot of headaches with learning the new firmware and getting it configured for my machine.

I am a believer in the Cohen-coon model now.  I've never had my printer hold temp this well.  It will regularly not deviate by even 0.1˚ for minutes at a time!

It would be great if more PID tuning models were developed for Marlin, with the ability for the person configuring firmware to choose a different model that may work better for them.

descipher, 2021-01-14:
May I suggest commenting out  #define SLOW_PWM_HEATERS I believe this is all or none.

BenKitesurfs, 2021-01-14:
So I can't use SLOW_PWM_HEATERS for the bed only?  My bed is fairly underpowered, so I was having problems with it timing out during PID tuning.

descipher, 2021-01-14:
At the moment they are not compatible features, I will be submitting a issue report and look at what can be done in the short term . I believe the timeouts can be adjusted. Having a look later today..

descipher, 2021-01-14:
Add this to your configuration.h
#define MAX_CYCLE_TIME_PID_AUTOTUNE 300L

descipher, 2021-01-14:
Are you certain it was a timeout the default is 20 mins.

descipher, 2021-01-14:
I recommend setting the BANG_MAX @ 255 and the PID range @ 10

descipher, 2021-01-15:
SLOW_PWM_HEATERS solves a mechanical relay problem with PWM. If you have a solid state relay it's likely not required. Setting BANG_MAX less than 255 on an under powered bed would make it even more under powered.

closed 2021-02-22 by github-actions[bot]