MarlinFirmware/Marlin #20662: Onboard SD not working for STM32F401VE_STEVAL platform

rcdaley, 2021-01-03:
### Bug Description

I am unable to get the onboard SD working on the STM32F401VE_STEVAL platform with the steval-3dp001v1 board. When I configure for SPI mode, M21 appears to init SD card ok, M500 says it saved settings, gives byte count and crc.  M20 returns an empty file list. When I look at the SD card on a PC, it has an EEPROM.DAT file with 0 bytes.

Things I've tried:
SPI mode gives the results above.  I've tried enabling 4bit mode by defining SDIO_D1_PIN, SDIO_D2_PIN and SDIO_D3_PIN in the pin configuration file.  With this configuration, Marlin won't boot with the SD card inserted.

I've tried changing the SD CARD SPI SPEED in configuration.h (not sure if this effects on board sd spi speed) but none of the settings had any impact.

I undefined SDSS in the pin configuration.  This seemed to have no impact.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Configure Marlin/platformio for STM32F401VE_STEVAL
2. Enable SDSUPPORT
3. ENABLE ONBOARD SD support

**Expected behavior:**

Normal SD operations, saving settings to SD.

**Actual behavior:**

EEPROM version error on bootup. Unable to save seetings with M500.  Unable to view files with M20.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5762362/Configuration.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

rhapsodyv, 2021-01-31:
Note: that board isn't SPI SD, but SDIO. 

Try add this:
```
#define SDIO_CLOCK                       4500000
#define SDIO_READ_RETRIES                     16
```

rcdaley, 2021-02-02:
Thanks for your help.  I made the changes you suggested but no joy.  M21 appears to work. M500 appears to work but it only creates the EEPROM.DAT file with 0 bytes.

rhapsodyv, 2021-02-02:
Did M20 work?

rcdaley, 2021-02-02:
It appears to (no error) but it only returns and empty file list.  When I check the card in a card reader, I can see where it created the 'EEPROM.DAT' file.

rhapsodyv, 2021-02-02:
Do you have any gcode file in the sd card? (Just to be sure, sorry)

rcdaley, 2021-02-02:
No, I formatted the SD card a couple times now.

rcdaley, 2021-02-02:
I'm looking at the file Sd2Card_sdio_stm32duino.cpp and wondering if this is a work in progress? I see a function called init_SDIO_pins() that is commented out.

rhapsodyv, 2021-02-02:
If you don’t have any gcode file inside the sd, it’s right to show a empty file list..........

try copying a gcode file to the sd and try again

rcdaley, 2021-02-02:
I copied a couple files to the SD card but M20 did not show anything

rhapsodyv, 2021-02-02:
Sdio code is to be very stable now, with the lasts changes.
I will take a look. Be sure to test on bugfix branch.

AndersDoverud, 2021-03-07:
I think I have the same problem, repetier writes to to the card, it doesn't shows on the display/repetier, so I can't print it...

rcdaley, 2021-03-08:
I'm trying the latest build to see if that fixes anything.

rcdaley, 2021-03-10:
I downloaded the latest bugfix-2.0 release and the board will boot with the SD card inserted so I guess that's progress.  M20 reports no media and M21 will cause marlin to crash.

rcdaley, 2021-03-12:
I hooked up a logic analyzer to the SDIO pins and on reboot I seen it switch from 400kHz SPI mode to 25MHz SDIO mode.  I seen activity on the SDIO pins so that is encouraging even though I had the SDIO_CLOCK 4500000 defined, it didn't have any impact on the actual SDIO clock rate.  I'm starting to wonder if I have a hardware issue like a bad solder joint on the SD card holder.  I'll check it out when this print finishes.

ebraiman, 2021-03-14:
I'm seeing this issue as well.

Configuration.

GTR v1.0
LCD Same as above.

Recorded a video of issue.
Drive is formatted as FAT32 for SD card and has gcode files previously printed.

This also occurred on MKS sGen L v2.0

Enclosing GTR firmware and will post link to video.

https://youtu.be/XziirWkws3Y

[Marlin-2.0.x (2).zip](https://github.com/MarlinFirmware/Marlin/files/6137619/Marlin-2.0.x.2.zip)

ebraiman, 2021-03-14:
Also tested on RAMPS 1.4 and works file.

ebraiman, 2021-03-14:
Also check FAT formatted drive with same result.

rhapsodyv, 2021-03-14:
The video is private. I can't watch.

There're some confusing information in this thread.

Onboard SD:

The onboard SD either SDIO or SPI, but not both. They are different things. Even if you can use SPI to access a SD attached to SDIO mcu pins, Marlin you need to select at compile time which protocol to use.

To use SDIO, the sd should be connected to specific mcu pins. So, each board schematics will tell if the SD is connected to the SDIO hardware of the MCU or in the SPI hardware.

Each option has different set of options to tune it. 
SDIO:
 ```
#define SDIO_CLOCK                       4500000
#define SDIO_READ_RETRIES                     16
```
SPI:
```
  /**
   * SD Card SPI Speed
   * May be required to resolve "volume init" errors.
   *
   * Enable and set to SPI_HALF_SPEED, SPI_QUARTER_SPEED, or SPI_EIGHTH_SPEED
   *  otherwise full speed will be applied.
   *
   * :['SPI_HALF_SPEED', 'SPI_QUARTER_SPEED', 'SPI_EIGHTH_SPEED']
   */
  //#define SD_SPI_SPEED SPI_HALF_SPEED
```

So, to use SPI with a SDIO card, you will need to configure a custom SPI pins.

If you are using a SD on the LCD, it will surely be SPI and you should set the `#define SDCARD_CONNECTION LCD`

And one last question: are you sure you are using the last bugfix? Because only recently the code got support for running SDIO clock.

ebraiman, 2021-03-14:
Just changed to unlisted for video.
Is RepRap Discount Smart controller SPI?  I think it's SDIO, but please educate me as I'm unsure.

rhapsodyv, 2021-03-14:
if the SD is on the `RepRap Discount Smart controller`, it's not SDIO and it's not ONBOARD. You should set: `#define SDCARD_CONNECTION LCD`.

ebraiman, 2021-03-14:
2.0.7.2 is the Marlin version.

rhapsodyv, 2021-03-14:
> 2.0.7.2 is the Marlin version. 

It doesn't have the SDIO fix.

But first of all, you need know what you have! Where is the SD located? On the board? On the LCD?

And try bugfix.

ebraiman, 2021-03-14:
I'm using LCD on display and one exists in GTR v1.0.

ebraiman, 2021-03-14:
It's reading media off micro SD card for GTR v1.0 and not off RepRap Discount Smart Controller.
What's the correct process to make SD card on reader work?

ebraiman, 2021-03-14:
Just rechecked in 2.0.7.2 for Marlin firmware.

rhapsodyv, 2021-03-14:
You should seek for support on Facebook group or marlin discord server. It will be a lot more easy than using issue queue, because people that have the same hardware are you can will be able to help better.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

If you conclude that it's a bug, you can reopen this.

rcdaley, 2021-03-15:
Since his worked with the Marlin 1.x version that shipped with the board it seems logical that it is a bug that is why I posted here.

rcdaley, 2021-03-15:
For your information, every SD card is initialized in SPI mode and then switched over to SDIO mode if the hardware supports it and the firmware requests it.  That is how SD cards are able to operate in either mode.

Thanks for the links but that was not necessary.  I reviewed the web quite extensively before coming here.

rhapsodyv, 2021-03-15:
See, I didn’t say you don’t find a bug. But there’re lots of confusion going on here. Marlin 1.x isn’t 8 bit only??
Took a lot of messages to understand what the SD are you using, which peripheral, etc, neither the board, as the issue start with one, and in the middle we have configs of another one.

So, the best I can do is to tell you to seek for help in the groups, as users that have exactly the same hardware as you, can help to properly set all the configs and do all the tests.

rhapsodyv, 2021-03-15:
@rcdaley ok, sorry. I didn’t see that it was two different users.
I will keep this open for you, you are right: @rcdaley 

But @ebraiman should look for help in the links I commented.

rcdaley, 2021-03-15:
ST Micro has a version of Marlin that they provide for the STEVAL board. I believe it reports Marlin version 1.0.  It lacked a lot of Marlin features which is why I updated to 2.0.x

rcdaley, 2021-03-15:
If you know anyone has an STEVAL-3DP001 platform running Marlin 2.0.x with functional SDIO, then I think you should close this issue.  I don't want to waste anyone's time solving my problem if it's not a bug.

rhapsodyv, 2021-03-15:
> If you know anyone has an STEVAL-3DP001 platform running Marlin 2.0.x with functional SDIO, then I think you should close this issue. I don't want to waste anyone's time solving my problem if it's not a bug.

Don't worry. I got really confused by the lasts message, mixing LCD/onboard and GTR here. It was my fault. Sorry. Let's keep it open until we get a solution.

bigtreetech, 2021-03-15:
> If you know anyone has an STEVAL-3DP001 platform running Marlin 2.0.x with functional SDIO, then I think you should close this issue. I don't want to waste anyone's time solving my problem if it's not a bug.

Hello,  In my test, the code in SDIO may have bugs.

You can change `SDIO_ReadBlock` and `SDIO_WriteBlock` to the following code to verify whether there is a problem with your hardware

https://github.com/MarlinFirmware/Marlin/blob/b9065195f1babf00ee453c39410206af90a525ae/Marlin/src/HAL/STM32/Sd2Card_sdio_stm32duino.cpp#L258
```

bool SDIO_ReadBlock(uint32_t block, uint8_t *dst) {

    uint32_t start = millis();

    HAL_StatusTypeDef stat = HAL_SD_ReadBlocks(&hsd, (uint8_t *)dst, block, 1, 5000);
    if (stat != HAL_OK) {
      return false;
    }

    while(HAL_SD_GetCardState(&hsd) != HAL_SD_CARD_TRANSFER && millis() - start < 5000)
    {
      TERN_(USE_WATCHDOG, HAL_watchdog_refresh());
    }
    if (HAL_SD_GetCardState(&hsd) != HAL_SD_CARD_TRANSFER)
    {
      return false;
    }

    return true;
  }

  bool SDIO_WriteBlock(uint32_t block, const uint8_t *src) {

    uint32_t start = millis();

    HAL_StatusTypeDef stat = HAL_SD_WriteBlocks(&hsd, (uint8_t *)src, block, 1, 5000);
    if (stat != HAL_OK) {
      return false;
    }

    while(HAL_SD_GetCardState(&hsd) != HAL_SD_CARD_TRANSFER && millis() - start < 5000)
    {
      TERN_(USE_WATCHDOG, HAL_watchdog_refresh());
    }
    if (HAL_SD_GetCardState(&hsd) != HAL_SD_CARD_TRANSFER)
    {
      return false;
    }

    return true;
  }
```

rhapsodyv, 2021-03-15:
> Hello, In my test, the code in SDIO may have bugs.
>
> You can change `SDIO_ReadBlock` and `SDIO_WriteBlock` to the following code to verify whether there is a problem with your hardware [...]

The main difference seems to be the timeout, isn’t?

bigtreetech, 2021-03-15:
> The main difference seems to be the timeout, isn’t?

No, it will wait a little bit for status OK, and the previous code will be repeated run `HAL_SD_WriteBlocks`

bigtreetech, 2021-03-15:
This code is just to test if his hardware is OK, Because It's similar to what I've encountered according to his describes. And I temporarily repaired it in this way

And I learned from search engines that STM32F4 SDIO needs to disable interrupt to prevent FIFO overflow if it doesn't use DMA to read and write. So this may not be the final solution, and we need to change it to DMA transmission

rcdaley, 2021-03-15:
Thanks, I will check this out tomorrow and let you know how it works.

rhapsodyv, 2021-03-15:
> This code is just to test if his hardware is OK, Because It's similar to what I've encountered according to his describes. And I temporarily repaired it in this way
>
> And I learned from search [...]

Interesting. Recently I fixed the clock and retries of SDIO and it got a lot stable for F1 users. Sadly, I don't have any F4 boards with SDIO. Only F1.

I will take a look where I can get a F4 board with SDIO. @bigtreetech Any of your F4 boards use SDIO?

ebraiman, 2021-03-15:
If you need to work the issue, I'd be glad to help over Discord chat.  I have an oscope and a logic analyzer someplace.

ebraiman, 2021-03-15:
Also several different types of displays.

rcdaley, 2021-03-15:
I replaced the two functions with the code above and there was a slight difference in operation.  It still won't boot with SD card inserted but if I insert the SD card after Marlin boots I can do M21 which results in volume.init failed.  the second time I try M21, Marlin reboots.  With the original code M21 would immediately result in a crash/reboot.

rcdaley, 2021-03-15:
For reference, here is the SD initialization code the ST provides with the board.  They are using DMA.

[stm32f4xx_3dprinter_sd.c.txt](https://github.com/MarlinFirmware/Marlin/files/6144052/stm32f4xx_3dprinter_sd.c.txt)

rhapsodyv, 2021-03-15:
Maybe the issue is with the variant... `buildroot/share/PlatformIO/variants/STEVAL_F401VE/`

bigtreetech, 2021-03-16:
@rcdaley 
Can you replace  ` Sd2Card_sdio_stm32duino.cpp` with the file in my compressed package? It has been modified to DMA mode. It is normal to test read and write on my side (BTW: This package is only for STM32F4 test and may not work on F1)

[Sd2Card_sdio_stm32duino.zip](https://github.com/MarlinFirmware/Marlin/files/6146647/Sd2Card_sdio_stm32duino.zip)

bigtreetech, 2021-03-16:
> I will take a look where I can get a F4 board with SDIO. @bigtreetech Any of your F4 boards use SDIO?

Yes. Some new F4 boards(E3 RRF, etc...), But may not on sale yet

rcdaley, 2021-03-16:
Absolutely.   I'll check it out in the morning.

rcdaley, 2021-03-16:
Some good news!  With the latest Sd2Card_sdio_stm32duino.cpp, Marlin will boot with the SD card inserted.  M21 initializes the SD card and M20 reports an empty file list.  M500 appears to work but no data actually gets saved to the SD card.  I verified that the SD card was empty on a PC.  On the upside, it doesn't crash anymore.  I saved a file to the SD card with a PC but M20 still reported an empty file list.  I feel like we're getting closer. Just let me know what I can do to help.

rhapsodyv, 2021-03-16:
Where I can get a F4 board with SDIO?

rcdaley, 2021-03-16:
Amazon:
https://www.amazon.com/BIGTREETECH-Direct-Expansion-Printer-Control/dp/B08QHK7RSD/ref=asc_df_B08QHK7RSD/?tag=hyprod-20&linkCode=df0&hvadid=475790020230&hvpos=&hvnetw=g&hvrand=16136725721006483602&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9026841&hvtargid=pla-1181464297342&th=1

Digikey has the STEVAL-3DP001
https://www.digikey.com/en/products/detail/stmicroelectronics/STEVAL-3DP001V1/6023689?Enterprise=44&Vendor=497&WT.z_cid=sp_497_0928_buynow&lang=en&s=N4IgTCBcDaIMoBUCiA1AggGQLQGYAiACgAxECMKpIAugL5A

rcdaley, 2021-03-16:
I hooked up a logic analyzer and there is SD bus activity for M20/M21 but no bus activity for M500.

rhapsodyv, 2021-03-17:
@bigtreetech this board has sdio BIGTREETECH E3 RRF V1.1 Control Board RRF IDEX V1.0 With TMC2209 UART 3D Printer Parts For Ender 3/5 Pro Upgrade BTT SKR V1.4
https://a.aliexpress.com/_mNOtHaB

bigtreetech, 2021-03-17:
> @bigtreetech this board has sdio? BIGTREETECH E3 RRF V1.1 Control Board RRF IDEX V1.0 With TMC2209 UART 3D Printer Parts For Ender 3/5 Pro Upgrade BTT SKR V1.4

Yes, it has SDIO (4 bits width)  The firmware is here https://github.com/bigtreetech/BTT-E3-RRF

If you need, we can send you this board for free to support open source. You can contact tech@biqu3d.com  email if necessary

rhapsodyv, 2021-03-17:
> > @bigtreetech this board has sdio? BIGTREETECH E3 RRF V1.1 Control Board RRF IDEX V1.0 With TMC2209 UART 3D Printer Parts For Ender 3/5 Pro Upgrade BTT SKR V1.4
> 
> Yes, it has SDIO (4 bits width) The firmware is here https://github.com/bigtreetech/BTT-E3-RRF
> 
> If you need, we can send you this board for free to support open source. You can contact [tech@biqu3d.com](mailto:tech@biqu3d.com) email if necessary

Thanks! It would be very helpful, as we are getting more and more powerful boards!

bigtreetech, 2021-03-17:
> Some good news! With the latest Sd2Card_sdio_stm32duino.cpp, Marlin will boot with the SD card inserted. M21 initializes the SD card and M20 reports an empty file list. M500 appears to work but no [...]

Did you get the correct answer "SD card OK" from M21 before M20?

rcdaley, 2021-03-17:
Yes.

rcdaley, 2021-03-24:
I tried playing with the SDCLOCK frequency to see if that made any difference.  I tried several frequencies from as low as 700kHz to 10~12MHz.  They all seemed to work exactly the same so I don't think that's an issue.  I may have given some bad information above regarding M20.  I said it always reports an empty file list but when I created the /SYS directory it will report the  M_CFG.G file in that directory.  I guess this is how m20 works. Also, it's been my observation that with M28 and M500 it will create files but they always have 0 bytes. I tried configuring the SDIO driver to use polling mode.  I figured this would be the most reliable method from a firmware perspective but the results were the same.  I have the debugger working in VS code and the logic analyzer connected so hopefully I can track this down.

rhapsodyv, 2021-03-29:
@rcdaley can you test this branch? 

https://github.com/rhapsodyv/Marlin/tree/fix-sdio-stm32

rcdaley, 2021-03-29:
No joy.  Ended up in infinite loop, a home for unexpected interrupts (see screenshot). This happened after SDIO_ReadBlock() was called.  I'll dig into it more tomorrow.

[image]

rhapsodyv, 2021-03-29:
@rcdaley I did one more commit, changing the irq handler for F4. Can you test again?

I don't have a F4 board yet, but I rewrote the code to work with dma. And, in my testes, everything is working on F1.

If you have discord, we can chat.

[7 more replies]

PR #21476 (2021-03-30): Fix and add DMA for SDIO on STM32
### Description

MKS just discovered that write block on STM32 SDIO is not working.

While fixing it, I already added DMA support for it, as we have on STM32F1.

I'm using DMA with interrupts.

Add two debug codes, to test write and read.

`D101` will generate a file `TEST.GCO` with a long sequence of A-Z, with about 2.1MB. Test Write.
`D102` will read this file and check if the A-Z sequence is valid. Test Read.

### Requirements

F1 board with SDIO: Nano V2 
F4 board with SDIO

### Benefits

Faster and Stable SDIO
Fix #20662

### Related Issues
closed 2021-03-30 by rhapsodyv