MarlinFirmware/Marlin #20664: [BUG] Archim1 MS1/2/3 settings not flashing correctly with Platform.io

V1EngineeringInc, 2021-01-04:
### Bug Description

When flashing the Archim1 I have been having issues for some time. I can not seem to track down when this started but now after erasing and re-flashing using a current VSCode and Platform.io M350 reports, MS1|2|3 Pins X:000 Y:001 Z:000 E0:111 E1:000 when it should be X:111 Y:111 Z:111 E0:111 E1:111.

Everything works fine when issuing,
M351 S1 X1 Y1 Z1 E1 B1
M351 S2 X1 Y1 Z1 E1 B1
M351 S3 X1 Y1 Z1 E1 B1
  
*Side note, M500 and M502 also do not seem to be working, but I only tried with Repetier host. Related?

### Configuration Files

[archim1configs.zip](https://github.com/MarlinFirmware/Marlin/files/5762978/archim1configs.zip)

### Steps to Reproduce

Erase and re-flashing using a current VSCode and Platform.io

**Expected behavior:**

M350 reports, MS1|2|3 Pins X:111 Y:111 Z:111 E0:111 E1:111
The steppers moving as far as expected.

**Actual behavior:**

M350 reports, MS1|2|3 Pins X:000 Y:001 Z:000 E0:111 E1:000
Steppers are a mess and try to rip the machine apart.

#### Additional Information

Thanks for any help you can provide.

sjasonsmith, 2021-01-04:
Try uncommenting the 32 microstep line in `Configuration_adv.h`. I believe the default HIGH,LOW,HIGH will work fine. DRV8825 have several ways to select 32 microsteps.

`#define MICROSTEP32 HIGH,LOW,HIGH`

It looks like for Archim it isn't auto-enabling 32-microsteps by default, and there isn't a sanity check to tell you the config is broken.

V1EngineeringInc, 2021-01-04:
Thanks!

Using just that line and not the other fails to compile. Doing both works. Thank you!

closed 2021-01-04 by V1EngineeringInc