MarlinFirmware/Marlin #20666: Ender3 v2 pause does not work with cura[BUG] (short description)

Funfury32, 2021-01-04:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

dimonoid, 2021-01-16:
It works. You just need to wait a little(10-20 seconds). 
I assumed you are clicking pause on the printer display.

Bluelinegecko, 2021-01-20:
I think the original poster is probably referring to problems using the pause-at-height scripts in Cura. If so it is probably related to #19870

Funfury32, 2021-01-28:
> I think the original poster is probably referring to problems using the pause-at-height scripts in Cura. If so it is probably related to #19870 

Ok thanks

closed 2021-03-07 by github-actions[bot]