MarlinFirmware/Marlin #20667: [BUG] IDEX DETECT_BROKEN_ENDSTOP causes halt

RZN-FFEvo, 2021-01-04:
### Bug Description

Dual X carriages fail to home when DETECT_BROKEN_ENDSTOP is enabled. Interestingly this only happens when X0 is home.

### Configuration Files

https://github.com/RZN-FFEvo/Marlin/raw/IDEX-Ender-3/Marlin/EndStopBugConfig.zip

Also enabled I2C_EEPROM

### Steps to Reproduce

1. home X with X0 endstop triggered

**Expected behavior:**

Both carriages should home

**Actual behavior:**

As soon as X1 releases its endstop
Recv: echo:Bad X Endstop?
Recv: Error:Printer halted. kill() called!

#### Additional Information

https://github.com/MarlinFirmware/Marlin/blob/ba2cadb47934fa83f576aece2c45f1b6cb5829cf/Marlin/src/module/motion.cpp#L1638

Seems to be the issue as it doesn't know or care about X1, looks like this will happen with other multi endstop configurations as well.

ellensp, 2021-01-04:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/20237

ellensp, 2021-01-04:
On send though looks different enough to be a separate issue

DrumClock, 2021-01-07:
Hi,
Yes,
I confirm when the X-min endstop is closed (X1 carriage - parkinkg)
after sending the g-code G28, the printer stops with the error "parking failed".

Other X-max, Y-min sensors work properly.

closed 2021-02-14 by github-actions[bot]