MarlinFirmware/Marlin #20669: [BUG] Retract V in LCD menu is in mm/m

f8k8, 2021-01-04:
### Bug Description

When firmware retract is enabled, the feedrates in Configuration_adv.h are in mm/s:

```
#define RETRACT_FEEDRATE 25             // (mm/s) Default feedrate for retracting
#define RETRACT_RECOVER_FEEDRATE 8      // (mm/s) Default feedrate for recovering from retraction
#define RETRACT_RECOVER_FEEDRATE_SWAP 8 // (mm/s) Default feedrate for recovering from swap retraction
```

but in the LCD menu, they are in mm/m.

### Steps to Reproduce

1. Enable FWRETRACT
2. Set feedrates in mm/s in Configuration_adv.h
3. Try and adjust the retraction V in the LCD menu.

**Expected behavior:**

The retraction speeds in the LCD menu should be in mm/s.

**Actual behavior:**

The retraction speeds in the LCD menu are in mm/m, so the minimum retraction speed that can be set is 60mm/s (i.e. by setting retraction V to 1 in the menu, it sets the retraction speed to 60mm/s), so there is no way to get any finer grained control of the retraction speed.

**Config:**
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5764440/Config.zip)

qwewer0, 2021-01-04:
Please test it on the latest bugfix-2.0.x
And what test did you do that showed that results?

f8k8, 2021-01-04:
Sorry, I think this is actually an issue with Repetier Server. It looks like they have the wrong units displayed when editing the EEPROM values. As far as I can see, Marlin is doing it correctly, so will close the issue.

closed 2021-01-04 by f8k8