MarlinFirmware/Marlin #20670: [BUG]Part Cooling Fan not Running - after clicking on Menu it starts with 100%, not adjustable - SKR Mini E3 V2

Bastlwastl84, 2021-01-04:
BTT SKR Mini E3 V2
Marlin Bugfix 2.0 from 3.1.2021
starting Printjob in Marlin-Mode
Part Cooling 
Fan not starting after Layer 1
Fan startes at 100% (was set to 30%) if klicking the Wheel at BTT TFT35 E3 V3

[SKR Mini E3 V3 .zip](https://github.com/MarlinFirmware/Marlin/files/5764697/SKR.Mini.E3.V3.zip)

pv501977, 2021-01-06:
Try 

#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN PC7

and/or try switch FANs

Bastlwastl84, 2021-01-06:
[image]
[image]

Bastlwastl84, 2021-01-06:
> USE_CONTROLLER_FAN

the Stepper-Cooler is running, but not the part-cooling fan at hotend. it starts if clicking in marlin-mode the wheel at display. the coldend fan is permanently on (at 24v output and cant be regulated)

Bastlwastl84, 2021-01-06:
[image]

pv501977, 2021-01-06:
Entruder Fan
 - I have connected to FAN1
 - It start run when I move the extruder e.g. on Axis X
 - I could modify speed from TFT - Configuration - Controller Fan - Idle/Active Speed

Printed parts FAN
 - I have connected to FAN0
 - I could modify speed from TFT (Temperature - FAN speed) or by M106 P0 Sxx

Bastlwastl84, 2021-01-06:
> Entruder Fan
>
> * I have connected to FAN1
> * It start run when I move the extruder e.g. on Axis X
> * I could modify speed from TFT - Configuration - Controller Fan - Idle/Active Speed
>
> Printed parts [...]

worked with older bugfix, but not now.
Steppercooler are running
Partcooling not. if i do adjust in TFT to 100% it doesn't starts

If i'm in MarlinMode it does not start, too. if i'm clicking the Menu-Wheel in Touchmode or MarlinMode the Fan starts running at 100%. if i try to adjust, nothing changes. running at 100% (while printing). it also doesn't stops running, till i click on the wheel, than it stops.
In MarlinMode, if the Fan started and i klick the wheel, it stops and after leaving menu it starts again.

empakoso, 2021-01-06:
I concur with the issue above exactly as it is reported.  I have Marlin 2.0.7.2 stable and bugfix and both are doing this on my BTT SKR mini E3 v2 board (with BTT TFT35 display).

I have read that if you disable the speaker option in configuration.h this problem goes away.  I have not tried that.  For my speaker to work I had to adjust timers.  Can anyone confirm this?

autonumous, 2021-02-06:
I have just noticed this issue today, thinking it was a damaged board.
When using BTT 2.0.72 precompiled firmware https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3, fan worked as expected.  Using Bugfix 2.0.x, the fan is on or off.
I have tried 2.0.x branch (2.0.7.2) with the same config files (configuration.h and configuration_adv.h) and fan works.

With latest bugfix, i disabled speaker, and fan become fully operational.   
I notice lastest commit on BTT GitHub is commented "fix speaker damaged fan bug”  although not sure what actually was done.....

autonumous, 2021-02-07:
for me at least i think i’ve found issue/fix

i’ve changed settings in platformio.ini, and now same code works.

Also beeper works, and an odd behaviour with button, where by actions only seem to work on second button push..

Original 
```
#
# HAL/STM32F1 Common Environment values
#
[common_stm32f1]
platform          = ststm32@~10.0
board_build.core  = maple
build_flags       = !python Marlin/src/HAL/STM32F1/build_flags.py
  ${common.build_flags}
  -DARDUINO_ARCH_STM32
build_unflags     = -std=gnu11 -std=gnu++11
src_filter        = ${common.default_src_filter} +<src/HAL/STM32F1>
lib_ignore        = SPI, FreeRTOS701, FreeRTOS821
lib_deps          = ${common.lib_deps}
  SoftwareSerialM
platform_packages = tool-stm32duino
```

used the config from BTT GitHub

```
#
# HAL/STM32F1 Common Environment values
#
[common_stm32f1]
platform      = ststm32@~6.1
build_flags   = !python Marlin/src/HAL/STM32F1/build_flags.py
  ${common.build_flags} -std=gnu++14 -DHAVE_SW_SERIAL
  -DTONE_CHANNEL=4
  -DTONE_TIMER=4
build_unflags = -std=gnu11 -std=gnu++11
src_filter    = ${common.default_src_filter} +<src/HAL/STM32F1>
lib_ignore    = SPI
lib_deps      = ${common.lib_deps}
  SoftwareSerialM
custom_marlin.NEOPIXEL_LED = Adafruit NeoPixel=https://github.com/bigtreetech/Adafruit_NeoPixel
```

empakoso, 2021-02-07:
I concur! automumous fix works.  Using the BTT platformio settings allows speaker and M106 to work together.  It also fixes the neopixel issue I had ealier.  This simplifies all the custom changes made for all three issues.  BLtouch and UBL all working here as normal using the TFT35 E3 V3

DarkSnart, 2021-02-13:
@autonumous I was ecstatic when I finally came across your post after DAYS of searching, interrogating and compiling; all the while hope was rapidly fading. YES! FINALLY!! I then realized I've indeed been using the BTT platformio for the past few days so unfortunately this didn't help me 😪 

My situation is a part cooling fan which runs constant with no way to turn it off. When I increase the fan speed through the control interface (BTT TFT35) the fan actually slows down, but the display shows increase as I would expect. Same if I execute M106 S255, the fan slows down, and conversely M106 S0 speeds it back up. Output at FAN0 when M106 S255 is 7.3v, and M106 S0 shows 12.1v which again is totally bass-ackwards of any logic I hold within. Any thoughts on this? Please? Anyone?

btt_configs.zip

pitti98, 2021-03-02:
After a long research I finally found this ticket, and I am happy to report that @autonumous fix works!

The controller fan works and is adjustable manually or by USE_CONTROLLER_FAN, and the part fan also works via the temperature menu and via M106 and is adjustable.

The only thing is that if they are not on 100% you can hear a high pitched sound (from the PWM?). I will try tweaking the PWM settings and see if I can make it go away.

But anyway, I'm happy the fans work at all. My last prints were all without part cooling... Thanks @autonumous!

Rudi-GH, 2021-03-03:
I can also confirm that the workaround from @autonumous works. Full working Cooling Fan !! Including the high pitched sound mentioned by @pitti98  :-( but that is a minor setback compared to having no fan at all....

thisiskeithb, 2021-04-10:
Please see https://github.com/MarlinFirmware/Marlin/pull/21488. The boards will be migrated out of the STM32F1 HAL soon, so these fixed won’t be necessary.

strayr, 2021-05-01:
@autonumous 's ini fix just worked for me on bugfix 2.0.x commit 082fce5e3e68c7b8657201e5de5adba0e4f86a33 
thanks everyone

slowbro, 2021-07-08:
#21488 was merged, I think this can be closed.

Doormat1, 2021-07-08:
@autonumous fix doesnt work any more as the platformio.ini file has changed and doesnt include this information any more. Any suggestions how to fix this? Ive downloaded the bugfix (config ended in 901) about a month ago and just noticed my fan isnt working again.

slowbro, 2021-07-12:
The platformio configs were broken out into separate files in the `ini/` dir. It should be in `ini/stm32f1.ini`

Doormat1, 2021-07-12:
Thanks very much. I will take a look

Freethefire, 2021-07-25:
I'm sorry for bringing this up again, but I don't understand how to use @autonumous fix now that the code has been moved to the ini files. Cannot find the "original" code to replace with the custom one anymore? Thanks.

thisiskeithb, 2021-07-25:
> I'm sorry for bringing this up again, but I don't understand how to use @autonumous fix now that the code has been moved to the ini files. Cannot find the "original" code to replace with the custom one anymore? Thanks.

You don't have to edit anything. Use a one of the non-`*_maple` environments.

Freethefire, 2021-07-25:
> > I'm sorry for bringing this up again, but I don't understand how to use @autonumous fix now that the code has been moved to the ini files. Cannot find the "original" code to replace with the custom one anymore? Thanks.
> 
> You don't have to edit anything. Use a one of the non-`*_maple` environments.

Thanks for quick reply!
Trying to compile latest bugfix with the correct example config files, but the fan on Fan0 is always on and Fan1 does not work at all or is at least not responding to lcd inputs. But maybe I've missed something?

Freethefire, 2021-08-19:
Sorry for replying late.
The problem was broken fan MOSFETs. Not software related.

slowbro, 2021-08-19:
Understood - thanks for updating us!

GJSchaller, 2021-11-13:
As a heads up for anyone looking into this, there is a similar issue (#23094) that is not directly related to the SPEAKER or build environment, it seems.  I thought I'd solved this, but my fan stopped working again, and I found the new issue mentioned above.

IrnScorpion, 2021-11-17:
Using 2.0.9.2 downloaded 2 days ago and my cooling fan also does not work. Running a BTT SKR E3 DIP. 
env:STM32F103RC_btt

Is there a workaround for this?

ellensp, 2021-11-17:
@IrnScorpion you have to apply the PR https://github.com/MarlinFirmware/Marlin/pull/23125 or wait till it is properly finished and added into marlin.

holocronology, 2021-11-17:
Also have this issue as of 2.0.9.x ... thought it was the fan itself but now believe it to be this issue.

IrnScorpion, 2021-11-19:
There is a lot going on in this feed and I am having a hard time deciphering what they changed. 
I tried the nightly build, but that gave me a whole new set of problems.   

Can someone please explain how I apply this fix to the current 2.0.9.2 "latest release" build?

thisiskeithb, 2021-11-23:
#23125 has been merged.