MarlinFirmware/Marlin #20672: [FR] Probe offset wizard should measure at the same point as the probing took place

HiLivin, 2021-01-04:
### Description

Probe offset wizard should measure the Z offset at the same point at which the probe homed.
With Z_SAFE_HOMING and PROBE_OFFSET_WIZARD enabled, the probe offset wizard autohomes, probes the center point (Z_SAFE_HOMING_X_POINT, Z_SAFE_HOMING_Y_POINT) of the bed and without moving in XY waits for the user to manually lower the nozzle to the bed and establish the Z offset. Given the bed is not perfectly levelled and the nozzle is lowered at a different point than the probe (NOZZLE_TO_PROBE_OFFSET), there could be a significant error in the measurement.

### Feature Workflow

After homing and before the user's measurement the carriage should move to (Z_SAFE_HOMING_X_POINT+NOZZLE_TO_PROBE_OFFSET[0], Z_SAFE_HOMING_Y_POINT+NOZZLE_TO_PROBE_OFFSET[1]), so that the nozzle is above the same point as the probe was.

qwewer0, 2021-01-04:
Did you test it on the latest bugfix-2.0.x?

HiLivin, 2021-01-04:
No, sorry. It was actually resolved in #20239.

closed 2021-01-04 by HiLivin