MarlinFirmware/Marlin #20674: BTT_SKR_V1_4_TURBO and  TFT35-v2

Garyr14, 2021-01-04:
Bug Description

Running BTT SKR V1.4 and BTT TFT35_V2 using Bugfix 2.0x dated May 26, 2020.  Everything has been working fine.  Compiled latest version of Bugfix 2.0x and updated configuration.h and configuration_adv.h accordingly.  Screen comes on with Logo, but seems to continuously "reboot".  Can initiate Touchscreen mode, but get the "no printer attached" message.

I do not think the problem is with the configuration files, as I can use them in the May 26 version (I have to change the Homing_Feedrate variables for backwards compatibility).

I have made no changes to any other files in the repository.  

While I don't think this is a bug per se, I can't seem to get help elsewhere.  I think it might have something to do with changes to the pins_BTT_SKR_V1_4.h file, but I'm not sure.

### Configuration Files
See below:

### Steps to Reproduce
Compiled using the same configuration files in the most recent repository and one dating back to May, 2020.  The older one works fine.  The new one simply keeps "rebooting".

**Expected behavior:**
Expect controller to boot up normally!

**Actual behavior:**

#### Additional Information

`/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 020008

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * https://reprap.org/wiki/Calibration
[... 4735 lines cut ...]
  #define PAUSE_PARK_NOZZLE_TIMEOUT           45  // (seconds) Time limit before the nozzle is turned off for safety.
  #define FILAMENT_CHANGE_ALERT_BEEPS         10  // Number of alert beeps to play when a response is needed.
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT           // Enable for XYZ steppers to stay powered on during filament change.

  //#define PARK_HEAD_ON_PAUSE                    // Park the nozzle during pause and filament change.
  //#define HOME_BEFORE_FILAMENT_CHANGE           // If needed, home before parking for filament change

  //#define FILAMENT_LOAD_UNLOAD_GCODES           // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.
  //#define FILAMENT_UNLOAD_ALL_EXTRUDERS         // Allow M702 to unload all extruders above a minimum target temp (as set by M302)
#endif

// @section tmc

/**
 * TMC26X Stepper Driver options
 *
 * The TMC26XStepper library is required for this stepper driver.
 * https://github.com/trinamic/TMC26XStepper
 */
#if HAS_DRIVER(TMC26X)

  #if AXIS_DRIVER_TYPE_X(TMC26X)
    #define X_MAX_CURRENT     1000  // (mA)
    #define X_SENSE_RESISTOR    91  // (mOhms)
    #define X_MICROSTEPS        16  // Number

ellensp, 2021-01-04:
please remove these config files from the body of the message, it makes it unusable. Just drag and drop the config files to attach them to the post 

you can view the entire history of that file via git eg git log -p Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_4.h

Garyr14, 2021-01-08:
Thank you for the advice.  Much appreciated.
I was able to narrow it down to changes made between 9aee6674bb and 4fccb92e07 (October 21 and October 22.  Sorry - I assume there is a proper way to refer to these commits..  
I've tried comparing what changed between these two and while it may have something to do with TMC Interpolation per driver, I can't seem to figure out why it is affecting my setup to the point where the board will not connect to the printer.

Garyr14, 2021-01-09:
@ManuelMcLure
I narrowed this down to basically the same as #19974.  The only difference is that I'm using TMC2130's.  When I uncomment #define HYBRID_THRESHOLD, all I get is a boot loop.  Disabling it allows everything to work fine.  I used the identical config files in the commit prior to 4fccb92 and it runs fine with Hybrid_threshold enabled. 
I compared the two folders and the only files changed were TMC_UTIL.CPP, conditionals.post.h and trinamic.cpp.

Also tried fresh install as the original poster did, but did not have the same good fortune.  
Any ideas?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5789817/Configuration.zip)

ManuelMcLure, 2021-01-09:
As I mentioned in the other issue, there should be no change in behavior between the two versions unless you specifically change one of the per-driver interpolation variables. The only thing I can think of is that this is causing a slightly different memory layout that's causing some other bug (like a buffer overflow) to trigger a bootloop. If that's the case, this is going to be a bear to track down.

ManuelMcLure, 2021-01-09:
One possible thing to try - can you try unplugging the serial port to the TFT? I don't _think_ the TFT sends any commands over serial to Marlin when it's in emulation mode, but just in case it is and it's trying to get data and that data code is causing the bootloop...

Garyr14, 2021-01-09:
Thanks for the quick reply.  I'll try the serial port.

ManuelMcLure, 2021-01-09:
Since I recently added a TFT35 E3 V3 to my printer, I just tested latest `bugfix-2.0.x` enabling `HYBRID_THRESHOLD` on my build (I have a Re-ARM which uses the same processor as the SKR1.4) and saw no issues. So this seems to be triggered by very specific configurations. This makes it more likely that it's being caused by memory being in a very slightly different layout.

Garyr14, 2021-01-09:
Thanks again.  Would you mind sharing your config files with me?  I'm curious to see what the differences are.

ManuelMcLure, 2021-01-09:
Check out https://github.com/ManuelMcLure/Marlin/tree/bf-2.0.x-configured - the last commit will always be my current configuration. The checked in version does not have `HYBRID_THRESHOLD` enabled, but I die enable that when testing.

closed 2021-01-18 by Garyr14