MarlinFirmware/Marlin #20685: How to change configuration to a delta robot not a printer?

12343954, 2021-01-05:
### I wanna make a delta robot, not printer, via Marlin.

Motherboard : MKS SGEN L V2.0
Stepper driver :  TMC2225
Stepper motor : NEMA 23

Like this,
[image]

but lots of marlin delta printers like this, with sliding tracks.
[image]

`M665` has 8 parameters, I don’t know what these parameters mean. Is anybody who can tell me? thank you!
[image]

ellensp, 2021-01-05:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

12343954, 2021-01-05:
@ellensp OKey, I see.

closed 2021-01-05 by 12343954

ellensp, 2021-01-05:
But come to the discord server, more than happy to talk about it there.

12343954, 2021-01-05:
@ellensp really? can you help me? how to connect with you on discard?