MarlinFirmware/Marlin #20688: [BUG] Support of TMC 2208 in standalone mode is broken

Nalorokk, 2021-01-05:
### Bug Description

Extruder driver seems to stop during first layers of almost every print, sometimes it can work a little longer. It happens for me only on Marlin, and I never saw this problem on actual Klipper or Chitu firmware which clearly tells this is firmware bug. Seems like here is fix in Klipper firmware [git](https://github.com/KevinOConnor/klipper/pull/2597). While you may say that it is problem of driver itself, but it works for me in Klipper with harsh pressure advance and bigger accelerations, I never saw it stopping like that in other firmwares, while it often shut downs even from extruder movement in menu in Marlin.

### Configuration Files

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5771506/config.zip)

I tried different MINIMUM_STEPPER_PULSE, MINIMUM_STEPPER_POST_DIR_DELAY, MINIMUM_STEPPER_PRE_DIR_DELAY, LIN_ADVANCE, E0_DRIVER_TYPE. It seems like increasing delays may somewhat little increase driver stability, but it won't solve problem.

Seems like solution described in this quote from someone genius:

> When using stealthChop, please always make sure, that you use velocity ramping. A hard stop will cut away motor back-EMF at once. As stealthChop is a voltage based chopper, it cannot respond to this at once, like spreadCycle. The result is an overcurrent, and the motor driver goes to overcurrent switch off, until it becomes disabled / enabled again.

>To resolve the problem, please use at least a tiny velocity ramping, when hard stopping the motor, e.g. within a few / a few ten microsteps.

### Steps to Reproduce

1. Start any print on marlin
2. Soon enough extrusion will stop

**Expected behavior:**

I would prefer if it would not stop during prints.

**Actual behavior:**

Extruder motor is powered off completly until printer reboot.

samtny, 2021-01-06:
Was just coming here to open the same issue.

If I turn off LIN_ADVANCE in the config then Marlin works okay (at least so far), this seems to be supported by others I have been reading. "No Linear Advance with TMC2208".

However:

I had found this commit where the issue was "fixed" in Klipper: https://github.com/KevinOConnor/klipper/commit/c9cb462f90a68deb73cacb179f7bbcde2cb9aeed

But @Nalorokk 's link above may be a more up-to-date fix, not sure yet:

https://github.com/KevinOConnor/klipper/pull/2597

The new v4.2.7 "Silent" boards on Amazon have these TMC2208 / TMC2249 (drop-in replacement for 2208) in them. Seems many of us got a 3D printer for the holidays and want to compile our own Marlin, but running into this issue on these boards. ;)

Nalorokk, 2021-01-06:
> If I turn off LIN_ADVANCE in the config then Marlin works okay (at least so far), this seems to be supported by others I have been reading. "No Linear Advance with TMC2208".

Well, for me without LIN_ADVANCE it just works for little longer, but still unreliable even for test cube.

samtny, 2021-01-06:
24 hours of prints in and extruder continuing to work for me. 

I know if I turn LIN_ADVANCE on it will freeze up immediately.

I also have the S CURVE stuff turned off, junction deviation off, etc.

@Nalorokk Out of curiousity, what are the voltages you have coming out of your stepper drivers (the 2208's)? Mine is at about 1.1V - I thought this was rather high compared to "stock" for my setup (Ender 3, 4020 motors), but I dunno if these TMC standalones are supposed to be set higher than other drivers.

devin122, 2021-01-09:
FWIW, I'm having the same issues with with my 2208s in non-standalone. Some days it works fine, some days it seems to consistently stop extruding during the first few layers. I ended up tracking it down to the drivers built in short-circuit protection, which fits well with the Klipper explanation given

samtny, 2021-01-13:
The solution for Klipper looks ... pretty involved to me, unfortunately I don't know enough about how Marlin / stepper logic works to even begin trying to port it.

Two commits from @KevinOConnor seem to be involved in the fix:

- https://github.com/KevinOConnor/klipper/commit/c9cb462f90a68deb73cacb179f7bbcde2cb9aeed
- https://github.com/KevinOConnor/klipper/commit/acd165cbea2f48812b7fd96c28b81622e5c13665

The second appears to be an improvement / fix against the first.

It seems he is putting the computed steps into a queue, evaluating the queue for rapid "step + change direction + step" instructions, then pausing the queue at the direction change and inserting a delay before the next step, so as not to overwhelm / trigger voltage overage on the steppers.

ellensp, 2021-01-13:
that system in completely incompatible with marlin at this time.  Re posting it everywhere is not helping anyone

sjasonsmith, 2021-01-17:
Duplicate of #16076

closed 2021-01-17 by sjasonsmith