MarlinFirmware/Marlin #20693: Babystep and bed leveling.

DJDOUBLEG, 2021-01-06:
Marlin 2.0.7.2 (also with bugfix 2.0.x)
1. Baby stepping doesn't really work( Under printing double click). No matter where I set it (plus or minus), the head just moves away from the table. (think fw bug)
2. At the same time, the leveling is also shit. It weighs the table and although it looks like the Z engine is making corrections (i see on the OLED display) but not enough, not so, and not well. There are places where the filament flattens out, and from where it is almost extruded into the air.
Z = Closed loop driver
Touch-Mi bed level sensor
BTT GTR v1.0
XYE = 2209
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5774714/config.zip)

ellensp, 2021-01-06:
```
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

<THE TEMPLATE>

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

rhapsodyv, 2021-01-06:
It's working, your config that is "wrong". You set `BABYSTEP_MULTIPLICATOR_Z` to 1...so, each baby step command will move only one single step! You set 682 steps per mm in your Z... So you need to command 682 times to move a SINGLE MILLIMETER.

Probably you have issues in the levelling configs too.

So, next time, pay attention to your configs before say that something is "shit". 

I will close this, as it's not a bug, and lock because it don't follow issues template and it's offensive.

closed 2021-01-06 by rhapsodyv