MarlinFirmware/Marlin #20698: M84 or M18 not turning off drivers for Azteeg X5 GT

cope413, 2021-01-06:
### Bug Description

Neither M84 nor M18 disable the motors on Azteeg X5 GT
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5777565/Configuration.zip)

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

Default Azteeg_X5_GT pins file

### Steps to Reproduce

As soon as printer is powered on, motors are engaged and will not disengage.

**Expected behavior:**
Motors should turn off with M84 or M18

#### Additional Information

Using TMC 2660 drivers (Panucatt Bigfoot drivers)
I've played with STEPPER_DEACTIVE_TIME with no effect.
Also played with HOLD_MULTIPLIER

M43 pin log included
[pin log.txt](https://github.com/MarlinFirmware/Marlin/files/5777573/pin.log.txt)

ellensp, 2021-01-07:
reading https://panucattdevices.freshdesk.com/support/solutions/articles/1000227573-bigfoot-bsd2660
"alpha_en_pin - needs to be set to nc"
This would imply these drivers are always enabled. Not firmware controllable.

cope413, 2021-01-07:
I was confused by that as it is all smoothie firmware, but I suppose I should have thought about how that would apply to Marlin code.

SOFTWARE_DRIVER_ENABLE fixed it, thank you.

closed 2021-01-07 by cope413