MarlinFirmware/Marlin #20700: [BUG] Ringing on circles while printing fast

gandzia44, 2021-01-06:
### Bug Description
While printing circles with high-speed 150mm/s, I can sense/hear/see vibrations, the same circle with speed around 80mm/s works fine.

Jerk: 12
Acceleration: 3500

When I increase Jerk to 40, circles are being printed way smoother, but any angles are painful.

The machine I'm dealing with has like 100kg weight and 4,5A Nemas on axes, it is super stiff, but it can't print fast because from the second room it sounds like "shooting" with gun ^^

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5778436/Marlin.zip)

### Steps to Reproduce

1. Build a huge, stiff machine with huge nemas
2. Tell it to do circles fast
3. I was testing it on https://www.thingiverse.com/thing:2046448/files scaled 200%

**Expected behavior:**

Super smooth printing

**Actual behavior:**

Super ringing printing

#### Additional Information

Setup, that I'm using:
- SKR PRO 1.1 connected with Raspberry Pi (Repetier Server)
- External stepper drivers (they are for sure working fine - industrial grade)

TESTED:
- SD card printing
- Increasing BLOCK_BUFFER_SIZE to 64

Maybe I don't understand jerk and acceleration right?

MasterPIC, 2021-01-07:
Similar issue on my cheap Anet A8 printer (also on small circles).
I increased baudrate to 250000bps and BLOCK_BUFFER_SIZE to 64.
Then I switched from (Octoprint) serial to SD but uselessly.
By postprocessing GCODE to make use of ARC commands I'm now able to print by Octoprint too with no issues.
I know that the hardware is different, also mechanical specs differ; so I'm not sure my report can be of help.

gandzia44, 2021-01-08:
I'm pretty sure, that postprocessing Gcode to arc commands would work in this situation too, but it's not a solution 😞

gandzia44, 2021-02-10:
up

closed 2021-03-22 by github-actions[bot]