MarlinFirmware/Marlin #20703: [FR] Allow runtime / eeprom configuration of PROBING_MARGIN

CRCinAU, 2021-01-07:
### Description
Given that users often have different clips or devices, it would be nice to be able to configure the probing margin to be used without having to rebuild a firmware to change this margin.

It would be good to have this as a storable setting in the EEPROM vs a G29 command option to allow for set / forget when using various scripts / example codes that only say to use the G29 option.

SpectrumTech, 2022-02-16:
Got my vote!

eneene, 2025-05-19:
Got my vote too!
I run fairly large printers (550x550) And i use beds that are smaller than the print area (350x350) since most of my models touch a small part of the bed and grow outwards as they are printed.
I use smaller beds because they are cheaper, heat up faster and consume a lot less power for the same temperature.
Normally i set a big probing margin (110mm) so it probes the smaller bed but keeps the whole build size.
It would be amazing if i could set this margin at runtime. I could then swap the bed with a full size one when needed, and could still use ubl !!!