MarlinFirmware/Marlin #20706: [BUG] Babystep/Z probe offset problem

DJDOUBLEG, 2021-01-07:
Marlin 2.0.7.2 (also with bugfix 2.0.x)
Baby stepping/Z probe offset doesn't really work. No matter where I set it (plus or minus), the head not move from the table.

Z = Closed loop driver
Touch-Mi bed level sensor
BTT GTR v1.0
XYE = TMC2209

Previous answer:
"It's working, your config that is "wrong". You set BABYSTEP_MULTIPLICATOR_Z to 1...so, each baby step command will move only one single step! You set 682 steps per mm in your Z... So you need to command 682 times to move a SINGLE MILLIMETER."

Why show in display -4.171 mm ?
And if I turn the knob, the distance doesn't change either.
See the attached video.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5781223/Marlin.zip)

https://user-images.githubusercontent.com/48370718/103892954-b8ed5e80-50ec-11eb-84f4-0c6c2e83a9e7.mp4

rq3, 2021-01-07:
@DJDOUBLEG , you can spin that encoder until the cows come home, but don't you have to actually press the knob to make the input take effect? Just curious.

DJDOUBLEG, 2021-01-08:
> 
> 
> @DJDOUBLEG , you can spin that encoder until the cows come home, but don't you have to actually press the knob to make the input take effect? Just curious.

No, Babystep (if you double-click on the menu) is already alive as you rotate the knob. No need to press.

rq3, 2021-01-08:
> No, Babystep (if you double-click on the menu) is already alive as you rotate the knob. No need to press.

Interesting. Our printers work differently. On mine, there are two ways to get to babystep. One is a double click from the main screen, and the other is through the "Tune" menu. In both cases, the babystep doesn't actually happen until I enter the required Z axis distance, and THEN enter it. The z axis doesn't actually move while the babysteps are being entered.

DJDOUBLEG, 2021-01-08:
> Interesting. Our printers work differently. On mine, there are two ways to get to babystep. One is a double click from the main screen, and the other is through the "Tune" menu. In both cases, the [...]

So far, in the case of double-clicking, Babystep from the main menu worked without pressing a button.

sortagreg, 2021-01-10:
I am having this exact same issue. I can spin that knob to be +-2.000mm and nothing changes.

Norien, 2021-01-10:
Posting to follow along as I also can't get live z-offset or babystep to work. Thanks

rhapsodyv, 2021-01-10:
Can you test disabling `INTEGRATED_BABYSTEPPING`?

And, again, enabling `BABYSTEP_MILLIMETER_UNITS` and changing `BABYSTEP_MULTIPLICATOR_Z` to `0.1`.

DJDOUBLEG, 2021-01-11:
> 
> 
> Can you test disabling `INTEGRATED_BABYSTEPPING`?
> 
> And, again, enabling `BABYSTEP_MILLIMETER_UNITS` and changing `BABYSTEP_MULTIPLICATOR_Z` to `0.1`.

Now works with this settings, but the Probe Z Offset not synchronized with babystep.
How can i syncronizing this two value?

rhapsodyv, 2021-01-11:
Hum, what shows `M851`?

Because, having `BABYSTEP_ZPROBE_OFFSET` enable, it sincronizes with Probe Z Offset. 

Are you storing the settings, after doing baby steps? It doesn't save automatically. You need send M500, or go to menu configuration -> store settings.

DJDOUBLEG, 2021-01-11:
Send: M851
Recv: Probe Offset X47.0000 Y-7.0000 Z-4.0000

In the first video see the problem. Baby step and Probe Z offset is not in sycnron.

Sh7777777, 2021-01-15:
> Marlin 2.0.7.2 (also with bugfix 2.0.x)
> Baby stepping/Z probe offset doesn't really work. No matter where I set it (plus or minus), the head not move from the table.
>
> Z = Closed loop driver
> Touch-Mi [...]

DJDOUBLEG, 2021-01-23:
Ping.
Any update?

DJDOUBLEG, 2021-01-23:
it doesn't matter where I turn the Z-probe, the bed just moves down. It looks perfectly good in the video.

https://user-images.githubusercontent.com/48370718/105610221-d853df00-5dae-11eb-9800-3a6ac8f983b3.mp4

rq3, 2021-01-23:
@DJDOUBLEG , your video shows what I would expect to see. But babystepping or off-setting isn't saved unless you explicitly do so (at least with my set-up). After a calibration, I babystep to get an excellent first layer, then add (or subtract) the babystep distance to the z-probe offset, and save that to the eeprom. For example, my "default" z-probe offset is -16.3 mm (the microswitch triggers 16.3 mm below the nozzle). I generally have to correct that to -16.25 mm after calibration, and save the change, to get consistent and reproducible first layers.

DJDOUBLEG, 2021-01-23:
That's not how it's worked so far. As you can see in the second video,
babystepping could be adjusted on the fly. It could be now, but the table's
just moving away from the table.

rq3 <notifications@github.com> ezt írta (időpont: 2021. jan. 23., Szo
21:57):

kevinpawsey, 2021-02-18:
I had this similar issue, it wasn't 'live' stepping... I had to set **BABYSTEP_WITHOUT_HOMING** and **BABYSTEP_ALWAYS_AVAILABLE** to get it to work.

One thing that I am finding is that I can't do the double-click to get to the Z-Offset adjustment menu quickly (at the moment I am just living with it... although would be nice to have it back).  If I try and enable **DOUBLECLICK_FOR_Z_BABYSTEPPING**, it conflicts with (I think) **BABYSTEP_ALWAYS_AVAILABLE**, which then means that there has to be movement for it to work.

Anyway, might be worth checking the first two options to make sure that they are enabled for live babystepping?

djdoubleg, 2021-06-19:
The bug still exists in 2.0.9. Wherever I turn the knob the bed just moves up.

Itox001, 2021-12-14:
@Sh8tan maybe related to #19899?

n9jcv, 2022-03-12:
@kevinpawsey Thank You, this solved my problem

Sh8tan, 2022-03-13:
@Itox001 did you finally solve the problem ?

Still have the problem with ABL.

Z_probe_offset are change when baby stepping but not considered when start a new print.

Z_probe_offset set to 0.1 then g29, during print changing by 0.18, so perfect layer pr0n but next print or when turn off the printer I can see the new z_probe_offset from 0.1 to 0.18 but still have bad first layer...

Itox001, 2022-03-13:
@Sh8tan I did not solve this particular issue, but realized that perhaps my understanding of how it's supposed to work was not right. I think that the probe Z offset is only consideresd if you are probing a mesh again, and is doesn't work as a persistent home z offset as you have found, and as I initially thought.

In the end I applied the offset to my saved mesh and it was correctly set for the subsequent prints. Anyways I have moved on to klipper now so I did not debug further.

Sh8tan, 2022-03-13:
@Itox001 by moving to kipper you finally solved it 👌

Sh8tan, 2022-03-13:
@Itox001 can you just tell us how did you do to add offset directly to your saved mesh without doing a g29 ?

Itox001, 2022-03-13:
@Sh8tan I used UBL, and did it with the g-code G29 P6 Cx.x: https://marlinfw.org/docs/gcode/G029-ubl.html

DiskDoctor, 2022-04-29:
> @Sh8tan I used UBL, and did it with the g-code G29 P6 Cx.x: https://marlinfw.org/docs/gcode/G029-ubl.html

Very interesting...

Can you please give a list of the Babystep related options you have enabled and disabled to make this work?

Due to the ABL not working properly, I am switching Ender 3 Pro w/CRTouch over to UBL.

thisiskeithb, 2022-04-29:
Closing since OP deleted their account.

closed 2022-04-29 by thisiskeithb

DiskDoctor, 2022-04-29:
> Closing since OP deleted their account.

Not a good reason to close an open BUG that still exists, as does a possible workaround. Please reopen and let this play out so we can summarize the information.

thisiskeithb, 2022-04-29:
> Not a good reason to close an open BUG that still exists

That was not confirmed and OP isn’t around to respond.

We get a lot of “me too” reports with no additional information or configs (just like in this very issue.)

Please open a new bug report and follow the template carefully, including listing the **exact steps** to reliably reproduce bug and be sure to **include your config**.

I Have not seen this bug and cannot reproduce the issue on my end.

DiskDoctor, 2022-04-29:
Gotcha. Thanks.