MarlinFirmware/Marlin #20709: [BUG] MAX6675 and Touch UI on board

rafaljot, 2021-01-07:
I tried to configure Mingda MD-16 board it is similar to MKS_ROBIN with TFT and thermocouple amplifier on board. All works great expect  MAX6675 and Touch UI together. 
I found solution, it is modification in temperature.c line 1672 from 

  SPIclass<MAX6675_DO_PIN, **SD_MOSI_PIN**, MAX6675_SCK_PIN> max6675_spi;

to
  SPIclass<MAX6675_DO_PIN, **MAX6675_MOSI_PIN**, MAX6675_SCK_PIN> max6675_spi;

and assign MAX6675_MOSI_PIN to any unused PIN

I'm not sure how to correct this in right way because it is read only SPI device and do not need MOSI PIN. Especially it doesn't need connection to SD card ;)

rafaljot, 2021-01-07:
https://github.com/MarlinFirmware/Marlin/pull/20711

sjasonsmith, 2021-01-17:
This issue is being closed because it is incomplete. When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.

For reference, these are the instructions included in the template:

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

closed 2021-01-17 by sjasonsmith