MarlinFirmware/Marlin #20710: [BUG] (Babystepping only applies to the first layer)

vguttmann, 2021-01-07:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Start a print
2. Babystep Z down before the first layer using the double click babystep (maybe also up and through the "Tune" menu, but I didn't test that) 
3. Measure the first layer thickness (I did it with stopping the print and using a micrometer)
4. Start a second print and measure the combined layer thickness (which should be the desired layer height plus the babystep height (yes, plus. I babystepped .15mm down, and my layer thickness was 0.35 instead of 0.4))

**Expected behavior:** The print should measure 0.4mm in thickness

**Actual behavior:** The print measured 0.35mm in thickness, which is 0.4mm + -0.15mm. Seems like Babystep only applies to the first layer.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing 
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5783575/Configurations.zip)
to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

rq3, 2021-01-07:
@vguttman, your description is exactly as babystepping should work. It allows you to correct the very first layer of the print in real time, to get the proper "squish" and adhesion.

The babystep results are not saved, and must be repeated for any further print. You can make the change permanent by adding it to your z-axis probe offset, and saving that with M500.

By way of example, if you start a print with all layers 0.2 mm thick, and babystep the first layer 0.05 mm thinner, then the first layer will be 0.15 mm thick, and all other layers 0.2 mm. To save that change, you would have to correct your z-height offset by the babystep amount, and save that change via M500.

There is no bug here, just a mis-understanding of babystepping, what it does, and how to permanently incorporate its effect.

vguttmann, 2021-01-07:
Oh my god, this is embarrassing: I just noticed two errors: first of all, 0.4 - 0.15 is not 0.35. And if I babystep down, the second layer can't even be stashed into the first one by ignoring babysteps. 

What lead me to believe that there is a problem was the fact that I had an Overextrusion Problem with the second layer which I could solve by moving up 0.15mm. Mind you that it ONLY affected - the second layer. So there is definitely _something_ going on, but I don't know why. Sorry, in the moment I really thought that there was a bug in Marlin.

closed 2021-01-07 by vguttmann

rq3, 2021-01-07:
@vguttmann, not embarrassing at all. There are so many variables in Marlin that are so poorly defined that it's impossible to keep track of them all. The latest issues with touch displays is indicative. The main Marlin development team does a phenomenal job, but they can't do any of it without feedback.

I hear your issue. Try placing a large and slow skirt in your slicer, which will allow you time to play with babystepping before the real print starts. Save the result to a properly calculated z offset. That should get you into the "hit print and have a coffee" category, unless you change any parameter (print surface, thickness, nozzle change, etc.).

vguttmann, 2021-01-07:
I mean, I got an ender 3, which is notorious for a bad bed level. Basically I now use a custom filament gcode in prusaslicer that babysteps down and sets flow to 85%, which weirdly enough gives me a perfect first layer, and then babystep up and set flow to 100%, which makes the rest of the print perfect. I know it's not a beautiful solution, but it's quick, and if I were tech-illiterate, I wouldn't have configured my own firmware (and I also wouldn't be using Linux). Once I got my babystep dialed in (using one-layer prints and a nice ratcheting micrometer for consistency), I can usually keep that until I remove the glass surface for cleaning off the glue stick I have to use. It came that way, and a magnetic spring steel surface is on the way, so I hope it gets better soon.