MarlinFirmware/Marlin #20716: [BUG] UBL loses track probe position and probes the wrong places

zmiguel, 2021-01-07:
### Bug Description

When using UBL I'm unable to complete the mesh creation (G29 P1) because the printer starts probing outside of the bed area.
This happens because some of the movements are wrong, it consistently does the first 10 correctly, often does the first 20 correctly, and then starts messing them up, doing either too short or too long movements. Many times being off by more than 10cm from the value reported by the printer vs the actual position of the extruder/probe.

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5784411/Configuration.zip) - Config files for latest bugfix
[Link to console log while running G29 P1](https://gist.github.com/zmiguel/5a381d09a1ada5abedb22f4844e65fc9?fbclid=IwAR3kbqCqvByQ6rwo1RtdGZKRiAJzIYdQilmit0yHP77auEW-vGe9ltKzgsE) (after running M502, M500 and G28)

### Steps to Reproduce

1. Run marlin with UBL enabled
2. Create new UBL mesh
3. Look at the printer probe outside the bed and in random places

**Expected behavior:**

The printer should probe all points on the bed and successfully create a new UBL mesh

**Actual behavior:**

The Printer probes the first few point correctly and then does either too small or too large movements and ends up with the probe in the wrong position vs the expected XY

#### Additional Information
My hardware:
SKR 1.4 Turbo + TMC 2209 + BLTouch

I've tried creating the mesh both via pronterface and LCD

I've tried this both on 2.0.7.2 and the latest bugfix-2.0 as of writing this (3009707723d490c7dcc71093f5c461c782c16b18)

I've had this issue before about 1 year ago, see #16686, and ended up returning to AUTO_BED_LEVELING_BILINEAR as I could not get UBL to work.

I know my drivers are fine, I don't have any issues with printing, only while doing the UBL mesh creation, other auto bed levelling methods work fine.

If it would help I could make a video of the printer attempting to probe the bed, let me know if you want this.

francisduvivier, 2021-01-11:
I'm having this issue as well, and it occurs both with 3 point leveling as with UBL, It's inconsistent, I can't reproduce it reliably but I never have any problem during printing so I don't understand how this can go so wrong during leveling

francisduvivier, 2021-01-12:
Is this the same issue as https://github.com/MarlinFirmware/Marlin/issues/20497 ?

zmiguel, 2021-01-12:
> Is this the same issue as #20497 ?

No, here during probing the print head/probe end up in a different position than the expected as displayed on the coordinates. Making it probe the wrong positions and even outside of the bed.

zmiguel, 2021-01-21:
Is no one interested in having a look at this? It would be great to have UBL working.

slowbro, 2021-01-21:
@zmiguel I am actually upgrading to a 1.4 turbo soon and should be able to try to replicate this.

francisduvivier, 2021-01-22:
Hi, seems that with me this issue had to do with endstop noise because I was able to fix it with by uncommenting `#define ENDSTOP_NOISE_THRESHOLD 2`

I did however also sometimes have issues with homing before uncommenting that, and those are also fixed by it, but I didn't think it would also affect the leveling moves.

zmiguel, 2021-01-22:
@francisduvivier I'll try that, but I don't think I have endstop issues.

bepstein111, 2021-02-08:
Same problem, sensorless homing thought so I can't enable endstop noise threshold. Very annoying, this has been a problem for months that usually just dissipates by trying enough times, but with my last update (upgrading from a standard-style filament runout sensor to a smart runout sensor) I can't make it past move three or four without a jerky/incomplete movement.
BTW for me, it's ALWAYS an incomplete movement, never too long. This one issue is literally preventing me from using my printer, so I'd say this is serious enough for somebody to take a look at.

closed 2021-03-21 by github-actions[bot]