MarlinFirmware/Marlin #20717: [BUG] disabled steppermotors do not trigger untrusted state

steljwagh, 2021-01-08:
I did fix the problem with a pull request: https://github.com/MarlinFirmware/Marlin/pull/20714
But as tinkyhead pointed out, this should be working correctly already.

Problem:
When auto bed-leveling has been done, there are 2 ways to disable the steppers. Either wait for the time-out to occur or disable them from the menu manually.

After that if you execute a new bed-level for various reasons, the new auto bed-level starts without homing the axes first. This can be really dangerous for your steppers while axes could have moved in between. (For example, the Z axis drops a bit when the stepper is disabled).

A auto bedlevel is started with a G28O, I don't know why it still thinks it has a trusted axes state, but I think that has something to do with it.

Solution, https://github.com/MarlinFirmware/Marlin/pull/20714, but as pointed out by tinkyhead it should already work with the code in inderection.h

`#define  ENABLE_AXIS_X() if (SHOULD_ENABLE(x))  {  ENABLE_STEPPER_X();  ENABLE_STEPPER_X2(); AFTER_CHANGE(x, true); }
#define DISABLE_AXIS_X() if (SHOULD_DISABLE(x)) { DISABLE_STEPPER_X(); DISABLE_STEPPER_X2(); AFTER_CHANGE(x, false); set_axis_untrusted(X_AXIS); }
#define  ENABLE_AXIS_Y() if (SHOULD_ENABLE(y))  {  ENABLE_STEPPER_Y();  ENABLE_STEPPER_Y2(); AFTER_CHANGE(y, true); }
#define DISABLE_AXIS_Y() if (SHOULD_DISABLE(y)) { DISABLE_STEPPER_Y(); DISABLE_STEPPER_Y2(); AFTER_CHANGE(y, false); set_axis_untrusted(Y_AXIS); }
#define  ENABLE_AXIS_Z() if (SHOULD_ENABLE(z))  {  ENABLE_STEPPER_Z();  ENABLE_STEPPER_Z2();  ENABLE_STEPPER_Z3();  ENABLE_STEPPER_Z4(); AFTER_CHANGE(z, true); }
#define DISABLE_AXIS_Z() if (SHOULD_DISABLE(z)) { DISABLE_STEPPER_Z(); DISABLE_STEPPER_Z2(); DISABLE_STEPPER_Z3(); DISABLE_STEPPER_Z4(); AFTER_CHANGE(z, false); set_axis_untrusted(Z_AXIS); Z_RESET(); }`

steljwagh, 2021-01-08:
In other words, this code in G28.cpp is not working:

`  // Home (O)nly if position is unknown
  if (!axes_should_home() && parser.boolval('O')) {
    if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("> homing not needed, skip");
    return;
  }`

steljwagh, 2021-01-08:
Upon further investigation by debugging the procedure via serial.

The "set_axis_never_homed" is never executed, not after stepper time-out and not after manual disabling the steppers.

steljwagh, 2021-01-09:
I created this pull request as proposed fix
https://github.com/MarlinFirmware/Marlin/pull/20728

thinkyhead, 2021-01-11:
It's important to respect the `HOME_AFTER_DEACTIVATE` setting. When steppers sleep they are not "trusted" but they are still considered homed, and that should remain a fixed behavior. We don't want to change which flags get set. However, we do pick and choose how to use those flags throughout Marlin, in most cases based on the `HOME_AFTER_DEACTIVATE` setting.

Parts of Marlin that care about the `HOME_AFTER_DEACTIVATE` setting will call `axes_should_home` or one of the sub-functions that calls it, such as `homing_needed_error`.

Parts of Marlin that only care about whether the axis is "trusted" will call `axis_is_trusted`.

So, start from the specific behavior you're looking to adjust and see if changing `HOME_AFTER_DEACTIVATE` gives the behavior you want. If neither on or off gives you what you want, then look at which one of the above functions is being called and adjust that according to what seems more appropriate.

steljwagh, 2021-01-11:
I didn't knew about the `HOME_AFTER_DEACTIVATE` function, so thank you for sharing the rationale behind this position.
Just tested and discarded my 'fix', to find out this is exactly what I need.

I does raise another question, how about parking the head (for example filament change), I have to dig in to that to find out what's the best setting to keep, because the Z axis usually drops a step after the stepper are released. But, you don't want to get autohoming in the middle of your print....

closed 2021-01-11 by steljwagh