MarlinFirmware/Marlin #20718: [FR] get the real time z offset with given x y when all axes are moving

19504643, 2021-01-08:
I'm using Marlin2.0-bugfixed branch, and after I do the auto level, I get 4*4 grid data array
Bilinear Leveling Grid:
         0          1         2         3
 0 -3.907 -3.314 -2.507 -1.675
 1 -3.972 -3.292 -2.382 -1.667
 2 -3.895 -3.227 -2.513 -1.736
 3 -3.484 -3.282 -2.518 -1.786

how can I get the real time z offset when all axes are moving
that's to say, I want to get z offset when X=100mm, Y=100mm
is there already a function to return the calculated z offset with given (x, y)

rhapsodyv, 2021-01-08:
Enable `M114_DETAIL` in the configs. Then enabled levelling with `M420 S`. Now go to any position with `G1 X10 Y10` and run `M114 D`. It will tell:
```
Send: M114 D
Recv: X:0.00 Y:-22.00 Z:-2.60 E:0.00 Count A:-880 B:880 Z:-505
Recv: 
Recv: Logical: X: 0.000 Y:-22.000 Z:-2.602
Recv: Raw:     X: 0.000 Y:-11.000 Z:-1.340
Recv: Leveled: X: 0.000 Y:-11.000 Z:-1.262
Recv: UnLevel: X: 0.000 Y:-11.000 Z:-1.339
Recv: Stepper: X:-880 Y:880 Z:-505 E:0
Recv: FromStp: X: 0.000 Y:-11.000 Z:-1.262 E: 0.000
Recv: Diff:    X: 0.000 Y: 0.000 Z:-0.001 E: 0.000
```

Is it enough?

19504643, 2021-01-09:
@rhapsodyv ok, thanks

closed 2021-01-09 by 19504643