MarlinFirmware/Marlin #20719: [BUG] ? bugfix-2.0.x - BTT GTR V1.0 - stepper motors at re-start

DrumClock, 2021-01-08:
After turning on the power or restarting the printer:

Some stepper motors "jerk" and then lock for a few seconds.
As soon as the FW starts, the stepper motors are released.

It does not matter if the driver is controlled by UART, SPI or STEP/DIR.

rhapsodyv, 2021-01-10:
I don't know what are you reporting here. What is the issue? What is your configs?

Please, follow the issue template.

```
<!--

Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

-->

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

```

DrumClock, 2021-01-11:
HI, ... when you need problem templates .....

**Before reporting an error** 

1. Tested using the `bugfix-2.0.x` branch, dated `2021-01-11`.
2. HW configuration: `BTT GTR V1.0` board and `12864 LCD` controller
3. The problem still exists

**Bug Description**

Some stepper motors "jerk" and then lock for a few seconds.
As soon as the FW starts, the stepper motors are released.

**Configuration files**

It does not matter which configuration is used.
In the original files `Configuration.h` and` Configuration_adv.h` 
I changed only this:

```
#define SERIAL_PORT -1
#define MOTHERBOARD BOARD_BTT_GTR_V1_0
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
```

if I use any other configuration nothing will change on the error

**Steps to Reproduce**

 
1. define the BTT GTR V1.0 board
2. load the FW into the board
The 3rd error occurs immediately after turning on or restarting the printer

**Expected behavior:**

Stepper motors should be released.

**Actual behavior**

Some stepper motors "jerk" and then lock for a few seconds.
It is released after FW initialization.

**Additional information**

It does not matter if the driver is controlled by UART, SPI or STEP / DIR.
For example, the X, Y, E0 motors are blocked for me

rhapsodyv, 2021-01-11:
If it is release after initialisation, why `Some stepper motors "jerk" and then lock for a few seconds.` is a problem?? I can't see this as an issue.

> HI, ... when you need problem templates .....

Did it hurt? did you loose a finger following the template?? ... There're reasons for the template. Just now I understood what you are telling that is a "issue", that I didn't see as one...

DrumClock, 2021-01-12:
Hi @rhapsodyv  It's not a big issue.
It's just like ...... getting in the car, inserting the key and he would run the wiper himself .... 
it's not a problem either, but it's annoying.

closed 2021-01-15 by DrumClock