MarlinFirmware/Marlin #20736: [BUG] (UBL G29 P1 levelling fails to complete)

casiobearing, 2021-01-10:
Environment Summary - I've recently updated my delta to Marlin 2.0.7.2 from an old 8-bit marlin bugfix version.
Ported over config settings etc, everything works apart from G29 P1 (and G33 P2 but ignoring that for now).
The old 8-bit code worked fine for UBL. There no hardware problems. My delta uses the nozzle \ uSwitch for probing.
The bed is set up as level as possible and delta Z height set so nozzle at bed is Z=0 .
Controller board is a AtMega 2560 RUMBA.

I'm currently digging into 2.0.7.2 code, so this is a placeholder for feedback on what I have found so far....
The codebase has changed a lot from when I last looked into it so I'm still figuring out what does what, so any fixes below are temporary and might have better solutions :o)

### Bug Description

After issing a G29 P1, the 1st mesh point is probed correctly. Any subsequent mesh point probes fail (perhaps at the 2nd probe, or maybe the 5th, 6th or later probes.

### Relevant settings in Configuration.h
// Feedrate (mm/min) for the first approach when double-probing (MULTIPLE_PROBING == 2)
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z / 2
// Feedrate (mm/min) for the "accurate" probe of each point
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 4)

#define Z_CLEARANCE_BETWEEN_PROBES  6 // Z Clearance between probe points

### Possible code Issues found so far In "probe.cpp"
   Section covered by Lines 543 - 553 #elif Z_PROBE_SPEED_FAST != Z_PROBE_SPEED_SLOW

This section of code appears to be causing a gradual increase in physical nozzle height every time a mesh point is probed. Each mesh point probed increases the nozzle Z height by the mesh point Z error just triggered. This explains why sometimes it will probe many points before failing over... the larger the mesh point Z errors found the quicker it fails once the acculumated Z gets larger than the max probe depth.

As a temporary fix for this, I patched out the above section lines and then added after that section the following code to ensure the probe attempts ALWAYS start at a fixed height.
  do_blocking_move_to_z(Z_CLEARANCE_BETWEEN_PROBES, MMM_TO_MMS(Z_PROBE_SPEED_FAST));

This makes bed levelling work as I would expect, but it now shows a 2nd problem that is occuring.
The leveling now fails with a SLOW Probe fail! - Triggered early. Looking at leveling debug output I have now narrowed this failure down to a problem in "set_current_from_steppers_for_axis(Z_AXIS);"  (line 488 in probe.cpp)

When mesh probing is operating correctly, the above line extracts the triggered mesh point Z value, e.g.
13:32:07.582 : current_position= X13.56 Y-13.56 Z-2.40 : probe starts at +6mm, asked to probe down to max depth -2.40
13:32:07.582 : current_position= X13.56 Y-13.56 Z-0.15 : after trigger, set_current_from_steppers_for_axis() get -0.15.
13:32:07.582 : current_position= X13.56 Y-13.56 Z-0.15 : sync_plan_position

All good so far, but when the failure occurs I get the following....
13:32:08.491 : current_position= X40.67 Y-13.56 Z-2.40 : As previously, starts at +6mm (Z_CLEARANCE_BETWEEN_PROBES)
13:32:08.493 : current_position= X40.67 Y-13.56 Z6.00 : ERROR.... set_current_from_steppers_for_axis() has returned +6mm.
13:32:08.495 : current_position= X40.67 Y-13.56 Z6.00 : sync_plan_position (will use bad Z resulting in SLOW probe fail)

Hot off the press, so i've not yet dug further into set_current_from_steppers_for_axis(), but this seems to occur intermittantly - resulting in the SLOW Probe fail, effectively stopping the bed level process.

Would be useful if anyone else can concur with these observations possibly causing their UBL levelling problems.

casiobearing, 2021-01-10:
Latest update on 2nd issue set_current_from_steppers_for_axis() 
   From a printer powerup, the failure always occurs on the 9th mesh probe. Thereafter a G29 P1 failure will occur on the 5th mesh probe. On each run (8 done so far with 2 power ups) bed errors are slightly different (to be expected) but the fails consistenly happen on 9th probe for a fresh power up, or the 5th probe for a rerun of G29 P1.
This might be an initialisation issue, or\and a race condition happening somewhere.

casiobearing, 2021-01-11:
After more digging... its not set_current_from_steppers_for_axis() causing the problem, its the Z probe reading as triggered before it starts the z probe! At least its a dead consistent... From power up G29 P1 starts off ok until the beginning the 9th probe and then Z_MIN reads as triggered before it starts... hmmm.

sjasonsmith, 2021-01-17:
> delta Z height set so nozzle at bed is Z=0

Does your probe have a Z offset, or is the nozzle itself the probe? If you have a Z offset your Z height needs to be the distance from the bed to your probe, not to the nozzle.

As reported, I don't think anybody is going to help you with the issue. You have not attached configurations as the template requires. You are also missing the Expected/Actual behavior sections, which are important for somebody trying to quickly understand what the problem is.

I'll leave this open to give you a chance to update the description to include the missing information. If it's not added in a few days this will likely be closed.

closed 2021-02-24 by github-actions[bot]