MarlinFirmware/Marlin #20739: [BUG] Reboot on Z height

softwarecrash, 2021-01-10:
### Bug Description

Delta Printer style with TMC 2130 Drivers, BTT SKR 1.4 TURBO Board, 

The printer (Delta) makes at a specific Z Height a reboot, in the sample gcode file at Z37.7-Z38.5
its not a error of the gcode or slicer, testet wit cura slicer and simplify3d
testet with createt model and ready to use stl files

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/5793027/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/5793028/Configuration_adv.h.txt)
[CFFFP_Chat_Noir_Bell_Top.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/5793029/CFFFP_Chat_Noir_Bell_Top.gcode.txt)
or
https://www.file-upload.net/download-14434373/bug.zip.html

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [Testet with old Marlin config version 200006 and 200000 in both version this error not come up, but another error (extruder randomly pulls out and turn back) marlicn version config 20007 and upper this bug shows up]
2. [createt varius files height >50mm the printer abort with a reboot]
3. [tested with a minimal config to run the printer, withe the same result]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

rhapsodyv, 2021-01-11:
I noted you are using `#define SDCARD_CONNECTION LCD`. This issue happens if you use the ONBOARD sd?

softwarecrash, 2021-01-11:
> 
> 
> I noted you are using `#define SDCARD_CONNECTION LCD`. This issue happens if you use the ONBOARD sd?

tested now, with the internal SD Card the error comes not up,
the internal sd i can only acces over usb

bill-orange, 2021-01-11:
Are you sure that this relates to the SD card?  Can you print from Proterface or Octopi?

softwarecrash, 2021-01-11:
> 
> 
> Are you sure that this relates to the SD card? Can you print from Proterface or Octopi?

i think its not a SD problem

rhapsodyv, 2021-01-11:
Enable this: `NO_SD_HOST_DRIVE`, so your onboard SD will not be shared with host, then you can print with it.

I suspect your issue is on the LCD SD slot, or in the SPI, or the cable.

softwarecrash, 2021-01-11:
> 
> 
> Enable this: `NO_SD_HOST_DRIVE`, so your onboard SD will not be shared with host, then you can print with it.
yeah i found it and testet
> I suspect your issue is on the LCD SD slot, or in the SPI, or the cable.

since i have edit the microsteps the printer runs at the moment.
LCD SD slot was replaced by another new lcd, cable checked and replaced, all witghout success

softwarecrash, 2021-01-11:
Update: After some test with the postet changes, the printer restarts again. the issue still alive

softwarecrash, 2021-01-14:
Next Update:
I have replaced Board, Display, cable, Driver, SD Cards without success.

The board reboot, Show the Bootscreen and weit for User Input Like the printer switched on.

So i think its Not a Hardware issue

softwarecrash, 2021-01-16:
attached the gcode who stopped the printer and a video of the issue.
[stopping gcode.txt](https://github.com/MarlinFirmware/Marlin/files/5824944/stopping.gcode.txt)

softwarecrash, 2021-01-16:
the gcode area who stopps the printer
[stopping gcode.txt](https://github.com/MarlinFirmware/Marlin/files/5824950/stopping.gcode.txt)
and a video

softwarecrash, 2021-01-17:
i have found the problem!!! :D

my extruder driver overheating, thats not a problem, in the config is set that by overheat it decerase the current

the print abort after this message
00:30: E driver overtemperature warning! (1000mA)

without the decerase message.

so i think the funktion has a malefunction or the message that the driver current decerase was disturbs the communication with the sd card.

when i decerase the current the print end normaly.

closed 2021-02-27 by github-actions[bot]