MarlinFirmware/Marlin #20746: [BUG] MKS ROBIN NANO v2.0 TS35 SPI with encoder - garbage screen

filichev, 2021-01-11:
### Bug Description

If you turn on the stepper motor E0 and then use the encoder, garbage appears on the screen.

### Configuration Files

Included
[Configuration_and_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5794617/Configuration_and_adv.zip)
[pins_MKS_ROBIN_PRO_V2.h.zip](https://github.com/MarlinFirmware/Marlin/files/5794758/pins_MKS_ROBIN_PRO_V2.h.zip)

### Steps to Reproduce

1. Switch to the mode of movement of axes and press E- or E +
2. Rotate or push the encoder

**Expected behavior:**
Filament feeding.
Normal encoder response

**Actual behavior:**
When the encoder is connected, when you press E + or E-, garbage appears at the bottom of the screen. 
When the encoder is activated, garbage fills the entire screen. This happens ONLY if the extruder motor has been turned on before. If the extruder motor has not started yet, the encoder is working normally.

#### Additional Information
video https://youtu.be/9RClAyWFGi0
tmc2209 uart, tmc2208 standalone - There is a problem.
TFT_480x320_SPI or Upscaled 128x64 Marlin UI - There is a problem.

If the encoder is unsoldered from the board, there is no problem - everything works fine. Replacing the encoder, additional noise filters 100nf capacitors do not help.

thinkyhead, 2023-07-19:
ColorUI has been revamped a lot. Please confirm the latest `bugfix-2.1.x` code works (or not) and you can go ahead and close this issue if we solved the problem.

thisiskeithb, 2023-07-19:
I just tested this on a Robin Nano V2 + TS35 V2.0 SPI display and the TFT is fully white/non-functional on current bugfix.

mrjayy, 2023-10-22:
Confirmation:
    TS35 V2.0 SPI + bugfix-2.1.x results in white screen on the display

thisiskeithb, 2023-10-22:
> Confirmation:
> TS35 V2.0 SPI + ROBIN NANO V3.1 bugfix-2.1.x results in white screen on the display

That’ll be fixed in https://github.com/MarlinFirmware/Marlin/pull/26052

thinkyhead, 2026-10-08:
Thanks for the detailed report and video! The Color UI SPI/DMA path for the Robin Nano and TS35 has been reworked several times since this was filed, including #26052, which fixed the white-screen problem mentioned above. Could you flash the current bugfix-2.1.x and try the same steps (Move screen, E+/E-, then the encoder)? If the garbage still shows up, please attach your current configs and a photo, and tell us whether it still happens with the E0 motor unplugged. That will show whether it's electrical noise or a firmware bug.

closed 2026-10-13 by thinkyhead