MarlinFirmware/Marlin #20750: [BUG] #error "no match MCU defined" when enable USB host to GTR_V1.0

DJDOUBLEG, 2021-01-11:
GTR_v1.0
default_envs = BIGTREE_GTR_V1_0_usb_flash_drive
Can not compile the code.
I get this error: #error "no match MCU defined" (any many more pin map error e.g.:error: 'PB12' undeclared here (not in a function); did you mean 'PB_12'?)
When i set back to: default_envs = BIGTREE_GTR_V1_0
Then compiles well.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5798454/Marlin.zip)

Sh7777777, 2021-01-15:
Yes

sjasonsmith, 2021-01-17:
This happens because the `usb_flash_drive` environment doesn't inherit the build_flags from the normal GTR environment, so it is missing some MCU defines.

Adding those doesn't completely solve the problem, though. It just gets you to the next failures.

margaale, 2021-02-25:
Sorry to bring this to life, but how could i get it to compile?

gaby64, 2021-02-26:
I need help with this aswell

thisiskeithb, 2021-02-26:
Is this something you can look at @rhapsodyv?

gaby64, 2021-02-26:
Im having the same problem but with BIGTREE_SKR_PRO

thisiskeithb, 2021-02-26:
We don't need more "me toos" since the bug is confirmed & easily reproducible. I pinged rhapsodyv since they added this feature initially for MKS boards, but it should have also worked on others like the SKR Pro or GTR with flash drive support.

rhapsodyv, 2021-02-26:
Fixed here: #21197

DJDOUBLEG, 2021-02-26:
New build: 
[image]
:(

rhapsodyv, 2021-02-26:
You need to download my branch and not use -1 as serial port.

rhapsodyv, 2021-02-26:
Merged

PR #21197 (2021-02-26): Fix build for GTR and SKR PRO with usb flash drive support
### Description

Fix missing flags. Fix #20750
closed 2021-02-26 by rhapsodyv