MarlinFirmware/Marlin #20754: [BUG] Mesh Bed Leveling Defaults to "OFF" on (re)boot

Firedrops, 2021-01-12:
### Bug Description

I have a Sapphire Plus (MKS Nano V1.2), which I have flashed Marlin firmware to enable various features, such as manual Mesh Bed Leveling.

Every time the printer starts, the "Bed Leveling" option is hidden. I have to manually run "Auto Home" to make it visible, then change it from OFF (default) to ON. 

To clarify, this "Bed Leveling" option I'm talking about sets whether or not previously-generated mesh is loaded, NOT the one that puts the printer into Mesh Bed Leveling mode to generate a new mesh.

I do have "#define RESTORE_LEVELING_AFTER_G28" uncommented/active, but from further reading it appears that this line simply enables persistent memory of existing mesh values, rather than enable the "Bed Leveling" feature entirely.

What can I do to make the printer boot with "Bed Leveling" on by default?

### Configuration Files

[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/5800731/Config.Files.zip)

### Steps to Reproduce

1. Boot the Printer
2. Try any print, straight away

**Expected behavior:**

Nozzle is on the bed, following mesh bed leveling values.

**Actual behavior:**

Nozzle is ~1mm above bed

fettesb, 2021-01-24:
Put `M420 S` after your `G29` 😉

Firedrops, 2021-01-25:
Would that be in Cura's "Start G-code"? 

Or is there a way to permanently hard-tie `G28` with `M420 S` in Marlin?

ellensp, 2021-01-25:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-01-25 by ellensp

Firedrops, 2021-01-25:
You're kind of painting in broke strokes here to brush this off as a user-support question, no?

How do you separate code that does not function as intended from bugs? It doesn't make sense to me that the intent behind Mesh Bed Leveling was to have it disabled by default.