MarlinFirmware/Marlin #20756: [BUG] G29 not stopping on BLTouch error

mrseeker, 2021-01-12:
### Bug Description

When the BLTouch encounters an error, it halts and requires manual intervention from the user (by sending M999). However, the underlying G29 routine keeps going. The error itself might be related to #20736, however, I am focussing here mostly on the underlying routine and not on the probe failure itself.

### Configuration Files

<zipfile will follow soon>

### Steps to Reproduce

1. Turn machine on
2. Do a probe using G29 P1
3. Create an unrecoverable fault with the probe (keep the pin stowed or deployed)
4. See error about BLtouch not being able to probe
5. Try to clear M999 error.

**Expected behavior:**
The firmware should halt all execution and requires M999 to reset the probe. After that, it should redo it's last command in the queue.

**Actual behavior:**

The G29 command keeps continuing with probing, does not respond to commands from Octoprint and gets disconnected.

slowbro, 2021-01-20:
Based on your Config version, it looks like you're using the release branch; can you test with the bugfix-2.0.x branch and see if the error persists?

I tried replicating this on my Ender 3 Pro w/ Cr v4.2.7 board and I got the 'stop' LCD message when preventing the bltouch from deploying (after releasing it - I held it for some time, but the error appeared after releasing it)... but after the stop() was called, the machine halted and did not continue probing.

mrseeker, 2021-01-21:
Issue does not seem to appear in Bugfix branch, can't reproduce.

closed 2021-01-21 by mrseeker