MarlinFirmware/Marlin #20760: [BUG] M114 R & D not reporting real-time E values

cope413, 2021-01-12:
### Bug Description

Cannot get real time E values from either M114 R or M114 D.

### Configuration Files

[M114 Config files.zip](https://github.com/MarlinFirmware/Marlin/files/5805181/M114.Config.files.zip)

Tried on both Einsy RAMBo and Azteeg X5GT

### Steps to Reproduce

G92 E0
G1 Ennn Fnnn
M114 R during the move

**Expected behavior:**

M114 R should report real time E values

**Actual behavior:**

During an Extruder move, M114 R reports 0 until the move is over. Once the move is finished, it reports the current position
X, Y, and Z all report real-time positions

#### Additional Information

[M114 R terminal log.txt](https://github.com/MarlinFirmware/Marlin/files/5805197/M114.R.terminal.log.txt)
 (Line 178 shows the G1 E50 move, followed by M114 R reporting position of 0. Then reporting 50 after it's finished.
 Lines above 178 show X and Y moves reporting in real-time with M114 R)

cope413, 2021-02-12:
Still not reporting live E values

thisiskeithb, 2021-04-10:
Can you try #21579? `LIN_ADVANCE` broke realtime reporting.

thisiskeithb, 2021-04-10:
> Can you try #21579? `LIN_ADVANCE` broke realtime reporting.

Closing as the fix has been merged.

closed 2021-04-10 by thisiskeithb