MarlinFirmware/Marlin #20761: [BUG][Ender 3 v2][dwin] MYSERIAL1' was not declared in this scope

CRCinAU, 2021-01-12:
This seems to be introduced since yesterday, but I haven't found a commit yet that has broken this in bugfix-2.0.x

Possible its a Configuration update that's broken it.

Configs: https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3%20V2

```
Marlin/src/lcd/dwin/dwin_lcd.cpp: In function 'void DWIN_Send(size_t&)':
Marlin/src/lcd/dwin/dwin_lcd.cpp:85:20: error: 'MYSERIAL1' was not declared in this scope
   LOOP_L_N(n, i) { MYSERIAL1.write(DWIN_SendBuf[n]); delayMicroseconds(1); }
                    ^~~~~~~~~
Marlin/src/lcd/dwin/dwin_lcd.cpp:85:20: note: suggested alternative: 'MYSERIAL0'
   LOOP_L_N(n, i) { MYSERIAL1.write(DWIN_SendBuf[n]); delayMicroseconds(1); }
                    ^~~~~~~~~
                    MYSERIAL0
Marlin/src/lcd/dwin/dwin_lcd.cpp:86:20: error: 'MYSERIAL1' was not declared in this scope
   LOOP_L_N(n, 4) { MYSERIAL1.write(DWIN_BufTail[n]); delayMicroseconds(1); }
                    ^~~~~~~~~
Marlin/src/lcd/dwin/dwin_lcd.cpp:86:20: note: suggested alternative: 'MYSERIAL0'
   LOOP_L_N(n, 4) { MYSERIAL1.write(DWIN_BufTail[n]); delayMicroseconds(1); }
                    ^~~~~~~~~
                    MYSERIAL0
Marlin/src/lcd/dwin/dwin_lcd.cpp: In function 'bool DWIN_Handshake()':
Marlin/src/lcd/dwin/dwin_lcd.cpp:97:10: error: 'MYSERIAL1' was not declared in this scope
   while (MYSERIAL1.available() > 0 && recnum < (signed)sizeof(databuf)) {
          ^~~~~~~~~
Marlin/src/lcd/dwin/dwin_lcd.cpp:97:10: note: suggested alternative: 'MYSERIAL0'
   while (MYSERIAL1.available() > 0 && recnum < (signed)sizeof(databuf)) {
          ^~~~~~~~~
          MYSERIAL0
*** [.pio/build/STM32F103RET6_creality/src/src/lcd/dwin/dwin_lcd.cpp.o] Error 1
```

thinkyhead, 2021-01-13:
This will be solved when #20738 is merged.

closed 2021-01-13 by thinkyhead

CRCinAU, 2021-01-13:
Can confirm - is working ok now.