MarlinFirmware/Marlin #20762: [BUG] (short description)

h8dyt, 2021-01-13:
### Bug Description

I am trying to get my Ender3 v2 working with a bltoutch and currently get a "Needs a programmable LCD controller" or playing around getting Boot_screen needs graphical display

"

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5805761/Marlin.zip)
If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Enable the bltouch
2.  z_min_probe_endstop_inverting to true
3. comment out USE_PROBE_FOR_Z_HOMING
4. comment out Min_software_endstop_z
5. enable auto_bed_leveling_bilinear
6. grid_max_points x5
7. lcd_bed_leveling enable
8. level_bed_corners enable
9. z_safe_homing enable
[image]

**Expected behavior:**

I expected to be able to add the lcd control and corner manual leveling to the configuration for the use of a bltouch

**Actual behavior:**

Just keeps saying it needs a programmable lcd controller or a graphical display

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-01-13:
That feature needs a marlin programmable LCD not a serial smart lcd such as the DWIN_CREALITY_LCD.

You simply cannot use these features

closed 2021-01-13 by ellensp

ellensp, 2021-01-14:
But you may be in luck (in time), someone wants this feature added so much they have created https://github.com/MarlinFirmware/Marlin/issues/20773