MarlinFirmware/Marlin #20764: [BUG] MP_SCARA doesn't work

svsergo, 2021-01-13:
Marlin-bugfix-2.0.x_13.01.2021
For MP_SCARA to work, you need to make changes.
		
* motion.cpp

-1331       #if ENABLED(IS_SCARA)
     +1331  #if ENABLED(MORGAN_SCARA)

-1395            set_axis_homed(axis);
     +1395    #if !ENABLED(MP_SCARA) 
     +1396       set_axis_homed(axis);
     +1397    #endif 

-1404       #if ENABLED(MORGAN_SCARA)
     +1404  #if  IS_SCARA

-1555         #if IS_SCARA
     +1555	  #if ENABLED(MORGAN_SCARA)

* Conditionals_post.h
  
-146          #define QUICK_HOME
     +146  #if ENABLED(MORGAN_SCARA) 
     +147     #define QUICK_HOME
     +148  #endif	
  

* Conditionals_LCD.h
-938       #if ENABLED(MORGAN_SCARA) 
      +938 #if (ENABLED (MORGAN_SCARA) || ENABLED (MP_SCARA))
 
* Configuration.h

-1542          #define VALIDATE_HOMING_ENDSTOPS
       +1542 //#define VALIDATE_HOMING_ENDSTOPS

LexRozov, 2021-01-13:
Hello. Thank you for fixing the MP SCARA. After your changes, kinematics turned on, but a bright problem remained. I put different numbers in all lines of the scara section, but after G28 it invariably "SCARA Theta:18.64  Psi+Theta:-43.74"
In printrun:
"
Connecting...
start
Printer is now online.
Marlin bugfix-2.0.x
echo: Last Updated: 2021-01-12 | Author: (none, default config)
echo:Compiled: Jan 13 2021
echo: Free Memory: 5181  PlannerBufferBytes: 1232
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X480.00 Y480.00 Z200.00 E100.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z2.50 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X1000.00 Y1000.00 Z10.00 E20000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1000.00 R1000.00 T1000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X5.00 Y5.00 Z0.30 E3.00
echo:; SCARA settings: S<seg-per-sec> P<theta-psi-offset> T<theta-offset>
echo:  M665 S200.00 P0.00 T0.00 Z0.00
echo:; Material heatup parameters:
echo:  M145 S0 H180 B70 F255
echo:  M145 S1 H240 B100 F255
echo:; PID settings:
echo:  M301 P24.50 I1.72 D87.73
echo:  M304 P630.14 I121.71 D815.64
>>> G28
SENDING:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
SCARA Theta:18.64  Psi+Theta:-43.74
"
Always the same...

Please help fix this completely. Scara is needed by many people!!!

Sh7777777, 2021-01-13:
Yes

LexRozov, 2021-01-13:
After a few hours of practicing this firmware version with your changes, I found out that the position after the G28 command depends only on the values ​​in the lines: #define MANUAL_X_HOME_POS 0 and #define MANUAL_Y_HOME_POS. Changes to SCARA_OFFSET_X and SCARA_OFFSET_Y do not affect anything, nor do the other lines in the scara block.
Below are examples of how my printer calculates angles after G28 with only these two variables changing:

#define MANUAL_X_HOME_POS 0
#define MANUAL_Y_HOME_POS 0
SCARA Theta: -4473924.50 Psi + Theta: -4473924.50

#define MANUAL_X_HOME_POS 45
#define MANUAL_Y_HOME_POS 0
SCARA Theta: 83.54 Psi + Theta: -83.54

#define MANUAL_X_HOME_POS 45
#define MANUAL_Y_HOME_POS 45
SCARA Theta: 125.85 Psi + Theta: -35.85

#define MANUAL_X_HOME_POS 0
#define MANUAL_Y_HOME_POS 45
SCARA Theta: 173.54 Psi + Theta: 6.46

#define MANUAL_X_HOME_POS 15
#define MANUAL_Y_HOME_POS 60
SCARA Theta: 157.07 Psi + Theta: -5.14

#define MANUAL_X_HOME_POS -45
#define MANUAL_Y_HOME_POS 45
SCARA Theta: 215.85 Psi + Theta: 54.15

#define MANUAL_X_HOME_POS -21
#define MANUAL_Y_HOME_POS -21
SCARA Theta: 28.79 Psi + Theta: -150.22

#define MANUAL_X_HOME_POS 319.85
#define MANUAL_Y_HOME_POS -20.47
SCARA Theta: 46.16 Psi + Theta: -45.76

#define MANUAL_X_HOME_POS -245
#define MANUAL_Y_HOME_POS 215
SCARA Theta: 187.46 Psi + Theta: 97.46

#define MANUAL_X_HOME_POS -215
#define MANUAL_Y_HOME_POS 245
SCARA Theta: 180.00 Psi + Theta: 90.00

The last two configurations told me that the default firmware is left-handed robot.

My suggestions:
1- Add a line defining right-handed or left-handed to the scara block.
2- Repair the settings in the scara block. (In my opinion, it is easier to adjust the scara through the angles THETA_HOMING_OFFSET and PSI_HOMING_OFFSET than to draw the position of the hands and calculate the position relative to the origin.)

P.S.
Dear friends, please finish the scara. It doesn't matter how! The main thing is that it works !!! Marlin is the best in my opinion and I want to use it in my scara robots. I am convinced that I am not alone !!!

svsergo, 2021-01-14:
Try the attached file for right-handed SCARA.

[MP_SCARA.pdf](https://github.com/MarlinFirmware/Marlin/files/5812530/MP_SCARA.pdf)

[scara.cpp.txt](https://github.com/MarlinFirmware/Marlin/files/5812538/scara.cpp.txt)

https://www.youtube.com/watch?v=6M0xGTaFwww

myseg, 2021-01-14:
Sorry for dropping in out of the blue, but thanks man, you saved my day too! I really hope SCARA kinematics will be fixed soon officially.

LexRozov, 2021-01-14:
Everything is working! Thank you so much!!!

sjasonsmith, 2021-01-17:
If there are changes needed for SCARA, I invite one you to post pull requests with the changes.

I'm not aware of any key project contributors actively working with SCARA. The changes will be much more reliable if one you testing it can submit them.

thinkyhead, 2021-02-17:
I've created a PR based on recommended changes by @svsergo. Assuming the proposed changes here worked well, the linked PR should also get things working.

closed 2021-02-17 by thinkyhead