MarlinFirmware/Marlin #20765: [BUG]About "Auto bed level dont deploy" and "Z Safe Homing dont move to Center" and "IDEX Dual X Carriages" and "X Max Endstop".

xsuper9988, 2021-01-13:
[BUG]About "Auto bed level dont deploy" and "Z Safe Homing dont move to Center" and "IDEX Dual X Carriages" and "X Max Endstop".

Marlin Version is 2.0.7.2

= = = = = = = = = = = = = = = = = = = = 

setting situation 1.

normal XYZ axis
not IDEX mode
not X max endstop

result 1.

Z Safe Homing is OK
Auto bed level is OK

= = = = = = = = = = = = = = = = = = = = 

setting situation 2.

X1 X2 Y Z axis
IDEX mode
X max endstop

result 2.

Z Safe Homing is OK
Auto bed level is NG

When Auto Homing finish
then Goto MainPage Bed Level not Deploy

= = = = = = = = = = = = = = = = = = = = 

setting situation 3.

X1 X2 Y Z axis
IDEX mode
X max endstop
#define HOMING_BACKOFF_POST_MM { 10, 10, 10 }

result 3.

Z Safe Homing is OK
Auto bed level is NG

When Auto Homing finish 
(Now X Max Endstop not on，because HOMING_BACKOFF_POST_MM)
Bed Level not Deploy
and System crash restart

= = = = = = = = = = = = = = = = = = = = 

setting situation 4.

normal XYZ axis
not IDEX mode
X max endstop On(Because Machine hardwear is IDEX X2 at Max pos.)
#define HOMING_BUMP_MM      { 10, 10, 10 }

result 4.

Z Safe Homing is NG
Auto bed level is NG

X homing to X Min Endstop
and Not HOMING_BUMP_MM(Because X Max Endstop On)
X no homing finish(But System no Err MSG)
Then Y、Z Homing OK and finish
and not move to Z Safe POS.
(Because X Max Endstop On,and maybe program use move_xy function,so...)

Now Machine XY stop at (0,0)
But MainPage System Pos. at XY_CENTER

If now goto Bed Level
XY will be goto first point and dangerous bump
Because Machine POS. not equel System POS.

PS:When setting situation 4. X max endstop On,
i still can move X forward and backward in LCD_MENU!

= = = = = = = = = = = = = = = = = = = = 

OTHER TEST PS:
When i enable "LIN_ADVANCE" system is crash and restart,My Board is ATmega2560!

ellensp, 2021-01-13:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

ellensp, 2021-01-13:
```
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

<THE TEMPLATE>

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

xsuper9988, 2021-01-13:
> Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Sorry,I will be follow the instructions next time
and i will test "bugfix-2.0.x` branch" and report

xsuper9988, 2021-01-13:
> Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

bugfix-2.0.x branch Cant move Y、Z axis，but Y、Z axis has enable!

ellensp, 2021-01-13:
Please attach your configuration files. For bugfix

xsuper9988, 2021-01-13:
> Please attach your configuration files. For bugfix. Also any other files you edited.

https://drive.google.com/drive/folders/1gAF5XO7rOaSMMx9dJWwV7KRvaRfH0zfJ?usp=sharing

sjasonsmith, 2021-01-17:
I am closing this, because from the description I cannot identify the actual problem.

When reporting an issue, please be sure to follow all instructions at the top of the template, and leave the sections headings intact. In addition to configuration files, the **Expected Behavior** and **Actual Behavior** sections are extremely important, to help contributors understand the problem.

If you have multiple issues, I suggest you focus on one of them first. Reporting many different issues together makes it unlikely that people will be able to help.

For reference, these are the instructions included in the template:

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

closed 2021-01-17 by sjasonsmith

xsuper9988, 2021-01-18:
I’m sorry that I didn’t follow the template and caused trouble to everyone.

I am testing the latest branch version now.

But Y Z cant motion

Someone in the "Discord" has the same situation as me.

So there is no way to have new test results.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
不含病毒。www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>