MarlinFirmware/Marlin #20766: [BUG] Pt100 direct (147) reads 0 at room temp

karabas2011, 2021-01-13:
### Bug Description

I tried to connect Pt100 without amplifier to SKR1.3 ( pullup 4.7k)
I set 147 type
I got 0 reading
If I try to heatup I got E0 min temp failure.

So it easy to replicate. The same was described in 2016

ellensp, 2021-01-13:
```
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

<THE TEMPLATE>

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-01-13:
are you pulling up to 5v or 3.3v
I suspect 5v and that is your issue. This is s 3.3v controller (and analog pins are not 5v tolerant.) 
You need to pullup to 3.3v or the tables will not work.

karabas2011, 2021-01-13:
Sorry I have updated. As I understand 147 table is currently flat instead if Prusa version.

karabas2011, 2021-01-13:
Sorry, no pullup resistor actually

closed 2021-01-13 by karabas2011

oldman4U, 2021-01-14:
karabas2011

It would be great if you could take care for your other open tickets in the BTT TFT repository. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues

Thank you