MarlinFirmware/Marlin #20774: [BUG][DWIN] Encoder scroll seems to act as button press as well.

CRCinAU, 2021-01-14:
Builds after the 2021-01-12 date stamp seem to have a regression where selecting input values will exit the input field after scrolling one step.

This can be reproduced via navigating to: Prepare -> Motion -> Steps/mm -> Steps/mm E

Press the button to select, then scroll to change the value. The value will change by 0.1 and then exit the selection.

Configs:
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3%20V2

A quick browse of the commit history since the 12th throws this up as a likely candidate, but unable to confirm atm:
https://github.com/MarlinFirmware/Marlin/commit/4a897310252341f203371cb2e99f93bf9c035143

Cc: @thinkyhead (author of referenced commit)

dimonoid, 2021-01-14:
2b928b4 has the same problem

JohnClementine, 2021-01-14:
Just wanted to chime in and say I'm also experiencing this behavior

dimonoid, 2021-01-14:
What was your latest working commit? @CRCinAU

gmergulhao, 2021-01-14:
Have same issue. I can add that this was working fine on nightly build on 29.12 for the ender 3 v2.

dimonoid, 2021-01-14:
> 
> 
> Have same issue. I can add that this was working fine on nightly build on 29.12 for the ender 3 v2.

Have you flashed the monitor firmware only, or board also? For some reason the repository with monitor firmware got rebased, so that there is no old commits left.

gmergulhao, 2021-01-14:
I've flashed the board firmware only [Ender3-V2-BLTouch-20210114.bin](https://marlin.crc.id.au/firmware/Ender%203%20V2%20-%20BLTouch/)

DWIN_SET I'd have to compile first. Should I go for it? Still stock that came on the printer. Bought it around Christmas.

Thank you for your Input!

CRCinAU, 2021-01-14:
This is not related to the DWIN screen firmware.

dimonoid, 2021-01-15:
https://github.com/MarlinFirmware/Marlin/commit/4472ba2b6bc0835d0fcc039ad66d9dab40e13010

Using binary search, I found that this bug is in this commit. You are welcome.

SamLefebvre, 2021-01-15:
Thanks @dimonoid I use the old commit and it's working.

MD3D20, 2021-01-15:
i too am having this issue. what can i do to fix it im new to firmware programming

closed 2021-01-16 by CRCinAU