MarlinFirmware/Marlin #20778: [BUG] Cannot compile Marlin for SKR mini E3-DIP V1.1

Auravendill, 2021-01-14:
I cannot compile the current Marlin 2.0.x branch for my SKR mini E3-DIP V1.1. I've tried on both Windows and Linux and switched back to the example config from Configurations-release-2.0.7.2 for this board in an Ender 3 to minimize all error sources as far as I can. Compiling for other boards like the SKR v1.3 works totally fine.

This is the output in Platformio (VS Code, Debian 10):

Processing STM32F103RC_btt_USB (platform: ststm32@~6.1; board: genericSTM32F103RC; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 (6.1.1) > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <USBComposite for STM32F1> 0.91
|-- <TMCStepper> 0.7.1
|   |-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.3
|   |-- <Wire> 1.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <Wire> 1.0
|-- <FreeRTOS701>
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduinoVariant/board.cpp.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduinoVariant/wirish/boards.cpp.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduinoVariant/wirish/boards_setup.cpp.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduinoVariant/wirish/start.S.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduinoVariant/wirish/start_c.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduinoVariant/wirish/syscalls.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/HardwareTimer.cpp.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/Print.cpp.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/WString.cpp.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/avr/dtostrf.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/cxxabi-compat.cpp.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/ext_interrupts.cpp.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/hooks.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/itoa.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/adc.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/adc_f1.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/bkp_f1.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/dac.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/dma.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/dma_f1.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/exc.S.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/exti.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/exti_f1.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/flash.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/fsmc_f1.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/gpio.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/gpio_f1.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/i2c.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/i2c_f1.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/iwdg.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/nvic.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/pwr.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/rcc.c.o
Compiling .pio/build/STM32F103RC_btt_USB/FrameworkArduino/libmaple/rcc_f1.c.o
[... 428 lines cut ...]
At top level:
/home/daniel/.platformio/packages/framework-arduinoststm32-maple@2.10000.200103/STM32F1/libraries/FreeRTOS701/utility/port.c:285:6: warning: 'prvSetupTimerInterrupt' defined but not used [-Wunused-function]
 void prvSetupTimerInterrupt( void )
      ^~~~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/STM32F103RC_btt_USB/libc9e/FreeRTOS701/utility/tasks.c.o
Compiling .pio/build/STM32F103RC_btt_USB/libc9e/FreeRTOS701/utility/timers.c.o
Compiling .pio/build/STM32F103RC_btt_USB/lib393/Servo/Servo.cpp.o
Archiving .pio/build/STM32F103RC_btt_USB/lib393/libServo.a
Indexing .pio/build/STM32F103RC_btt_USB/lib393/libServo.a
Archiving .pio/build/STM32F103RC_btt_USB/libc9e/libFreeRTOS701.a
Indexing .pio/build/STM32F103RC_btt_USB/libc9e/libFreeRTOS701.a
Linking .pio/build/STM32F103RC_btt_USB/firmware.elf
/home/daniel/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32F103RC_btt_USB/firmware.elf section `.rodata' will not fit in region `rom'
/home/daniel/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 4476 bytes
collect2: error: ld returned 1 exit status
*** [.pio/build/STM32F103RC_btt_USB/firmware.elf] Error 1
============================================================ [FAILED] Took 97.79 seconds ============================================================

Environment          Status    Duration
-------------------  --------  ------------
STM32F103RC_btt_USB  FAILED    00:01:37.791
======================================================= 1 failed, 0 succeeded in 00:01:37.791 =======================================================
Der Terminalprozess "pio 'run'" wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.

qwewer0, 2021-01-14:
> `rom' overflowed by 4476 bytes

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: [https://discord.gg/n5NJ59y](https://discord.gg/n5NJ59y)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Auravendill, 2021-01-14:
But something in Marlin changed and now I cannot build for this one mainboard anymore. Neither with my own config files nor with the example files that are linked within Marlin itself. Reading documentations (again) on how to config marlin won't change that. I am usually more than capable to get my configs to work, my other printer all run just fine with 2.0.7.2, but with different mainboards.

On the printer in question is currently running Marlin 2.0.6.1. So it was obviously possible to compile for it back then.

In "Check All" it says:
Marlin/src/pins/pins.h:747: [high:error] #error Unknown MOTHERBOARD value set in Configuration.h [preprocessorErrorDirective] 
even though the mainboard is defined in boards.h (Not sure if that means anything though)

I can of course ask on Discord too, but I rather doubt, that they will find the issue... But I wouldn't mind getting proven wrong.

ellensp, 2021-01-14:
".rodata' will not fit in region rom'"

The firmware you have created is to large for the amount of flash on your controllers.
Disable some features.

ellensp, 2021-01-14:
also use bugfix for this board ststm32@10 produces much smaller code.

closed 2021-01-14 by ellensp

Auravendill, 2021-01-14:
But if compile with STM32F103RC_btt_512K_USB (which works (but might be unstable etc or will just crash)), it says, I am using 46,1% flash (241948 bytes), which means I should still be safe with only the 256k I normally have with the option I used before.

And the example config already uses less features than I am currently using in 2.0.6.1 and apparently doesn't fit anymore? Did 2.0.7.2 get so much worse in terms of space efficiency or are new features enabled in the background that I haven't noticed yet?

But if those issues will be fixed in the next release, I will be looking forward to it. If my board crashes with the 512K version, I might consider trying bugfix, but I prefer to use a stable branch as much as possible.

Thank you very much for your help!