MarlinFirmware/Marlin #20779: [BUG] (short description)

zwep, 2021-01-14:
### Bug Description

Compilation goes wrong... I really have no idea why or how... Here is the output

### Configuration Files

I have used the config files from 

https://github.com/MarlinFirmware/Configurations/archive/release-2.0.7.2.zip

for the Creality\Ender-3\BigTreeTech SKR Mini E3 2.0

### Steps to Reproduce
1. Download latest version from Github of Marlin
2. Get the config files stated above
3. Use windows 10 + Atom + PlatformIO + python + git
4. Make sure that the MOTHERBOARD parameter is set correctly, and that the platformio.ini file is updated to the proper board as well

**Expected behavior:**

Succesful compilation

**Actual behavior:**

Marlin\src\HAL\STM32F1\eeprom.cpp: In static member function 'static bool PersistentStore::write_data(int&, const uint8_t*, size_t, uint16_t*)':
Marlin\src\HAL\STM32F1\eeprom.cpp:49:14: error: 'eeprom_read_byte' was not declared in this scope
     if (v != eeprom_read_byte(p)) {
              ^~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\eeprom.cpp:49:14: note: suggested alternative: 'pgm_read_byte'
     if (v != eeprom_read_byte(p)) {
              ^~~~~~~~~~~~~~~~
              pgm_read_byte
Marlin\src\HAL\STM32F1\eeprom.cpp:50:7: error: 'eeprom_write_byte' was not declared in this scope
       eeprom_write_byte(p, v);
       ^~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\eeprom.cpp:50:7: note: suggested alternative: 'gpio_write_bit'
       eeprom_write_byte(p, v);
       ^~~~~~~~~~~~~~~~~
       gpio_write_bit
Marlin\src\HAL\STM32F1\eeprom.cpp: In static member function 'static bool PersistentStore::read_data(int&, uint8_t*, size_t, uint16_t*, bool)':
Marlin\src\HAL\STM32F1\eeprom.cpp:65:17: error: 'eeprom_read_byte' was not declared in this scope
     uint8_t c = eeprom_read_byte((uint8_t*)pos);
                 ^~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\eeprom.cpp:65:17: note: suggested alternative: 'pgm_read_byte'
     uint8_t c = eeprom_read_byte((uint8_t*)pos);
                 ^~~~~~~~~~~~~~~~
                 pgm_read_byte
Marlin\src\HAL\STM32F1\eeprom.cpp: In static member function 'static size_t PersistentStore::capacity()':
Marlin\src\HAL\STM32F1\eeprom.cpp:74:45: error: 'E2END' was not declared in this scope
 size_t PersistentStore::capacity() { return E2END + 1; }
                                             ^~~~~
*** [.pio\build\STM32F103RC_btt\src\src\HAL\STM32F1\eeprom.cpp.o] Error 1
========================= [FAILED] Took 15.97 seconds =========================
 
Environment      Status    Duration
---------------  --------  ------------
STM32F103RC_btt  FAILED    00:00:15.965

#### Additional Information

It looks a lot like this issue

https://github.com/MarlinFirmware/Marlin/issues/18765

But there they state that from 2.0.5 (or so) on it should be fixed. So I dont get how the bug is still there in 2.0.7 (or at least for me)

qwewer0, 2021-01-14:
> for the Creality\Ender-3\BigTreeTech SKR Mini E3 2.0

Was able to compile without any issue.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: [https://discord.gg/n5NJ59y](https://discord.gg/n5NJ59y)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ellensp, 2021-01-14:
Don't use atom, use vscode.

closed 2021-01-14 by ellensp

zwep, 2021-01-15:
Hi @ellensp ,

can you explain how that could change anything? Of course I am going to try it.. but I would like to understand how they compile the program differently

PS: sorry that I posted it here. The issue that I pointed at (#18765) also dealt with a compilation error, so I assumed that it was ok to post that here. Thanks for the clarification

ellensp, 2021-01-15:
Platform Io officially dropped support for atom some time back.

zwep, 2021-01-15:
Dangit, I have no Idea what I did differently. I just repeated the steps above.. download Marlin, download the configs, change platform.ini..
And now it compiles. Both on Atom and VScode btw.

Thanks for your time and sorry for the bother.