MarlinFirmware/Marlin #20781: [BUG] [Ender 3 v2] Manual movement along the y-axis makes extruder pull filament out of the bowden tube.

dimonoid, 2021-01-15:
### Bug Description

This happens only the first time after either:
1) I switch on printer and manually move y-axis up.
or
2) Stop print, and manually move y-axis up.

### Configuration Files

[bugYaxis.zip](https://github.com/MarlinFirmware/Marlin/files/5818311/bugYaxis.zip)
**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

The only thing I modified is bed size.

I am using this build
https://github.com/MarlinFirmware/Marlin/commit/7f3dcb3e8a8cc1a82c1d9315e9fada5ff59058f2

### Steps to Reproduce

**Expected behavior:**
1) Switch on printer
2) Move in settings carriage along y-axis
3) See that filament is not moving.

**Actual behavior:**
1) Switch on printer
2) Move in settings carriage along y-axis
3) See that extruder is pulling the filament really quickly from the bowden tube.
4) If try moving second/third/etc. times again, everything works normally again.

It looks

CRCinAU, 2021-01-16:
Is this the same issue as: https://github.com/MarlinFirmware/Marlin/issues/19944 ?

dimonoid, 2021-01-16:
> Is this the same issue as: #19944 ?

No, they are discussing retraction when moving along Z axis or X/Y/Z simultaneously. I experience same issue, but only with Y axis. I have no ideas why there is a difference.

Klaws--, 2021-01-17:
In #19944 it was recently confirmed that a move on any axis can cause this issue.

sjasonsmith, 2021-01-17:
After reviewing the latest comments in #19944 I think it covers this scenario exactly. Fixing that will almost certainly fix this as well.

sjasonsmith, 2021-01-17:
Duplicate of #19944

closed 2021-01-17 by sjasonsmith