MarlinFirmware/Marlin #20787: [BUG] MKS_GEN_L - Parts Fan 100% when preheat PLA or PETG, bed only selected.

2dbit, 2021-01-16:
### Bug Description

Adimlab Gantry Pro OEM with MKS GEN L board installed (RAMPS 1.4).   When selecting either PLA bed heat, or PETG bed heat, the bed begins to heat, but the parts cooling fan also goes to 100%.   Selecting End only or End/Bed preheat does not cause the parts fan to go 100%.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

[My Configuration and Pins.zip](https://github.com/MarlinFirmware/Marlin/files/5823908/My.Configuration.and.Pins.zip)

If you've made any other modifications describe them in detail here.
Outside of config.h and config.adv.h, the MKS_GEN_L.h file had to be modified with MOSFET_D_PIN 7 commented out to allow the caselight to work, as it is connected to that pin.   

### Steps to Reproduce

The version of Marlin that is compiled is 2.0.7.2, the latest "nightly build"   
I have reflashed the firmware several times, including following up with M502 and M500.  The issue persists.

**Expected behavior:**
Select preheat bed only and have the parts fan stay off.

**Actual behavior:**
parts fan goes to 100% when bed heats.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

sjasonsmith, 2021-01-17:
Are you able to properly control that fan by itself to the LCD menus?

2dbit, 2021-01-18:
> Are you able to properly control that fan by itself to the LCD menus?

Yes, once preheat bed only is selected, and the fan sets to 100%, you can go to the fan setting and turn it back to zero.   Bed continues to heat.

rhapsodyv, 2021-01-31:
Did you test bugfix?

Because it was fixed 3 months ago on bugfix, here: https://github.com/MarlinFirmware/Marlin/pull/19773 . But 2.0.x branch and 2.0.7.2 doesn't have the fix yet.

2dbit, 2021-01-31:
> Did you test bugfix?
> 
> Because it was fixed 3 months ago on bugfix, here: #19773 . But 2.0.x branch and 2.0.7.2 doesn't have the fix yet.

Yes, the version of Marlin being run is built from the latest “bugfix - nightly build”

rhapsodyv, 2021-01-31:
There's no such a thing as "nightly build".

Your configs says you aren't using bugfix. The config version on bugfix is 020008 and yours is 020007.

Please test again using Marlin bugfix branch.

2dbit, 2021-01-31:
Hmm, will confirm and update.

2dbit, 2021-02-01:
> There's no such a thing as "nightly build".
> 
> Your configs says you aren't using bugfix. The config version on bugfix is 020008 and yours is 020007.
> 
> Please test again using Marlin bugfix branch.

Thank you!    I don't know how I downloaded the latest release version thinking I was downloading bugfix, but you caught it!   You were absolutely correct.  The issue is resolved in the bugfix version.     I will go ahead and close this if deemed appropriate to do so at this time.   Again, thank you.

rhapsodyv, 2021-02-01:
It's ok. It's a little confusing all those version and bugfix. Thanks!

closed 2021-02-01 by rhapsodyv